Koenkk/zigbee2mqtt #8560: Heiman remote not properly recognized

akala57, 2021-08-29:
Hi,
Since several month, my three remote Heiman HM1RC-2-E doesn't work properly.  It's recognized, battery information is ok but actions and linkquality doesn't work.
[image]
I'm using Conbee II but I used CC2531 previously.

I already tried to :
- delete device and repair it
- reinstall from scratch zigbee2mqtt and mosquitto, then pair remote again
- change battery
- tried with another dongle (conbee II) on another installation
The problem remain the same.
What could I do ? 
Thank you in advance.
Sincerly.

geijt, 2021-09-17:
I have the same issue

bdherouville, 2021-09-27:
Hi ! I have the same issue. It is very difficult to make them pair. The interviewing process seems to be very long.

akala57, 2021-10-17:
Hi @Koenkk,
Do you have any Idea ? I can't open my garage door.
Thank you.

Koenkk, 2021-10-17:
Could you provide a sniff of the traffic starting from the point where you trigger an action on the remote?

Please provide the sniff here + your network key: https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

bdherouville, 2021-11-17:
These remotes now works perfectly fine and I use 4 of them. 
At the begining I had a lot of issues due to Wifi and USB3 interferences.
On my side the issue can be closed.

geijt, 2021-11-23:
I've replaced my Conbee II with a CC2652RB stick from Slaesh and now the remotes work perfectly. Only the reported signal strength of all devices is much lower than with the Conbee II. I still need to see or the lower signal strength will be an issue

closed 2021-12-31 by github-actions[bot]