Koenkk/zigbee2mqtt #8563: Blocklist doesn't work

VladDrakul1986, 2021-08-29:
while building up my network new i fetched some devices from my neighbours since i don't what to control them i add them to configuration.yaml.
```
serial:
  port: /dev/ttyUSB0

blocklist:
  - '0x5c0272fffe6165e1'
  - '0x00158d000192cfe8'
  - '0x00158d000192dbc1'
  - '0x00158d00041cc845'
  - '0x0017880106d41190'
  - '0x0017880106d41278'
  - '0x5c0272fffe863d62'
  - '0x847127fffe44aad5'
devices:
```
and delete the devices from my devices section and restart zigbee2mqtt.
But when iam watching the mqtt messages, i can still see new messages from the actually blocked devices :(

sjorge, 2021-08-29:
You need to tell the device to leave the network too, IIRC blocklist only tells z2m to not interview them.

VladDrakul1986, 2021-08-30:
ah thank you now it worked.

closed 2021-08-30 by VladDrakul1986