Koenkk/zigbee2mqtt #8566: Cover control from Home Assistant does not work correctly

yonaesp, 2021-08-29:
## What happened
After adding the cover to home assistant automatically I cannot raise or lower the cover from Home Assistant when it is fully open or closed (state 0 or 100).
It seems that Home Assistant locks the down button after raising the shade completely and locks the up button after lowering the shade completely.
as I have read in the tutorial about adding a cover:
https://www.zigbee2mqtt.io/devices/TS130F.html
I must open the cover completely and this should be 100, then close completely and it should be 0
But after calibrating several times, reversing the motor and so on, the problem persists, and open is 0 and closed is 100.
But it is strange because the buttons are not reversed.
From zigbee2mqtt if I can perfectly control the cover and also through mqtt commands.

## What did you expect to happen
I should be able to lower the cover after raising it all the way from home assistant.
## How to reproduce it (minimal and precise)
Raise the shade all the way and the down button will stop working from Home Assistant.
If I put the cover in a position other than 100 and 0, I can go up and down without problem again from Home Assistant.
## Debug info
Log: https://pastebin.com/SxxQ1s5z
Zigbee2MQTT version: 1.21.0
Adapter hardware: Conbee II
Adapter firmware version: deCONZ_RaspBeeII_0x26720700
[image]

danieledwardgeorgehitchcock, 2021-08-31:
Have you tried adding an `invert_cover` option to the device in your Zigbee2MQTT config file?

yonaesp, 2021-08-31:
> Have you tried adding an `invert_cover` option to the device in your Zigbee2MQTT config file?
> 
> Instructions on how to do this are provided in the link you gave above

You are right!
In the instructions I interpret that it is enabled from the panel as I show in my video.
But now I have added this option in the configuration.yaml file and it works perfectly.
Thank you so much

Code of configuration.yaml in case it can be of use to someone:
`devices:
  '0x086bdXXXXcd3f6e':
    friendly_name: Cover1n
    invert_cover: true`

danieledwardgeorgehitchcock, 2021-08-31:
The motor reversal exposed element is more for setting the device i.e. a window opener cover would have the window open at 100% whereas a roller shutter would have the shutter closed at 100% - the values would be reported back to Zigbee2MQTT the same in both scenarios whereas the invert_cover applies the setting at the server level and is therefore applied across the platform through to Home Assistant.

I hope this clears things up a bit and thanks for confirming that this works! Please close out the issue if you are now happy with the devices behaviour

closed 2021-08-31 by yonaesp