Koenkk/zigbee2mqtt #8575: Device Id TS110E Manufacturer name _TZ3210_ngqk6jia

asanchez156, 2021-08-30:
## Information about the device + link

**Device model ID:** TS110E
**Manufacturer name:** _TZ3210_ngqk6jia
**Aliexpress link:** https://es.aliexpress.com/item/4001279149071.html?spm=a2g0s.9042311.0.0.2ea863c00bivCw

## data/database.db entry of the device

```bash
{"id":28,"type":"Router","ieeeAddr":"0x00000000000000","nwkAddr":22263,"manufId":4417,"manufName":"_TZ3210_ngqk6jia","powerSource":"Mains (single phase)","modelId":"TS110E","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":257,"inClusterList":[5,4,6,8,57345,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS110E","manufacturerName":"_TZ3210_ngqk6jia","powerSource":1,"zclVersion":3,"appVersion":66,"stackVersion":0,"hwVersion":1,"dateCode":""}},"genOnOff":{"attributes":{"onOff":0,"moesStartUpOnOff":1}},"genLevelCtrl":{"attributes":{"61440":447,"64515":10,"64516":1000}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b00219fbd82","endpointID":1},{"cluster":8,"type":"endpoint","deviceIeeeAddress":"0x00124b00219fbd82","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":3600,"repChange":0}],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":66,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":1461352984},"lastSeen":1629900231798}
```

I figured out that in the object `genLevelCtrl` the fields values are:
{"61440":447,"64515":10,"64516":1000}}}

61440 for the current brightness.
64515 the minimum brightness available 10 => 0%.
64516 the maximum brightness available 1000 => 100%.
I make the on/off work with the following definition.

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    fingerprint: [{ modelID: 'TS110E', manufacturerName: '_TZ3210_ngqk6jia' }],
    model: 'TS110E_dimmer',
    vendor: 'TuYa',
    description: 'Smart dimmer module with neutral',
    extend: extend.light_onoff_brightness({noConfigure: true}),
    exposes: [e.light_brightness()],
    configure: async (device, coordinatorEndpoint, logger) => {
        await extend.light_onoff_brightness().configure(device, coordinatorEndpoint, logger);
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff', 'genLevelCtrl']);
        await reporting.onOff(endpoint);
    },
};

module.exports = definition;
```

Can someone help me make the brightness working, thank you very much.

danieledwardgeorgehitchcock, 2021-09-01:
Not sure what this line is doing, but I don't think it should be there:

`await extend.light_onoff_brightness().configure(device, coordinatorEndpoint, logger);`

iamhyde, 2021-09-05:
I also have the same one, the dimming does not work, it can be switched on and off

asanchez156, 2021-09-06:
> Could you try this and see what happens?
> 
> ```js
> const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
> const tz = require('zigbee-herdsman-converters/converters/toZigbee');
> const exposes = require('zigbee-herdsman-converters/lib/exposes');
> const reporting = require('zigbee-herdsman-converters/lib/reporting');
> const extend = require('zigbee-herdsman-converters/lib/extend');
> const e = exposes.presets;
> const ea = exposes.access;
> 
> const definition = {
>     fingerprint: [{ modelID: 'TS110E', manufacturerName: '_TZ3210_ngqk6jia' }],
>     model: 'TS110E_dimmer',
>     vendor: 'TuYa',
>     description: 'Smart dimmer module with neutral',
>     extend: extend.light_onoff_brightness({noConfigure: true}),
>     exposes: [exposes.numeric('brightness', ea.ALL).withValueMin(10).withValueMax(1000).withDescription('Brightness of this light'));],
>     configure: async (device, coordinatorEndpoint, logger) => {
>         await extend.light_onoff_brightness().configure(device, coordinatorEndpoint, logger);
>         const endpoint = device.getEndpoint(1);
>         await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff', 'genLevelCtrl']);
>         await reporting.onOff(endpoint);
>     },
> };
> 
> module.exports = definition;
> ```
> 
> I think this may be stretching what can be extended from the original light_onoff_brightness converter so, you may need to create a bespoke device handler. Let's see how we get on!

Hi @danieledwardgeorgehitchcock!
I tried this but it just added a brightness sensor text, with no relation with the real light brightness. 

However, thank you very much for your help!

Anatol6868, 2021-09-14:
Got this device. Connected an external convecter. Brightness is not controlled, only on/off.

danieledwardgeorgehitchcock, 2021-09-14:
If you look at the above database.db entry, it has a `genLevelCtrl` cluster - this would lead me to believe that the brightness level can be controlled..?

Anatol6868, 2021-09-14:
The physical switch changes the brightness. I'll check in the evening if the data changes in database.db

Anatol6868, 2021-09-14:
"genLevelCtrl":{"attributes":{"61440":133,"64515":10,"64516":1000,"currentLevel":3}} 
Changes when pairing attribute 61440, depending on the brightness set by the physical switch

asanchez156, 2021-09-14:
> "genLevelCtrl":{"attributes":{"61440":133,"64515":10,"64516":1000,"currentLevel":3}}
> Changes when pairing attribute 61440, depending on the brightness set by the physical switch

Yes, that's correct, @Anatol6868. It is what I wrote in the issue's description.

Thank you for your help! :)

Anatol6868, 2021-09-14:
In the log when adjusting the brightness with a physical switch:

danieledwardgeorgehitchcock, 2021-09-15:
Could I ask you to try the below:

```javascript
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const utils = require('zigbee-herdsman-converters/lib/utils');
const e = exposes.presets;
const ea = exposes.access;

const fzLocal = {
    ts110e_brightness: {
        cluster: 'genLevelCtrl',
        type: ['attributeReport', 'readResponse'],
        convert: (model, msg, publish, options, meta) => {
            if (msg.data.hasOwnProperty('61440')) {
                return {brightness: mapNumberRange(msg.data['61440'], 10, 1000, 1, 100, 0)};
            }
        },
    },
};

const definition = {
    fingerprint: [{ modelID: 'TS110E', manufacturerName: '_TZ3210_ngqk6jia' }],
    model: 'TS110E_dimmer',
    vendor: 'TuYa',
    description: 'Smart dimmer module with neutral',
    fromZigbee: [fz.on_off, fzLocal.ts110e_brightness],
    toZigbee [tz.on_off],
    exposes: [exposes.numeric('brightness', ea.ALL).withValueMin(10).withValueMax(1000).withDescription('Brightness of this light'));],
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff', 'genLevelCtrl']);
        await reporting.onOff(endpoint);
    },
};

module.exports = definition;
```

It appears we will probably need some custom converters on this one which I have added part of above.

I haven't done the toZigbee replacement yet, so the devices brightness won't be controllable via Z2M.

However, if you could try changing the brightness via the device, and see if the slider moves correctly in Z2M, I know that we are on the right track!

asanchez156, 2021-09-15:
> Could I ask you to try the below:
> 
> ```js
> const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
> const tz = require('zigbee-herdsman-converters/converters/toZigbee');
> const exposes = require('zigbee-herdsman-converters/lib/exposes');
> const reporting = require('zigbee-herdsman-converters/lib/reporting');
> const utils = require('zigbee-herdsman-converters/lib/utils');
> const e = exposes.presets;
> const ea = exposes.access;
> 
> const fzLocal = {
>     ts110e_brightness: {
>         cluster: 'genLevelCtrl',
>         type: ['attributeReport', 'readResponse'],
>         convert: (model, msg, publish, options, meta) => {
>             if (msg.data.hasOwnProperty('61440')) {
>                 return {brightness: mapNumberRange(msg.data['61440'], 10, 1000, 1, 100, 0)};
>             }
>         },
>     },
> };
> 
> const definition = {
>     fingerprint: [{ modelID: 'TS110E', manufacturerName: '_TZ3210_ngqk6jia' }],
>     model: 'TS110E_dimmer',
>     vendor: 'TuYa',
>     description: 'Smart dimmer module with neutral',
>     fromZigbee: [fz.on_off, fzLocal.ts110e_brightness],
>     toZigbee [tz.on_off],
>     exposes: [exposes.numeric('brightness', ea.ALL).withValueMin(10).withValueMax(1000).withDescription('Brightness of this light'));],
>     configure: async (device, coordinatorEndpoint, logger) => {
>         const endpoint = device.getEndpoint(1);
>         await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff', 'genLevelCtrl']);
>         await reporting.onOff(endpoint);
>     },
> };
> 
> module.exports = definition;
> ```
> 
> It appears we will probably need some custom converters on this one which I have added part of above.
> 
> I haven't done the toZigbee replacement yet, so the devices brightness won't be controllable via Z2M.
> 
> However, if you could try changing the brightness via the device, and see if the slider moves correctly in Z2M, I know that we are on the right track!

Hello @danieledwardgeorgehitchcock, first of all thank you again!
I would try this evening and tell you back if at least the slider's brightness moves by changing the brightness on the device. :)

danieledwardgeorgehitchcock, 2021-09-15:
Please note, I modified my last comment just before you commented.

Make sure you use that one :-)

Anatol6868, 2021-09-15:
no start zigbee2mqtt, 

npm[18159]: (node:18198) UnhandledPromiseRejectionWarning: SyntaxError: Unexpected token '['
 npm[18159]: (node:18198) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch
 npm[18159]: (node:18198) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the

danieledwardgeorgehitchcock, 2021-09-15:
Try this:

```javascript
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const utils = require('zigbee-herdsman-converters/lib/utils');
const e = exposes.presets;
const ea = exposes.access;

const fzLocal = {
    ts110e_brightness: {
        cluster: 'genLevelCtrl',
        type: ['attributeReport', 'readResponse'],
        convert: (model, msg, publish, options, meta) => {
            if (msg.data.hasOwnProperty('61440')) {
                return {brightness: utils.mapNumberRange(msg.data['61440'], 10, 1000, 1, 100, 0)};
            }
        },
    },
};

const definition = {
    fingerprint: [{ modelID: 'TS110E', manufacturerName: '_TZ3210_ngqk6jia' }],
    model: 'TS110E_dimmer',
    vendor: 'TuYa',
    description: 'Smart dimmer module with neutral',
    fromZigbee: [fz.on_off, fzLocal.ts110e_brightness],
    toZigbee [tz.on_off],
    exposes: [exposes.numeric('brightness', ea.ALL).withValueMin(1).withValueMax(100).withDescription('Brightness of this light')],
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff', 'genLevelCtrl']);
        await reporting.onOff(endpoint);
    },
};

module.exports = definition;
```

Anatol6868, 2021-09-15:
no start, log
вер 15 19:00:47  systemd[1]: Started zigbee2mqtt.
вер 15 19:00:48  npm[7224]: > zigbee2mqtt@1.21.1 start
вер 15 19:00:48  npm[7224]: > node index.js
вер 15 19:00:50  npm[7224]: (node:7254) UnhandledPromiseRejectionWarning: SyntaxError: Unexpected token '['
вер 15 19:00:50  npm[7224]: (node:7254) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 1)
вер 15 19:00:50  npm[7224]: (node:7254) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.
вер 15 19:00:50  systemd[1]: zigbee2mqtt.service: Service hold-off time over, scheduling restart.
вер 15 19:00:50  systemd[1]: zigbee2mqtt.service: Scheduled restart job, restart counter is at 75.
вер 15 19:00:50  systemd[1]: Stopped zigbee2mqtt.

danieledwardgeorgehitchcock, 2021-09-15:
This?

```javascript
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const utils = require('zigbee-herdsman-converters/lib/utils');
const e = exposes.presets;
const ea = exposes.access;

const fzLocal = {
    ts110e_brightness: {
        cluster: 'genLevelCtrl',
        type: ['attributeReport', 'readResponse'],
        convert: (model, msg, publish, options, meta) => {
            if (msg.data.hasOwnProperty('61440')) {
                return {brightness: utils.mapNumberRange(msg.data['61440'], 10, 1000, 1, 100, 0)};
            }
        },
    },
};

const definition = {
    fingerprint: [{ modelID: 'TS110E', manufacturerName: '_TZ3210_ngqk6jia' }],
    model: 'TS110E_dimmer',
    vendor: 'TuYa',
    description: 'Smart dimmer module with neutral',
    fromZigbee: [fz.on_off, fzLocal.ts110e_brightness],
    toZigbee: [tz.on_off],
    exposes: [exposes.numeric('brightness', ea.ALL).withValueMin(1).withValueMax(100).withDescription('Brightness of this light')],
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff', 'genLevelCtrl']);
        await reporting.onOff(endpoint);
    },
};

module.exports = definition;
```

Anatol6868, 2021-09-15:
"brightness": 97,
    "last_seen": "2021-09-15T19:17:45+03:00",
    "state": "ON",
    "linkquality": 108

{
    "brightness": 58,
    "last_seen": "2021-09-15T19:18:40+03:00",
    "state": "ON",
    "linkquality": 114
}
The value changes, the slider moves, the 'OnOff' slider is missing. There is no reverse control,
zigbee2MQTT:debug 2021-09-15 19:27:23: Received MQTT message on 'zigbee2mqtt/0xa4c1388c242cd824/set' with data '{"brightness": 98}'
                       вер 15 19:27:23  npm[21494]: Zigbee2MQTT:error 2021-09-15 19:27:23: No converter available for 'brightness' (98)

Anatol6868, 2021-09-15:
exposes: [exposes.numeric('brightness', ea.ALL).withValueMin(1).withValueMax(100).withDescription('Brightness of this light'),e.switch()],
Added on and off control

danieledwardgeorgehitchcock, 2021-09-15:
Ok,

that’s as expected - I will code up the toZigbee for you guys to test tomorrow, unless one of you wants a go at it?

it is pretty much the brightness converter in toZigbee.js with the inverse of what I had done in fromZigbee above..?

Almost there :-)

Anatol6868, 2021-09-16:
The function works: 
fromZigbee: [tz.moes_power_on_behavior],
    toZigbee: [tz.moes_power_on_behavior],

Anatol6868, 2021-09-17:
If you write in toZigbee  :

 let dimmValue = Number(value);
 if (dimmValue > 100) dimmValue = 100;
 if (dimmValue < 0) dimmValue = 0;
 const bright = Math.round(dimmValue * 10);
 await entity.write('genLevelCtrl', {'61440': bright}, utils.getOptions(meta.mapped, entity));

Then it throws an error:
payload '{"message":"Publish 'set' 'brightness' to '0x000000000000000000000' failed: 'Error: Write 0xa000000000000000/1 genLevelCtrl({\"61440\":420}, {\"sendWhenActive\":false,\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":true,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (Write for 'BUFFER' not available)'","meta":{"friendly_name":"0x000000000000000000"},"type":"zigbee_publish_error"}'

Anatol6868, 2021-09-17:
or command :  await tuya.sendDataPointValue(entity, tuya.dataPoints.dimmerLevel, bright, 'setData', 1);
to
Zigbee2MQTT:info  2021-09-18 01:21:19: MQTT publish: topic 'zigbee2mqtt/bridge/log',
 payload '{"message":"Publish 'set' 'brightness' to '0x0000000000000'
 failed: 'TypeError: Cannot read property '61440' of undefined'","meta":{"friendly_name":"0x00000000000"},"type":"zigbee_publish_error"}'

Anatol6868, 2021-09-23:
> Ok,
> 
> that’s as expected - I will code up the toZigbee for you guys to test tomorrow, unless one of you wants a go at it?
> 
> it is pretty much the brightness converter in toZigbee.js with the inverse of what I had done in fromZigbee above..?
> 
> Almost there :-)

Help...

Anatol6868, 2021-10-21:
https://www.zigbee2mqtt.io/how_tos/how_to_find_tuya_data_points.html :

Tuya Data Points
Switch1 - 1
Brightness1 - 2
Minimum brightness1  - 3
Type of light source1 - 4
Maximum brightness1 - 5
Countdown1 - 6
Sleep mode - 22
Wakeup mode - 23
Mode - 13
Power-on Status Setting -14

Standard Instruction Set

Code | Type | Values
-- | -- | --
switch_led_1 | Boolean | "{true,false}"
sleep_mode | Raw | {}
wakeup_mode | Raw | {}
countdown_1 | Integer | {   "unit": "s",   "min": 0,   "max": 86400,   "scale": 0,   "step": 1 }
work_mode | Enum | {   "range": [     "light_white"   ] }
led_type_1 | Enum | {   "range": [     "led",     "incandescent",     "halogen"   ] }
bright_value_1 | Integer | {   "min": 10,   "max": 1000,   "scale": 0,   "step": 1 }
brightness_min_1 | Integer | {   "min": 10,   "max": 1000,   "scale": 0,   "step": 1 }
brightness_max_1 | Integer | {   "min": 10,   "max": 1000,   "scale": 0,   "step": 1 }
relay_status | Enum | {   "range": [     "off",     "on",     "memory"   ] }

Anatol6868, 2021-10-29:
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const utils = require('zigbee-herdsman-converters/lib/utils');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;

const DataType = {
    uint16: 0x21,
    uint32: 0x23,
    int16: 0x29,
    int32: 0x2B,
    enum8: 0x30
};

const toZigbee_OnOffSwitchCfg = {
   key: ['brightness', 'brightness_min', 'brightness_max', 'switch_level'],

   convertSet: async (entity, key, value, meta) => {
       let payload = {};
       let newValue = value;

       switch(key) {
           case 'brightness':
//               const newValue = utils.mapNumberRange(value, 1, 100, 10, 1000);
               let newValue = value;
               payload = {61440: {'value': newValue, 'type': DataType.uint16}};
               await entity.write('genLevelCtrl', payload, utils.getOptions(meta.mapped, entity));
               break;

           case 'brightness_min':
               payload = {64515: {'value': value, 'type': DataType.uint16}};
               await entity.write('genLevelCtrl', payload, utils.getOptions(meta.mapped, entity));
               break;

           case 'brightness_max':
               payload = {64516: {'value': value, 'type': DataType.uint16}};
               await entity.write('genLevelCtrl', payload, utils.getOptions(meta.mapped, entity));
               break;

           case 'switch_level':
               payload = {'currentLevel': {'value': value, 'type': DataType.uint16}};
               await entity.write('genLevelCtrl', payload, utils.getOptions(meta.mapped, entity));
               break;

           default:
               break;
       }

       result = {state: {[key]: value}};
       return result;
   },
   convertGet: async (entity, key, meta) => {
       if(key == 'switch_level') {
           await entity.read('genLevelCtrl', ['currentLevel']);
       }
       else {
           const lookup = {
               brightness: 61440,
               brightness_min: 64515,
               brightness_max: 64516
           };
           await entity.read('genLevelCtrl', [lookup[key]], utils.getOptions(meta.mapped, entity));
       }
   },
};

const fzLocal = {
    brightness: {
        cluster: 'genLevelCtrl',
        type: ['attributeReport', 'readResponse'],
        convert: (model, msg, publish, options, meta) => {
            if (msg.data.hasOwnProperty('61440')) {
//                return {brightness: utils.mapNumberRange(msg.data['61440'], 10, 1000, 1, 100, 0)};
                return {brightness: msg.data['61440']};
            }
        },
    },
};
const fzLoca = {
    brightness_min: {
        cluster: 'genLevelCtrl',
        type: ['attributeReport', 'readResponse'],
        convert: (model, msg, publish, options, meta) => {
            if (msg.data.hasOwnProperty('64515')) {
                return {brightness_min: msg.data['64515']};
            }
        },
    },
};
const fzLoc = {
    brightness_max: {
        cluster: 'genLevelCtrl',
        type: ['attributeReport', 'readResponse'],
        convert: (model, msg, publish, options, meta) => {
            if (msg.data.hasOwnProperty('64516')) {
                return {brightness_max: msg.data['64516']};
            }
        },
    },
};

const fzLc = {
    switch_level: {
        cluster: 'genLevelCtrl',
        type: ['attributeReport', 'readResponse'],
        convert: (model, msg, publish, options, meta) => {
            if (msg.data.hasOwnProperty('currentLevel')) {
                return {switch_level: msg.data['currentLevel']};
            }
        },
    },
};

const definition = {
    fingerprint: [{ modelID: 'TS110E', manufacturerName: '_TZ3210_ngqk6jia' }],
    model: 'TS110E_dimmer',
    vendor: 'TuYa',
    description: 'Smart dimmer module with neutral',
    fromZigbee: [fz.on_off, fzLocal.brightness, fzLoca.brightness_min, fzLoc.brightness_max, fz.moes_power_on_behavior,fzLc.switch_level],
    toZigbee: [tz.on_off, toZigbee_OnOffSwitchCfg, tz.moes_power_on_behavior],
    exposes: [exposes.numeric('brightness', ea.ALL).withValueMin(1).withValueMax(1000).withDescription('Brightness of this light'),
              e.switch(),
              exposes.enum('power_on_behavior', ea.ALL, ['off', 'previous', 'on']).withDescription('Controls the behaviour when the device is powered on'),
              exposes.numeric('brightness', ea.ALL),
              exposes.numeric('brightness_min', ea.ALL),
              exposes.numeric('brightness_max', ea.ALL),
              exposes.numeric('switch_level', ea.ALL)
              ],
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff', 'genLevelCtrl']);
        await reporting.onOff(endpoint);
    },

};

module.exports = definition;

**Command    await entity.write('genLevelCtrl', payload, utils.getOptions(meta.mapped, entity));** 
The value of the attribute 61440  changes when the command is sent, but in fact the brightness does not change!

Anatol6868, 2021-10-29:
Command await entity.write('genLevelCtrl', payload, utils.getOptions(meta.mapped, entity));
The value of the attribute 61440 changes when the command is sent, but in fact the brightness does not change!

[ts110e_new.txt](https://github.com/Koenkk/zigbee2mqtt/files/7445366/ts110e_new.txt)

Anatol6868, 2021-10-29:
Debug Received MQTT message on 'zigbee2mqtt/0xa4c1388c242cd824/set' with data '{"brightness":349}'
Info MQTT publish: topic 'zigbee2mqtt/0xa4c1388c242cd824', payload '{"brightness":829,"brightness_max":1000,"brightness_min":15,"last_seen":"2021-10-30T02:35:56+03:00","linkquality":96,"power_on_behavior":"off","state":"ON","switch_level":3}'
Info MQTT publish: topic 'zigbee2mqtt/0xa4c1388c242cd824', payload '{"brightness":349,"brightness_max":1000,"brightness_min":15,"last_seen":"2021-10-30T02:36:05+03:00","linkquality":111,"power_on_behavior":"off","state":"ON","switch_level":3}'

Anatol6868, 2021-10-29:
Debug Received MQTT message on 'zigbee2mqtt/0xa4c1388c550cd824/set' with data '{"brightness":349}'
Info MQTT publish: topic 'zigbee2mqtt/0xa4c1388c550cd824', payload '{"brightness":829,"brightness_max":1000,"brightness_min":15,"last_seen":"2021-10-30T02:35:56+03:00","linkquality":96,"power_on_behavior":"off","state":"ON","switch_level":3}'
Info MQTT publish: topic 'zigbee2mqtt/0xa4c1388c550cd824', payload '{"brightness":349,"brightness_max":1000,"brightness_min":15,"last_seen":"2021-10-30T02:36:05+03:00","linkquality":111,"power_on_behavior":"off","state":"ON","switch_level":3}'

Anatol6868, 2021-11-01:
genLevelCtrl  attributes 64514  uint8   0 - led , 1 - incandescent, 2- halogen

Anatol6868, 2021-11-26:
[image]

Anatol6868, 2021-11-26:
https://disk.yandex.by/d/nE7OD2tIewCOvA
запись , key 1f50ddb684c75fbe6258eb9f269398bf

Anatol6868, 2021-11-26:
In Tasmota, a low-level command causes the brightness to change:
 CMD: ZbSend {"device":"0xFB7B", "send": "0008!f0/f401"} 
 MQT: stat/tasmota_ZB/RESULT = {"ZbSend":"Done"}
 MQT: tele/tasmota_ZB/FB7B/SENSOR = {"Device":"0xFB7B","0008/F000":500,"Endpoint":1,"LinkQuality":154}
 MQT: tele/tasmota_ZB/FB7B/SENSOR = {"Device":"0xFB7B","0008/F000":498,"Dimmer":3,"Endpoint":1,"LinkQuality":154}
0008 cluster, f0 command,  0x01F4 - value 500

Anatol6868, 2021-11-27:
Configured the operation of this device. In the file .src/zcl/definition/cluster.ts you need to add a new command for the 0008 cluster:
.........
            moveToLevelTuya: {
                ID: 240,
                parameters: [
                    { name: 'level', type: dataType_1.default.uint16 },
                    { name: 'transtime', type: dataType_1.default.uint16 },
                ],
            },
...........

Works with external converter :

Anatol6868, 2021-11-27:
```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const utils = require('zigbee-herdsman-converters/lib/utils');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;

const DataType = {
    uint8: 0x20,
    uint16: 0x21,
};

const fzLocal = {
    ts110e_brightness: {
        cluster: 'genLevelCtrl',
        type: ['attributeReport', 'readResponse'],
        convert: (model, msg, publish, options, meta) => {
            if (msg.data.hasOwnProperty('61440')) {
 //               return {brightness: utils.mapNumberRange(msg.data['61440'], 10, 1000, 1, 100, 0)};
                return {brightness: msg.data['61440']};
            }
        },
    },
};

const fzLoca = {
    brightness_min: {
        cluster: 'genLevelCtrl',
        type: ['attributeReport', 'readResponse'],
        convert: (model, msg, publish, options, meta) => {
            if (msg.data.hasOwnProperty('64515')) {
                return {brightness_min: msg.data['64515']};
            }
        },
    },
};

const fzLoc = {
    brightness_max: {
        cluster: 'genLevelCtrl',
        type: ['attributeReport', 'readResponse'],
        convert: (model, msg, publish, options, meta) => {
            if (msg.data.hasOwnProperty('64516')) {
                return {brightness_max: msg.data['64516']};
            }
        },
    },
};

const fzled = {
     led_type: {
        cluster: 'genLevelCtrl',
        type: ['attributeReport', 'readResponse'],
        convert: (model, msg, publish, options, meta) => {
            const lookup1 = {0: 'led', 1: 'incandescent', 2: 'halogen'};
            if (msg.data.hasOwnProperty('64514')) {
                const property = utils.postfixWithEndpointName('led_type', msg, model);
                return {[property]: lookup1[msg.data['64514']]};
            }
        },
    },
};

const tzLocal = {
ts110e_brightness: {
   key: ['brightness'],
      convertSet: async (entity, key, value, meta) => {
               let level = value;
               const payload1 = {level, transtime: 0}
               await entity.command('genOnOff', 1, {}, utils.getOptions(meta.mapped, entity));
               await utils.sleep(1000)
               await entity.command('genLevelCtrl', 'moveToLevelTuya', payload1, utils.getOptions(meta.mapped, entity));

     },
     convertGet: async (entity, key, meta) => {
            await entity.read('genLevelCtrl', [61440]);
     },
   },
};

const tzLoca = {
ts110e_brightness_min: {
   key: ['brightness_min'],
      convertSet: async (entity, key, value, meta) => {
            let payload = {64515: {'value': value, 'type': DataType.uint16}};
            await entity.write('genLevelCtrl', payload, utils.getOptions(meta.mapped, entity));

     },
     convertGet: async (entity, key, meta) => {
           await entity.read('genLevelCtrl', [64515]);
     },
   },
};

const tzLoc = {
ts110e_brightness_max: {
   key: ['brightness_max'],
      convertSet: async (entity, key, value, meta) => {
            let payload = {64516: {'value': value, 'type': DataType.uint16}};
            await entity.write('genLevelCtrl', payload, utils.getOptions(meta.mapped, entity));

     },
     convertGet: async (entity, key, meta) => {
           await entity.read('genLevelCtrl', [64516]);
     },
   },
};

const tzLo = {
ts110e_led_type: {
   key: ['led_type'],
      convertSet: async (entity, key, value, meta) => {
           value = value.toLowerCase();
           const lookup1 = {'led': 0, 'incandescent': 1, 'halogen': 2};
           newValue = parseInt(lookup1[value]);
           payload = {'64514': {'value': newValue, 'type': DataType.uint8}};
           await entity.write('genLevelCtrl', payload, utils.getOptions(meta.mapped, entity));

     },
     convertGet: async (entity, key, meta) => {
           await entity.read('genLevelCtrl', [64514]);
     },
   },
};

const definition = {
    fingerprint: [{ modelID: 'TS110E', manufacturerName: '_TZ3210_ngqk6jia' }],
    model: 'TS110E_dimmer',
    vendor: 'TuYa',
    description: 'Smart dimmer module with neutral',
    fromZigbee: [fz.on_off, fzLocal.ts110e_brightness, fz.moes_power_on_behavior, fzLoca.brightness_min, fzLoc.brightness_max, fzled.led_type],
    toZigbee: [tz.on_off, tz.moes_power_on_behavior, tzLocal.ts110e_brightness,tzLoca.ts110e_brightness_min,tzLoc.ts110e_brightness_max, tzLo.ts110e_led_type],
    exposes: [exposes.numeric('brightness', ea.ALL).withValueMin(10).withValueMax(1000).withDescription('Brightness of this light'),
              e.switch(),
              exposes.numeric('brightness_min', ea.ALL),
              exposes.numeric('brightness_max', ea.ALL),
              exposes.enum('led_type', ea.ALL, ['led', 'incandescent', 'halogen']).withDescription('Controls the type led'),
              exposes.enum('power_on_behavior', ea.ALL, ['off', 'previous', 'on']).withDescription('Controls the behaviour when the device is powered on')],
        configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff', 'genLevelCtrl']);
        await reporting.onOff(endpoint);
   },
};

module.exports = definition;

                            
```

rsettembrini, 2021-11-28:
hi I am new to it and I have the same device. Could I do something to help you build a more or less final converter file? I don't know i.e. data point search and so on...

Anatol6868, 2021-11-29:
https://github.com/Koenkk/zigbee2mqtt/issues/9878

closed 2022-01-06 by github-actions[bot]

Gibbz, 2022-04-30:
Hi,

I have one of these devices on version 1.25 its still showing as not supported. 
How do you get this to work?