Koenkk/zigbee2mqtt #8580: Schneider Connected Socket 3025CSGZ

KPieboy, 2021-08-31:
Hello, 
I am attempting to support a new device for the first time...
it is Currently Not in the list of supported devices, it is a Schneider Connected Socket 3025CSGZ link below.

I have paired it successfully and created the .js file for the external converter file, 
I now receive in the log,

Zigbee2MQTT:debug 2021-08-31 14:30:33: Received Zigbee message from '0x680ae2fffe386ffd', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":1,"fileVersion":16843007,"imageType":62,"manufacturerCode":4190}' from endpoint 1 with groupID 0

Zigbee2MQTT:warn  2021-08-31 14:30:45: Received message from unsupported device with Zigbee model 'CH/Socket/2' and manufacturer name 'Schneider Electric'

My problem is i am unable to find a  converter (https://github.com/Koenkk/zigbee-herdsman-converters/blob/5c069a34beecc9250d642d02e84f2808af1b4fae/converters/fromZigbee.js#L304)
for this cluster 'genOta'

Any Ideas?

Many Thanks.

## Information about the device + link
..
https://download.schneider-electric.com/files?p_enDocType=Instruction+sheet&p_File_Name=NNZ6641201_Iconic_Connected_Socket_Twin_10A.pdf&p_Doc_Ref=NNZ664120

## data/database.db entry of the device
..

sjorge, 2021-08-31:
Try adding `ignore_genOta` to the fz list.

KPieboy, 2021-09-07:
Thanks sjorge, I have success!! my next thig is to work out where the end points are, currently i have endpoint 1 updating link quality.
the socket has two individual controllable outlets so i assume there at least 2 more endpoints

is there any simple way to work out what endpoints are available?
how do we usually find this info?

sjorge, 2021-09-07:
You need to enable multiendpoint and gave them a name (e.g. left and right or top and bottom) for an example look at the Ubisys S2 device.

KPieboy, 2021-09-08:
after my first go tinkering (config file as below), this is what i have, although when i attempt to send command from front end i get error 
Zigbee2MQTT:error 2021-09-08 21:58:22: Entity '0x680ae2fffe386ffd/Left' is unknown

What is the connection between the name Left and the endpoint 1 or 2? is it indexed by the code
endpoint: (device) => {
            return {'Left': 1, 'Right': 2};
         },

Actual file below

 const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['CH/Socket/2'], 
    model: 'Iconic 395', 
    vendor: 'Schneider Electric', 
    description: 'Connected 10a socket', 
    fromZigbee: [fz.ignore_genOta, fz.on_off], 
    toZigbee: [tz.ignore_genOta, tz.on_off, tz.command_toggle, tz.command_on, tz.command_off], 
    exposes: [ e.switch().withEndpoint('Left'), e.switch().withEndpoint('Right'),
             e.action(['toggle_s1', 'toggle_s2', 'on_s1', 'on_s2', 'off_s1', 'off_s2'])], 
   
    meta: {multiEndpoint: true},
    endpoint: (device) => {
            return {'Left': 1, 'Right': 2};
         },

       configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint1 = device.getEndpoint(1);
            await reporting.bind(endpoint1, coordinatorEndpoint, ['genOnOff']);
            await reporting.onOff(endpoint1);
            const endpoint2 = device.getEndpoint(2);
            await reporting.bind(endpoint2, coordinatorEndpoint, ['genOnOff']);
            await reporting.onOff(endpoint2);
        },
};

module.exports = definition;

sjorge, 2021-09-08:
Try lowercase L and R both the config and the mqtt command.

~ sjorge

KPieboy, 2021-09-09:
I tried just lower case l and r, same in the log so i changed them to 1 and 2 now i receive the below

Zigbee2MQTT:debug 2021-09-09 06:59:19: Received MQTT message on 'zigbee2mqtt/0x680ae2fffe386ffd/1/set' with data '{"state_1":"OFF"}'
Zigbee2MQTT:error 2021-09-09 06:59:19: Failed to call 'EntityPublish' 'onMQTTMessage' (TypeError: Cannot read property 'key' of undefined
    at converters.find (/opt/zigbee2mqtt/lib/extension/publish.js:191:56)
    at Array.find (<anonymous>)
    at EntityPublish.onMQTTMessage (/opt/zigbee2mqtt/lib/extension/publish.js:191:42)
    at Controller.callExtensionMethod (/opt/zigbee2mqtt/lib/controller.js:412:81)
    at process._tickCallback (internal/process/next_tick.js:68:7))

sjorge, 2021-09-09:
```node
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['CH/Socket/2'],
    model: 'Iconic 395',
    vendor: 'Schneider Electric',
    description: 'Connected 10a socket',
    fromZigbee: [fz.ignore_genOta, fz.on_off],
    toZigbee: [tz.ignore_genOta, tz.on_off,],
    exposes: [ e.switch().withEndpoint('l1'), e.switch().withEndpoint('l2')],
    meta: {multiEndpoint: true},
    endpoint: (device) => {
        return {'l1': 1, 'l2': 2};
     },

    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint1 = device.getEndpoint(1);
        await reporting.bind(endpoint1, coordinatorEndpoint, ['genOnOff']);
        await reporting.onOff(endpoint1);
        const endpoint2 = device.getEndpoint(2);
        await reporting.bind(endpoint2, coordinatorEndpoint, ['genOnOff']);
        await reporting.onOff(endpoint2);
    },
};

module.exports = definition;
```

Try like this. using l1 and l2 endpoints names

KPieboy, 2021-09-09:
OK Cool so when i first pair the socket i receive the correct state of the socket as expected below

Zigbee2MQTT:debug 2021-09-09 21:28:56: Received Zigbee message from '0x680ae2fffe386ffd', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2021-09-09 21:28:56: MQTT publish: topic 'zigbee2mqtt/0x680ae2fffe386ffd', payload '{"linkquality":120,"state_l1":"ON"}'
Zigbee2MQTT:debug 2021-09-09 21:28:56: Received Zigbee message from '0x680ae2fffe386ffd', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 2 with groupID 0
Zigbee2MQTT:info  2021-09-09 21:28:56: MQTT publish: topic 'zigbee2mqtt/0x680ae2fffe386ffd', payload '{"linkquality":118,"state_l1":"ON","state_l2":"ON"}'

But when i attempt to control it from the frontend i still receive this error

Zigbee2MQTT:error 2021-09-09 21:30:47: Failed to call 'EntityPublish' 'onMQTTMessage' (TypeError: Cannot read property 'key' of undefined
    at converters.find (/opt/zigbee2mqtt/lib/extension/publish.js:191:56)
    at Array.find (<anonymous>)
    at EntityPublish.onMQTTMessage (/opt/zigbee2mqtt/lib/extension/publish.js:191:42)
    at Controller.callExtensionMethod (/opt/zigbee2mqtt/lib/controller.js:412:81)
    at process._tickCallback (internal/process/next_tick.js:68:7))

sjorge, 2021-09-10:
I updated the one in https://github.com/Koenkk/zigbee2mqtt/issues/8580#issuecomment-915808662 plz try again, there is no ignore_genOta for toZigbee so I think it was failing on that.

KPieboy, 2021-09-14:
Awesome all is well now, many thanks