Koenkk/zigbee2mqtt #8581: Zigbee2MQTT:error Publish 'set' 'state' to ... failed: 'Error: Command 0x00158d0004218ea1/2 genOnOff.on({},

dt54171, 2021-08-31:
## What happened
all Zigbee devices stop working after unpredictable time. I use the latest version of Zigbee2mqtt addon and zStack3x0 20200921. All these running in the latest version of HA. In zigbee2mqtt addon Map menu it shows that there in no any connections at all.
Zigbee2mqtt addon reports the following error:
Zigbee2MQTT:error 2021-08-30 22:45:41: Publish 'set' 'state' to 'switch_hall' failed: 'Error: Command 0x00158d0004218ea1/2 genOnOff.on({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (SRSP - ZDO - extRouteDisc after 6000ms)'
After zigbee2mqtt addon restart in connectivity restores and all devises start working well.  zigbee2mqtt addon Map shows correct connectivity map.
## What did you expect to happen
no outages 
## How to reproduce it (minimal and precise)

## Debug info
Zigbee2MQTT version: 1.18.1-1
Adapter hardware: zStack3x0 
Adapter firmware version: 20200921

danieledwardgeorgehitchcock, 2021-08-31:
What Zigbee coordinator device are you using? How many devices are on your network?

dt54171, 2021-08-31:
usb stick TI CC2652P. Overall there are 12 devices. 4 of them are QBKG12LM, that work as a Router and they have 1 device connected to each of them.

Koenkk, 2021-08-31:
Since you are using outdated z2m and coordinator firmware version, I would suggest to first make sure you are using the latest ones.

If you are still on 1.18.1 check first point of https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.19.0

dt54171, 2021-08-31:
> Since you are using outdated z2m and coordinator firmware version, I would suggest to first make sure you are using the latest ones.
> 
> If you are still on 1.18.1 check first point of https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.19.0

Thank you! Updated zigbee2mqtt upto the latest version - 1.21.0-4, USB stick upto 20210708. So far so good. Will monitor it for a while.

dt54171, 2021-09-06:
I've monitored zigbee2mqtt for about a week and can conclude that similar issue occurred about once a day the following error occurred:
  Zigbee2MQTT:error 2021-09-06 14:26:11: Publish 'set' 'state' to 'switch_hall' failed: 'Error: Command 0x00158d0004218ea1/2 genOnOff.on({}, {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (SRSP - ZDO - extRouteDisc after 6000ms)'

After this error occurred all Zigbee connections are broken and Map in Zigbee2mqtt UI doesn't show any connections. 
After igbee2MQTT plugin restart everything works fine...

Koenkk, 2021-09-06:
Could you provide the herdsman debug logging of this issue?

See https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging on how to enable the herdsman debug logging. Note that this is only logged to STDOUT and not to log files.

dt54171, 2021-09-07:
log file attached
[zigbee-herdsman.txt](https://github.com/Koenkk/zigbee2mqtt/files/7117420/zigbee-herdsman.txt)

Koenkk, 2021-09-07:
What linux kernel are you using? (execute `uname -r`)

dt54171, 2021-09-07:
> What linux kernel are you using? (execute `uname -r`)

5.10.52-v7l+

Koenkk, 2021-09-08:
It seems something goes wrong with the communication of the USB stick.

- What system are you running z2m on? And are there more USB devices connected?
- Could you provide the log when running with `DEBUG=* npm start` (instead of `DEBUG=zigbee-herdsman* npm start`)
- What is the output of `dmesg` at the time of the crash?

dt54171, 2021-09-10:
it is Raspberry PI 4, 2Gb model
I have zigbee stick and APC UPS connected to USB2.0
SSD connected to USB3
[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/7143389/log.txt)
[dmesg.txt](https://github.com/Koenkk/zigbee2mqtt/files/7143391/dmesg.txt)

Koenkk, 2021-09-10:
Is the SSD connected via a powered hub? There have been many issues before with this (seems to draw too much power). Also make sure your stick is connected through an usb extension cable and positioned away from the SSD since this generated interference.

dt54171, 2021-09-13:
no, SSD connected using X857 Extension Board. 
I moved USB zigbee stick as far as possible using usb extension cable and now I have this error less offen. 
Actually the main problem is that when error appeared it blocks all zigbee devised. Watchdog enabled for the plugin doesn't help. Only manual plugin restart helps to restore connectivity.

dt54171, 2021-09-14:
+power adapter was changed as well
The interesting thing that now now any new zigbee devices are discovered in permit join mode...

gsasha, 2021-09-19:
I have a very similar problem. I'm leaving the state as is, so that if there are any diagnostics I can collect, it's still there.

Let me know if there is any information I can collect that will be useful.

I now have, out of my network, one 4-gang switch that is reporting the "No network route (205)" error. The rest seems to be fine.
The switch itself is connected to the network - the messages from it apparently arrive: if I flip the physical button, I see the state update in zigbee2mqtt, it's only the messages from coordinator to the switch that don't have the route.

I use 1.21.1 from the official docker on Raspberry Pi with stock OS. 

I have a CC2652P from a somewhat noname source in Aliexpress, but it seems otherwise to be working.

I tried restarting zigbee2mqtt, I did not try to restart the Raspi yet, and I'm holding off re-pairing of the switch.

The status page shows:

Zigbee2MQTT version 1.21.1 commit: eb5750d
Coordinator type zStack3x0
Coordinator revision 20210708
Frontend version 0.6.0

Tonio16, 2021-09-20:
Heĺlo

I have also this error, or I think so:
`Publish 'get' 'state' to '0x086bd7fffe0134d3' failed: 'Error: Read 0x086bd7fffe0134d3/1 genOnOff(["onOff"], {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 28440 - 1 - 233 - 6 - 1 after 10000ms)'`

I am running z2m with jeedom and zigbeelinker (= I don’t know how to have herdsman logs) on a pi3+ and a zigate  v1 (firmware 3.1d).

`Version Zigbee2MQTT
1.21.1 commit: 4a51e0c0
Type de Coordinateur
zigate
Révision du Coordinateur
1d03
Version Interface
0.6.0`

Antoine

goliath888, 2021-09-26:
Same here.
Just upgrade from CC2530 to a Zigbee Star v4 (runnin 1.21.0) with no issue. Sensor devices works fine (they report their status), but I cannot control lights and switches, it gives this error:

`Publish 'set' 'state' to 'hallway_front' failed: 'Error: Command 0x588e81fffec9ca45/1 genOnOff.on({}, {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 23164 - 1 - 5 - 6 - 11 after 10000ms)'`

Tried with container version 1.21.0 and 1.20.0
Upgraded the stick with the latest coordinator firmware  CC1352P2_CC2652P_launchpad_coordinator_20210708

closed 2021-11-03 by github-actions[bot]