Koenkk/zigbee2mqtt #8582: Cover device backlight issue

jvsystelab, 2021-08-31:
Good morning,

I have this device (it's a Tuya TS130F white label)

https://www.loratap.com/products/tuya-smart-life-zigbee-30-curtain-blind-switch-for-roller-shutter-electric-motor-google-home-alexa-echo-voice-control-diy-1

All entities works fine less the backlight entity.
I have devices with backlight on and anothers off. I don't know the reason.

The problem is:

When I change the entity value (from HA or Zigbee2mqtt frontend):

select.0x00000000000000_backlight_mode { LOW, MEDIUM, HIGH }

Nothing happens. The devices backlight not change the mode.

The log file has this:

Zigbee2MQTT:info  2021-08-31 12:03:58: MQTT publish: topic 'zigbee2mqtt/0x00000000000000/mode', payload '{"backlight":"MEDIUM"}'

Without errors but without changes.

## Debug info
Zigbee2MQTT version: 1.21
Adapter hardware: CC2652R
Adapter firmware version: 20210708

Thanks

yonaesp, 2021-08-31:
Same problem here with the same model but different design. 
https://www.loratap.com/products/tuya-smart-life-zigbee-30-curtain-blind-switch-for-roller-shutter-electric-motor-google-home-alexa-echo-voice-control-diy
## Debug info
Zigbee2MQTT version: 1.21.0
Adapter hardware: Conbee II
Adapter firmware version: deCONZ_RaspBeeII_0x26720700

yonaesp, 2021-10-04:
the problem persists

closed 2021-10-12 by github-actions[bot]

Pulpyyyy, 2023-05-22:
Issue still exists with my bseed device (_TZ3000_wptayaqr)

UniCizin, 2023-05-22:
Same issue here

Zigbee Modell
TS130F

Zigbee Hersteller
_TZ3000_wptayaqr

Pulpyyyy, 2023-09-22:
Just an update.
Has anyone found a fix or workaround?

Trebor87, 2023-11-28:
Months later and same problem. Back-light doesn't change regardless of the settings. Low, Medium. High
BSEED Zigbee curtain switch.

Danixu, 2024-05-18:
Same here with a LoraTap device.

Trebor87, 2024-11-26:
Thanks for the info 

The SC400ZB-EU-V2 only seems to be made by Loratap ok n AliExpress, and are only black or white, mine ar all gold coloured, so that's a pain, hopefully the ZigBee Software will have an update that controls yhe backlight one daye, though it seems unlikely

florianendrich, 2024-11-26:
If the hardware does not support it, i dont think any update will add this feature. 
The zigbee device junge is really annoying - and ordering on ali is a lottery on top as the pictures and product informations are not reliable.