Koenkk/zigbee2mqtt #8587: Unable to set up reporting for Xiaomi WXKG01LM button

antti365, 2021-08-31:
@Koenkk 

## What happened
Hi, after updating the stick firmware and updating the z2m to the latest dev version, I decided to configure the entire network from scratch, after adding the buttons, Xiaomi WXKG01LM noticed that the buttons work, but do not send reports, as before in version 1.18, for example. When I decided to set up reports manually, I found that there was practically nothing in the button settings on the "Clusters" tab ... something broke in the converters? I tried to re-pair the buttons several times - nothing changed ...

[image]

## Debug info
Zigbee2MQTT version - 1.21.0-dev commit: 05b19117
Coordinator type - zStack30x (2538)
Coordinator revision - 20201010

plaes, 2021-08-31:
@antti365 Can you check whether it similar to #8552 ?

antti365, 2021-08-31:
@plaes No, it doesn't look like, the buttons work without delays and the devices controlled by them work almost immediately (the button expects a double, triple click), I have a problem with missing clusters and their attributes and the inability to customize reports ... Although if I remember correctly, reports worked before without any additional settings ... What about the reports on these buttons?

plaes, 2021-08-31:
> @plaes No, it doesn't look like, the buttons work without delays and the devices controlled by them work almost immediately (the button expects a double, triple click), I have a problem with missing clusters and their attributes and the inability to customize reports ... Although if I remember correctly, reports worked before without any additional settings ... What do you have with the reports on these buttons?

Look similar as in your original post - only single `Endpoint 1` child...

closed 2021-10-08 by github-actions[bot]