Koenkk/zigbee2mqtt #8599: [FUT105Z] Help with adding new device Miboxer FUT105Z RGB+CCT Bulb

fluffymadness, 2021-09-01:
Bought the following bulb and now I'm trying to get it to work.
https://www.futlight.com/light/xqy/fut105z.html
This is what I have so far, but there are some issues:

```
{
        fingerprint: [{modelID: 'TS0505B', manufacturerName: '_TZ3210_wslkvrau'}],
        model: 'FUT105Z',
        description: '12W RGB+CCT LED Bulb',
        vendor: 'Miboxer',
        extend: extend.light_onoff_brightness_colortemp_color({disableColorTempStartup: true, colorTempRange: [153, 500]}),
		meta: {applyRedFix: true, enhancedHue: false},
 },
```
**Issues:**

Colors are all over the place. When red is set, the color is pink instead. 
Same happens to other colors. (I tried without the applyRedFix, enhancedHue line, didn't make any difference)

Brightness: zigbe2mqtt shows me a brightness slider that goes from 0-254. However brightness
stops changing after I think 127 (don't now the exact value, but this seems to be the spot, after trying different values)

What works: Changing the CCT Color Temperature.

Effects work all the same, doesn't matter when choosing blink or another. It just shortly dimms down and dimms up again.

Does someone have any hints what i could to to get this one working ?
Thanks

fluffymadness, 2021-09-01:
Brightness controls for color have the correct range from 0-254
only cct range seems to be halved.
However color mode only has half he brightness.
 It seems this device needs some extra work

I tried to add the following 
```
fromZigbee: [
        fz.ignore_basic_report, // Add this if you are getting no converter for 'genBasic'
        fz.tuya_data_point_dump, // This is a debug converter, it will be described in the next part
    ],
    toZigbee: [
        tz.tuya_data_point_test, // Another debug converter
    ],
```
in the hope to get some debug files. (Using zigbee2mqtt via homeassistant)
I connected to the docker container,but there is nothing in /data/
accessing the zigbee2mqtt directory via samba share. Also no  log files :(
Does anyone know where to find the logs.

fluffymadness, 2021-09-03:
Ok, color controls via homeassistant work fine, however are broken via webinterface.
If i set red via color xy in zigbee2mqtt interface it is pink, with homeassistant however it works fine

fluffymadness, 2021-10-10:
unstale

Koenkk, 2021-10-11:
I see, but that is not something we can be fixed in z2m (should be fixed by the manufacturer)

closed 2021-11-19 by github-actions[bot]