Koenkk/zigbee2mqtt #8600: Phillips Hue bulb firmware update 1.88.1 or 1.90.1 not available via OTA

Fisico, 2021-09-01:
## What happened
Most recent Phillips Hue bulbs firmwareupdates 1.88.1 or 1.90.1 are not available over MQTT OTA.

## What did you expect to happen
Upgrade Phillips Hue Filament e29 lights to firmware 1.88.1 or 1.90.1. MQTT says no update available and lamps stay on version 1.76.10.

The official changelog does not list this update. It is however available to upgrade via bridge and enables support for hue dynamic scenes etc... Various blogs, screenshots and Reddit posts confirmed that this update was released on august 14th 2021.
https://hueblog.com/2021/08/14/firmware-update-prepares-hue-lamps-for-dynamic-scenes/
https://www.reddit.com/r/Hue/comments/p8bpbm/hue_lights_updated_today/
[image]

## How to reproduce it (minimal and precise)
Check for updates on any Phillips Hue Bulbs 

## Debug info
Zigbee2MQTT version: 1.21.0 commit: 70891ee

MattWestb, 2021-09-01:
The OTAs is on hold waiting OK from the devs for 3 weeks now https://github.com/Koenkk/zigbee-OTA/pull/47.

ZHA have roiling the support for the new name change then its braking some function in the system and adding power on for on/off, light level and color for all X-Philips lights and updated to Signify for all devices that supporting this functions.

TheJulianJES, 2021-09-04:
It should be available now

Fisico, 2021-09-05:
Yes just updated my bulbs. Thank you

closed 2021-09-05 by Koenkk

odospace, 2022-11-25:
@TheJulianJES Can you tell me where to download the 1.88.1 firmware files?

TheJulianJES, 2022-11-26:
@odospace What is the "image type/id" of the bulb you need the older firmware for?

TheJulianJES, 2022-11-28:
@odospace This is the intermediate version for ``100-011D``:
http://fds.dc1.philips.com/firmware/ZGB_100B_011D/16785152/100B-011D-01001F00-ConfLight-ModuLumV2-EFR32MG13.zigbee

Do you also need this because Z2M is unable to directly upgrade your bulb to the latest version?
If so, please let me know if this works, so we can add it to the public ``zigbee-ota`` repo.

odospace, 2022-11-29:
Hi,  thank you. I want to install an older version. I tried with the following OTA overwrite file:[        {            "manufacturerCode": 4107,            "imageType": 285,            "url": "http://fds.dc1.philips.com/firmware/ZGB_100B_011D/16785152/100B-011D-01001F00-ConfLight-ModuLumV2-EFR32MG13.zigbee",            "force": true        }] But Z2M shows the following error: „File version mismatch“, any ideas?   Best regards Von: TheJulianJESGesendet: Montag, 28. November 2022 19:05An: Koenkk/zigbee2mqttCc: odospace; MentionBetreff: Re: [Koenkk/zigbee2mqtt] Phillips Hue bulb firmware update 1.88.1 or 1.90.1 not available via OTA ***@***.*** This is the intermediate version for 100-011D:http://fds.dc1.philips.com/firmware/ZGB_100B_011D/16785152/100B-011D-01001F00-ConfLight-ModuLumV2-EFR32MG13.zigbeeDo you also need this because Z2M is unable to directly upgrade your bulb to the latest version?If so, please let me know if this works, so we can add it to the public zigbee-ota repo.

TheJulianJES, 2022-11-29:
Not sure what exactly Z2M means with "File version mismatch", but do you maybe need to provide ``"fileVersion": 16785152`` too (in the override)?

odospace, 2022-11-29:
Hi,  this helps. However, it states Timeout: device did not request any image blocks.  I will give up. But thank you again for your time. Best regards Von: TheJulianJESGesendet: Dienstag, 29. November 2022 20:16An: Koenkk/zigbee2mqttCc: odospace; MentionBetreff: Re: [Koenkk/zigbee2mqtt] Phillips Hue bulb firmware update 1.88.1 or 1.90.1 not available via OTA (#8600) Not sure what exactly Z2M means with "File version mismatch", but do you maybe need to provide "fileVersion": 16785152 too (in the override)?

TheJulianJES, 2022-11-29:
On what firmware is the device currently?
Also, please respond to the issue via the GitHub app [or the web](https://github.com/Koenkk/zigbee2mqtt/issues/8600), as your e-mail responses are hard to read.

odospace, 2022-11-29:
sorry, the are really hard to read....

The current version is 1.93.11

TheJulianJES, 2022-11-29:
Oh, so why are you trying to downgrade?

odospace, 2022-12-05:
@TheJulianJES 
I found an additional hardware version, can you provide me the version 1.88.1 for this too?

Model
1746547P7
Hardware
100b-**111**

TheJulianJES, 2022-12-05:
From newest to oldest:
(not sure which firmware corresponds to 1.88.1)

http://fds.dc1.philips.com/firmware/ZGB_100B_0111/16784640/100B-0111-01001D00-ConfLight-ModuLum-EFR32MG13.zigbee (version 16784640)
http://fds.dc1.philips.com/firmware/ZGB_100B_0111/16783872/100B-0111-01001A00-ConfLight-ModuLum-EFR32MG13.zigbee (version 16783872)
http://fds.dc1.philips.com/firmware/ZGB_100B_0111/16783360/100B-0111-01001800-ConfLight-ModuLum-EFR32MG13.zigbee (version 16783360)
http://fds.dc1.philips.com/firmware/ZGB_100B_0111/16782848/100B-0111-01001600-ConfLight-ModuLum-EFR32MG13.zigbee (version 16782848)

odospace, 2022-12-15:
Thank you! Version 16783872 works.

brechtvhb, 2022-12-17:
Is it possible to use dynamic scenes when using zigbee2mqtt or is this only possible with the hue hub?