Koenkk/zigbee2mqtt #8602: Support issue with "ENKI LEXMAN E14 LED RGBW" (9CZA-P470T-A1A)

LittleBigDev, 2021-09-02:
## Information about the device + link

The device is **supposed to be already supported** (https://www.zigbee2mqtt.io/devices/9CZA-P470T-A1A.html)
Also found it in [adeo.js](https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices/adeo.js) devices file :
```
   {
        zigbeeModel: ['LXEK-3'],
        model: '9CZA-P470T-A1A',
        vendor: 'ADEO',
        description: 'ENKI LEXMAN E14 LED RGBW',
        extend: extend.light_onoff_brightness_colortemp_color({colorTempRange: [153, 370]}),
    }
```

## Db entry
```
{
   "id":6,
   "type":"Router",
   "ieeeAddr":"0x847127fffe981d27",
   "nwkAddr":18120,
   "manufId":4727,
   "manufName":"Adeo",
   "powerSource":"Mains (single phase)",
   "modelId":"LXEK-3",
   "epList":[
      1,
      242
   ],
   "endpoints":{
      "1":{
         "profId":260,
         "epId":1,
         "devId":269,
         "inClusterList":[
            0,
            3,
            4,
            5,
            6,
            8,
            768,
            4096,
            64642
         ],
         "outClusterList":[
            25
         ],
         "clusters":{
            "genBasic":{
               "attributes":{
                  "modelId":"LXEK-3",
                  "manufacturerName":"Adeo",
                  "powerSource":1,
                  "zclVersion":3,
                  "appVersion":1,
                  "stackVersion":2,
                  "hwVersion":1,
                  "dateCode":"20200526",
                  "swBuildId":"2.4.0"
               }
            }
         },
         "binds":[
            
         ],
         "configuredReportings":[
            
         ],
         "meta":{
            
         }
      },
      "242":{
         "profId":41440,
         "epId":242,
         "devId":97,
         "inClusterList":[
            
         ],
         "outClusterList":[
            33
         ],
         "clusters":{
            
         },
         "binds":[
            
         ],
         "configuredReportings":[
            
         ],
         "meta":{
            
         }
      }
   },
   "appVersion":1,
   "stackVersion":2,
   "hwVersion":1,
   "dateCode":"20200526",
   "swBuildId":"2.4.0",
   "zclVersion":3,
   "interviewCompleted":true,
   "meta":{
      
   },
   "lastSeen":1630562500495
}
```

## More info 

Here is the [vendor link](https://www.leroymerlin.fr/produits/decoration-eclairage/ampoule-et-led/ampoule-led/ampoule-e14/ampoule-connectee-led-spherique-e14-intensite-couleur-variables-enki-lexman-82473186.html).

Problem : marked as unsupported in HassOs, zigbee2mqtt dashboard :
[image]

Please tell me if I can provide anything more to help you fix this !

danieledwardgeorgehitchcock, 2021-09-02:
It looks like support for this device was only added 12 days ago.

Please could you check your version of Zigbee2MQTT? You should be running 1.21.1 which was released yesterday

LittleBigDev, 2021-09-02:
Hello @danieledwardgeorgehitchcock,

Thank you for the quick answer.

I'm currently running 1.18.1-1. Needs to be updated.

Auto update is enabled for the add-on, but it didn't update. I tried to go on Supervisor's add-ons store and "reload" to trigger the auto-update, but still no update incoming. I'll wait a little (end of my work time) and try again.  

I also tried removing and re-adding the z2m repo, and tried triggering updates again. Still nothing.  
I have a hunch that link between the add-on and the repo was lost/broken (the add-on icon changed from the z2m icon to the generic green one). I might need to reinstall it.

I know this is outside the scope of Koenkk/zigbee2mqtt, but if you are used to HA and know the traps and tricks of reinstalling an add-on without losing the config and paired devices, then any advice is welcome (I'm a Noob. Installed HA 3 days ago).

Thanks by advance !

danieledwardgeorgehitchcock, 2021-09-02:
You could also try switching to the Dev branch of Herdsman Converters which would keep you up-to-date with any new devices at the sacrifice of the software being potentially unstable (in all of the time I have used it, it has never fallen over). You can do that by following the how-to [here](https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html) - there is a specific section on how to do this in Home Assistant.

LittleBigDev, 2021-09-02:
My installed version of HA is the full image (HA OS including HA and supervisor). I didn't explore everything yet, but it looks like everything is dockerised in there and I don't really have control on it.
I installed the SSH terminal add on and access seems limited : no docker, docker-compose not systemctl commands, limited access to files : 
```
~ $ ls -la
total 12
drwxr-xr-x    1 root     root          4096 Sep  1 15:49 .
drwxr-xr-x    1 root     root          4096 Sep  1 15:49 ..
lrwxrwxrwx    1 root     root            19 May 26 15:23 .bash_history -> /data/.bash_history
lrwxrwxrwx    1 root     root            19 May 26 15:23 .bash_profile -> /data/.bash_profile
lrwxrwxrwx    1 root     root            10 May 26 15:23 .ssh -> /data/.ssh
-rw-r--r--    1 root     root           905 May 26 15:23 .tmux.conf
lrwxrwxrwx    1 root     root             7 Sep  1 15:49 addons -> /addons
lrwxrwxrwx    1 root     root             7 Sep  1 15:49 backup -> /backup
lrwxrwxrwx    1 root     root             7 Sep  1 15:49 config -> /config
lrwxrwxrwx    1 root     root             6 Sep  1 15:49 share -> /share
lrwxrwxrwx    1 root     root             4 Sep  1 15:49 ssl -> /ssl

```
Looks like ssh connection leads directly into a docker container... I still need to explore.

Anyway I'm diverging from the base subject : my add-on is out of date and the easiest solution seems to be a full reinstall. I'll give some time to auto-update to trigger, and if it doesn't, I'll have to restart from scratch I guess :(

LittleBigDev, 2021-09-02:
Ok, I couldn't wait.  
Backuped z2m add-on config, uninstalled and reinstalled the module. Got my config back from backup, rebooted MQTT broker add-on and started z2m add-on. 

Works like a charm, didn't loose my paired devices. Happy me.

Credits to @danieledwardgeorgehitchcock who pointed me to the very very recent device support add + new add-on version.

closed 2021-09-02 by LittleBigDev