Koenkk/zigbee2mqtt #8604: Home Assistant complaining that last_reset_topic and last_reset are deprecated

derekoharrow, 2021-09-02:
Since Home Assistant 2021.9.0 I am getting these errors in the Home Assistant log file:

```
2021-09-02 10:15:42 WARNING (SyncWorker_4) [homeassistant.components.google] The 'max_results' option is deprecated, please remove it from your configuration
2021-09-02 10:15:42 WARNING (SyncWorker_4) [homeassistant.components.google] The 'max_results' option is deprecated, please remove it from your configuration
2021-09-02 10:15:42 WARNING (SyncWorker_4) [homeassistant.components.google] The 'max_results' option is deprecated, please remove it from your configuration
2021-09-02 10:15:42 WARNING (SyncWorker_4) [homeassistant.components.google] The 'max_results' option is deprecated, please remove it from your configuration
2021-09-02 10:15:42 WARNING (SyncWorker_4) [homeassistant.components.google] The 'max_results' option is deprecated, please remove it from your configuration
2021-09-02 10:16:50 WARNING (MainThread) [homeassistant.components.mqtt.sensor] The 'last_reset_topic' option is deprecated, please remove it from your configuration
2021-09-02 10:16:50 WARNING (MainThread) [homeassistant.components.mqtt.sensor] The 'last_reset_topic' option is deprecated, please remove it from your configuration
2021-09-02 10:16:51 WARNING (MainThread) [homeassistant.components.mqtt.sensor] The 'last_reset_topic' option is deprecated, please remove it from your configuration
2021-09-02 10:16:51 WARNING (MainThread) [homeassistant.components.mqtt.sensor] The 'last_reset_topic' option is deprecated, please remove it from your configuration
2021-09-02 10:18:03 WARNING (MainThread) [homeassistant.components.sensor] Entity sensor.washing_machine_energy (<class 'homeassistant.components.mqtt.sensor.MqttSensor'>) with state_class measurement has set last_reset. Setting last_reset is deprecated and will be unsupported from Home Assistant Core 2021.11. Please update your configuration if state_class is manually configured, otherwise create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+mqtt%22
2021-09-02 10:19:40 WARNING (MainThread) [homeassistant.components.sensor] Entity sensor.tumble_dryer_energy (<class 'homeassistant.components.mqtt.sensor.MqttSensor'>) with state_class measurement has set last_reset. Setting last_reset is deprecated and will be unsupported from Home Assistant Core 2021.11. Please update your configuration if state_class is manually configured, otherwise create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+mqtt%22
2021-09-02 10:21:59 WARNING (MainThread) [homeassistant.components.sensor] Entity sensor.dishwasher_energy (<class 'homeassistant.components.mqtt.sensor.MqttSensor'>) with state_class measurement has set last_reset. Setting last_reset is deprecated and will be unsupported from Home Assistant Core 2021.11. Please update your configuration if state_class is manually configured, otherwise create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+mqtt%22
2021-09-02 10:22:29 WARNING (MainThread) [homeassistant.components.sensor] Entity sensor.cabin_heater_energy (<class 'homeassistant.components.mqtt.sensor.MqttSensor'>) with state_class measurement has set last_reset. Setting last_reset is deprecated and will be unsupported from Home Assistant Core 2021.11. Please update your configuration if state_class is manually configured, otherwise create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+mqtt%22
```

It appears that these errors all relate to my 4 power-monitoring smart plugs, which Z2M is populating in MQTT Home Assistant auto-discovery with config similar to the following:

**Topic:** homeassistant/sensor/0x001e5e0902145d6d/energy/config
`{"availability":[{"topic":"zigbee2mqtt/bridge/state"},{"topic":"zigbee2mqtt/Dishwasher/availability"}],"device":{"identifiers":["zigbee2mqtt_0x001e5e0902145d6d"],"manufacturer":"Salus Controls","model":"Smart plug (UK socket) (SP600)","name":"Dishwasher","sw_version":"Zigbee2MQTT 1.21.1-dev"},"device_class":"energy","json_attributes_topic":"zigbee2mqtt/Dishwasher","last_reset_topic":"zigbee2mqtt/Dishwasher","last_reset_value_template":"1970-01-01T00:00:00+00:00","name":"Dishwasher energy","state_class":"measurement","state_topic":"zigbee2mqtt/Dishwasher","unique_id":"0x001e5e0902145d6d_energy_zigbee2mqtt","unit_of_measurement":"kWh","value_template":"{{ value_json.energy }}"}`

From the Home Assistant Community forum, it seems I am not alone in experiencing this.

derekoharrow, 2021-09-02:
I've logged a corresponding Home Assistant Core issue [here](https://github.com/home-assistant/core/issues/55574)

Koenkk, 2021-09-02:
@frenck it's unclear for me what should change at the z2m side now, should `last_reset_topic` just be removed? It's still mentioned here: https://www.home-assistant.io/integrations/sensor.mqtt/

ssiergl, 2021-09-02:
@Koenkk see here

https://github.com/home-assistant/core/pull/55463

McGiverGim, 2021-09-03:
In this case, I think it is related with this: https://github.com/Koenkk/zigbee2mqtt/issues/8453 so I suppose you simply need to remove it, when moving from `state_class: measurement` to `state_class: total_increasing`.
I think they have left it in prevision of meters that can be negative, in this case they need a way to know when the meter has been reset. But I don't know if there is some device in zigbee2mqtt that can take negative measures of energy?

McGiverGim, 2021-09-03:
I've been looking at the Home Assistant PRs related with this:
- First, we had the `measurement` with `last_reset`
- They added the `total_increasing` to simplify this use case and removed the `last_reset` attribute
- They added a "new" `total` with `last_reset` to take into account the meters that can go negative.
- Finally they decided to remove this new `total` option.

I think:
- Only `measurement` and `total_increasing` are valid.
- The `last_reset` is accepted until `2021.11.0`, for this reason it continues giving warnings until removed.

So the changes needed:
- Change all `measurement` to `total_increasing` for all energy sensors.
- Remove `last_reset` from all the places where it appears.

AbeltjeNL, 2021-09-03:
> So the changes needed:
> 
> - Change all `measurement` to `total_increasing` for all energy sensors.
> 
> - Remove `last_reset` from all the places where it appears.
> 
> 

Agreed on this 👍🏻

fufar, 2021-09-03:
so it will be fixed in future versions or i need to fix it manually ? didnt add this topic manually ether

Koenkk, 2021-09-03:
@McGiverGim thanks, done now.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

andriej, 2021-09-05:
And what would be the estimation for next stable release? :-)

AbeltjeNL, 2021-09-05:
> And what would be the estimation for next stable release? :-)

Would like to know that to 😃 My temp fix in customize.yaml works great but it would be nice to see an update before the new major 😇

Koenkk, 2021-09-06:
A new release is made every 1st of the month, so next release will be 1 October

closed 2021-10-14 by github-actions[bot]