Koenkk/zigbee2mqtt #8617: HomeAssistant template warning and error for Hampton Bay/Wink/King of Fans 99432

mwav3, 2021-09-02:
## What happened
The MQTT template for the Hampton Bay/Wink/King of Fans 99432 was recently updated.  Thanks to the great work of @cmccambridge with this PR - https://github.com/Koenkk/zigbee2mqtt/pull/7359 , all fan controls and speeds are now able to be controlled in the Home Assistant template.  Home Assistant has also recently updated their templating requirements to be more rigid, and now expects a default and defined value for all template items.

However, two errors still persist in the logs as a result of the template loaded for this fan.

The light template creates the following error:

```
Exception in state_received when handling msg on 'zigbee2mqtt/Bedroom Fan': '{"fan_mode":"low"}' Traceback (most recent call last): File "/usr/src/homeassistant/homeassistant/components/mqtt/debug_info.py", line 34, in wrapper msg_callback(msg) File "/usr/src/homeassistant/homeassistant/components/mqtt/light/schema_json.py", line 299, in state_received if values["state"] == "ON": KeyError: 'state'
```

The fan template creates the following warning:
```
2021-09-01 21:44:02 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'fan_state' when rendering '{{ value_json.fan_state }}'
```

## What did you expect to happen
No errors in the logs and the template follows Home Assistant's requirements.  I was able to modify templates to control the fan and light and when I set them in my config file for Home Assistant the error and warning are cleared.  I would submit a PR to fix this but unfortunately I do not understand the zigbee2mqtt program enough to find the code and propose changes.  Hopefully someone can take these templates and put them in the files so they can be auto-discovered for other users.

Right now, everything works fine and these warnings do not impact performance, but the concern is that starting with the next release for 10/2021 for Home Assistant, any template warning or error will cause the template to fail, and the device will not work and render the device uncontrollable on the next update.

The light template that fixes the error is here:
```
light:
  - platform: mqtt 
    name: "Bedroom Fan"
    unique_id: "bedroom_fan"
    state_topic: "zigbee2mqtt/Bedroom Fan"
    command_topic: "zigbee2mqtt/Bedroom Fan/set"
    json_attributes_topic: zigbee2mqtt/Bedroom Fan
    brightness_state_topic: "zigbee2mqtt/Bedroom Fan"
    brightness_value_template: "{{ value_json.brightness |float | default('')}}"
    brightness_command_topic: "zigbee2mqtt/Bedroom Fan/set/brightness"
    brightness_scale: 254
    state_value_template: "{{ value_json.state | default('')}}"
```

The fan template that fixes the error is here:
```
fan:
  - platform: mqtt
    name: "Bedroom Fan"
    command_topic: "zigbee2mqtt/Bedroom Fan/set/fan_state"
    availability:
      - topic: "zigbee2mqtt/bridge/state"
    json_attributes_topic: "zigbee2mqtt/Bedroom Fan"
    state_topic: "zigbee2mqtt/Bedroom Fan"
    state_value_template: "{{ value_json.fan_state | default('')}}"
    unique_id: "bedroom_fan"
    speed_range_min: 1
    speed_range_max: 4
    percentage_state_topic: "zigbee2mqtt/Bedroom Fan"
    percentage_value_template: "{{ {'off': 0, 'low': 1, 'medium': 2, 'high': 3, 'on': 4, 'smart': 2}[value_json.fan_mode] | default('') }}"
    percentage_command_topic: "zigbee2mqtt/Bedroom Fan/set/fan_mode"
    percentage_command_template: "{{ {0: 'off', 1: 'low', 2: 'medium', 3: 'high', 4: 'on'}[value] | default('')}}"
    preset_mode_state_topic: "zigbee2mqtt/Bedroom Fan"
    preset_mode_value_template: "{{ 'smart' if value_json.fan_mode == 'smart' else 'normal' }}"
    preset_mode_command_template: "{{ value if value == 'smart' else 'medium' }}"
    preset_mode_command_topic: "zigbee2mqtt/Bedroom Fan/set/fan_mode"
    preset_modes:
      - "normal"
      - "smart"
```

## How to reproduce it (minimal and precise)

Error is displayed as soon as the fan is discovered in Home Assistant

Please let me know if  you need any other info.

## Debug info
Zigbee2MQTT version: 1.21.1
Adapter hardware: CC2652R1, - zzh stick
Adapter firmware version: CC2652R_router_20210128

mwav3, 2021-09-02:
My light template needs some work,  the status just stopped updating on me.

mwav3, 2021-10-03:
Not still, still an issue

OzGav, 2021-10-03:
So of all that is in your templates is it only the `| default('')` that is the change from what zigbee2mqtt creates normally? If so I believe I have found the lines in the code that would need this added...

mwav3, 2021-10-20:
Sorry for the delay.  I switched to the dev branch and still get the log errors for some reason.  I'm not sure why the code didn't fix it - tried restarting home assistant and zigbee2mqtt several times, and reinstalled the mqtt integration.

This is the discovery payload I get:

```
availability:
  - topic: zigbee2mqtt/bridge/state
command_topic: zigbee2mqtt/Bedroom Fan/set/fan_state
device:
  identifiers:
    - zigbee2mqtt_0x0022a3000029410e
  manufacturer: Hampton Bay
  model: Universal wink enabled white ceiling fan premier remote control (99432)
  name: Bedroom Fan
  sw_version: Zigbee2MQTT 1.21.2-dev
json_attributes_topic: zigbee2mqtt/Bedroom Fan
name: Bedroom Fan
percentage_command_template: '{{ {0:''off'', 1:''low'', 2:''medium'', 3:''high'', 4:''on''}[value] | default('''') }}'
percentage_command_topic: zigbee2mqtt/Bedroom Fan/set/fan_mode
percentage_state_topic: zigbee2mqtt/Bedroom Fan
percentage_value_template: >-
  {{ {'off':0, 'low':1, 'medium':2, 'high':3, 'on':4}[value_json.fan_mode] |
  default('None') }}
preset_mode_command_topic: zigbee2mqtt/Bedroom Fan/set/fan_mode
preset_mode_state_topic: zigbee2mqtt/Bedroom Fan
preset_mode_value_template: >-
  {{ value_json.fan_mode if value_json.fan_mode in ['smart'] else 'None' |
  default('None') }}
preset_modes:
  - smart
speed_range_max: 4
speed_range_min: 1
state_topic: zigbee2mqtt/Bedroom Fan
state_value_template: '{{ value_json.fan_state }}'
unique_id: 0x0022a3000029410e_fan_zigbee2mqtt
platform: mqtt
```

The new default option in the code change doesn't seem to feed into the state_value_template for some reason.

Koenkk, 2021-10-20:
@mwav3 could you post the Home Assistant warning? I expect that `fan_state` should be set to `null` now