Koenkk/zigbee2mqtt #8618: Calibration of TS130F

Jens-Wymeersch, 2021-09-02:
## What happened
Basically, I was able to configure 10 out of 11 TS130F curtain switches correctly. One of them doesn't totally open. 
I follow the procedure as described in the doc regarding calibration. Start from an open curtain to a closed one. 
However, when it opens again after the calibration, it stops at 90%. The only thing I can do is switching the calibration on again to get to the full 100%. I wonder if there is a way to make to switch aware that it needs to open longer ?

## What did you expect to happen

## How to reproduce it (minimal and precise)
I changed the switch with another one with the same result. I'm out of options.

## Debug info
Zigbee2MQTT version: I'm using HA Z2M version. 
Adapter hardware: ZZH CC1352R
Adapter firmware version: Current version: 1.21.1-1

McGiverGim, 2021-09-03:
I don't know if will work, but have you tried to do it at the inverse, calibrate when opening? If I'm not wrong it only records "the time" and maybe opening takes longer than closing.

Jens-Wymeersch, 2021-09-03:
@McGiverGim thank you for your reply. I tried this already to calibration the open side but without success. The thing is i have 11 of these switches, 10 of these work great but this one spoils the game. Do you know where the values are stored of the calibration ?

RoboMagus, 2021-09-10:
I've had the exact same issue as well. I'm still looking to improve upon it, but I do have a workaround for my setup which is to just run the calibration time a little longer than required. 
At least that way when opening from __fully__ closed will keep the switch 'opening' for long enough to actually fully open. This is however far from perfect, as dealing with any other position than fully closed will always keep my shutters a little bit away from opening completely...

Jens-Wymeersch, 2021-09-12:
@RoboMagus Today I went through all the curtains I have and added to all 1s extra, which basically did the trick. Thank you for the hint. It also let me realize that Z2M stores the number of calibration seconds and I should check each curtain, which I did. 
Thank you so much, Jens

closed 2021-09-12 by Jens-Wymeersch