Koenkk/zigbee2mqtt #8623: After update yestarday raspbian buster the zzh CC2652R becomes very slow in a raspberry pi 4

kitopopo, 2021-09-03:
Dear @Koenkk ,

After update raspbian buster to de last version the CC2652R becomes very slow. Orders sent from zigbee devices take up to 10 minutes to arrive to the zigbee2mqtt log .

I make a downgrade with a backup from july 2021 and everything goes back to normal.

Do you know how to solve the problem to have raspbian buster updated and be able to continue using zigbee2mqtt correctly? thanks in advance

## Debug info
Zigbee2MQTT version: zigbee2mqtt@1.21.1-dev 
Adapter hardware: CC2652R zzh
Adapter firmware version: First firmware after buy the device in March 2021

lkxnon, 2021-09-03:
Hi. I can confirm same behaviour. RPI4 latest buster update + ZZH stick latest coordinator firmware + zigbee2mqtt latest.
Xiaomi sensors (PIR, door) have delay 10-20seconds, button around 30seconds to get mqtt message out of it.

Regards,
Lukas

Koenkk, 2021-09-03:
I've pushed a possible fix to the latest-dev, can you check that?  (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

Seems related to #8552

kitopopo, 2021-09-03:
> I've pushed a possible fix to the latest-dev, can you check that? (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)
> 
> Seems related to #8552

Dear @Koenkk , thanks for your dedication and time but unfortunately I am currently in the lastet dev branch but the problem  persists.  
After execute npm ci i can see the message: 71 packages are looking for funding run npm fund for details, I do not know if this message has already appeared or if it is related to the problem, I do not know, but after executing npm fund i can see a tree with other devs versions. I don't understand what this tree means

we hope you can solve it soon. Thank you very much in advance. Best regars

robertalexa, 2021-09-04:
@Koenkk i am on rpi4 buster with zzh too, on latest stable z2m release and i am NOT experiencing any of these issues. Just figured i should post just to say it does not seem to be systemic, maybe just case specific

Koenkk, 2021-09-04:
@kitopopo so:
- When downgrading to zigbee2mqtt 1.20.0 the problem disappears
- When upgrading again to 1.21.1 the problem appears again

Koenkk, 2021-09-04:
@kitopopo I see now, I first want to check if it is something z2m related, if you update z2m to 1.21.1 on the raspbian buster 31th version, does everything work correctly?

kitopopo, 2021-09-04:
> @kitopopo I see now, I first want to check if it is something z2m related, if you update z2m to 1.21.1 on the raspbian buster 31th version, does everything work correctly?

Yes @Koenkk , I did that test Thursday night. With the raspbian version of August 31 th and the latest version of zigbee2mqtt everything works fine

Koenkk, 2021-09-04:
Interesting, what is the output of: `sudo apt-get dist-upgrade --dry-run`?

robertalexa, 2021-09-04:
@kitopopo is there any reason why you are running dist-upgrade instead of just upgrade? Dist-upgrade can create issues because not only it will upgrade your packages, it will also remove what it deems not needed, as it has a conflict resolution.

Also, i wouldn't advice ever installing or upgrading with -y because you will not get a chance to see what will happen, since you give consent ahead of time

Koenkk, 2021-09-04:
@robertalexa that triggers me

@kitopopo from the working situation can you do
- sudo apt-get upgrade -y
- see if z2m still works
- do a `sudo apt-get dist-upgrade --dry-run`     and provide the output here

robertalexa, 2021-09-04:
@Koenkk sorry? What i said?

kitopopo, 2021-09-04:
> Interesante, ¿cuál es el resultado de `sudo apt-get dist-upgrade --dry-run`:?

> * sudo apt-get upgrade -y

Dear @Koenk,

-> sudo apt-get upgrade -y
Leyendo lista de paquetes... Hecho
Creando árbol de dependencias       
Leyendo la información de estado... Hecho
Calculando la actualización... Hecho
0 actualizados, 0 nuevos se instalarán, 0 para eliminar y 0 no actualizados.
(z2m works correctly)

-> sudo apt-get dist-upgrade --dry-run
Leyendo lista de paquetes... Hecho
Creando árbol de dependencias       
Leyendo la información de estado... Hecho
Calculando la actualización... Hecho
0 actualizados, 0 nuevos se instalarán, 0 para eliminar y 0 no actualizados.

--> sudo apt-get -y update
--> sudo apt-get -y upgrade

Updated and Upgraded ok but z2m fails, is too slowly after reboot rp4. I'm going to try to update again because the output is too big and I can't copy it all, I'm going to do it. Thanks

kitopopo, 2021-09-04:
> @robertalexa that triggers me
> 
> @kitopopo from the working situation can you do
> 
> * sudo apt-get upgrade -y
> * see if z2m still works
> * do a `sudo apt-get dist-upgrade --dry-run`     and provide the output here

--> sudo apt-get -y update
```
Des:1 http://raspbian.raspberrypi.org/raspbian buster InRelease [15,0 kB]
Des:2 http://archive.raspberrypi.org/debian buster InRelease [32,6 kB]
Des:3 https://downloads.plex.tv/repo/deb public InRelease [6.685 B]                                                    
Des:4 https://repo.mosquitto.org/debian buster InRelease [12,4 kB]                                                     
Obj:5 https://deb.nodesource.com/node_10.x buster InRelease                   
Des:6 http://raspbian.raspberrypi.org/raspbian buster/main armhf Packages [13,0 MB]
Des:7 https://downloads.plex.tv/repo/deb public/main armhf Packages [432 B]
Des:8 http://archive.raspberrypi.org/debian buster/main armhf Packages [380 kB]
Des:9 https://repo.mosquitto.org/debian buster/main all Packages [4.174 B]
Des:10 https://repo.mosquitto.org/debian buster/main armhf Packages [35,5 kB]
Descargados 13,5 MB en 6s (2.241 kB/s)                                                                                 
Leyendo lista de paquetes... Hecho
N: Repository 'http://raspbian.raspberrypi.org/raspbian buster InRelease' changed its 'Suite' value from 'stable' to 'oldstable'
```
--> sudo apt-get -y upgrade

```
Leyendo lista de paquetes... Hecho
Creando árbol de dependencias       
Leyendo la información de estado... Hecho
Calculando la actualización... Hecho
Los siguientes paquetes se han retenido:
  mosquitto
Se actualizarán los siguientes paquetes:
  aspell bluez firefox-esr firefox-esr-l10n-es-es libaspell15 libbluetooth3 libgssapi-krb5-2
  libjavascriptcoregtk-4.0-18 libk5crypto3 libkrb5-3 libkrb5support0 libmosquitto1 libnftnl11 libpam-systemd
  libraspberrypi-bin libraspberrypi-dev libraspberrypi-doc libraspberrypi0 libsndfile1 libsystemd0 libudev-dev
  libudev1 libwebkit2gtk-4.0-37 linux-libc-dev mosquitto-clients openjdk-11-jdk openjdk-11-jdk-headless openjdk-11-jre
  openjdk-11-jre-headless raspberrypi-bootloader raspberrypi-kernel raspberrypi-sys-mods rp-bookshelf systemd
  systemd-sysv udev
36 actualizados, 0 nuevos se instalarán, 0 para eliminar y 1 no actualizados.
Se necesita descargar 402 MB de archivos.
Se utilizarán 2.511 kB de espacio de disco adicional después de esta operación.
Des:1 http://archive.raspberrypi.org/debian buster/main armhf bluez armhf 5.50-1.2~deb10u2+rpt1 [770 kB]
Des:3 https://repo.mosquitto.org/debian buster/main armhf mosquitto-clients armhf 2.0.12-0mosquitto1~buster1 [107 kB]
Des:2 http://ftp.cica.es/mirrors/Linux/raspbian/raspbian buster/main armhf systemd-sysv armhf 241-7~deb10u8+rpi1 [101 kB]
Des:4 https://repo.mosquitto.org/debian buster/main armhf libmosquitto1 armhf 2.0.12-0mosquitto1~buster1 [82,6 kB]     
Des:5 http://archive.raspberrypi.org/debian buster/main armhf libbluetooth3 armhf 5.50-1.2~deb10u2+rpt1 [94,0 kB]                 
Des:6 http://archive.raspberrypi.org/debian buster/main armhf libnftnl11 armhf 1.1.7-1~bpo10+1~0 [45,1 kB]
Des:7 http://archive.raspberrypi.org/debian buster/main armhf libraspberrypi-doc armhf 1:1.20210831-1 [31,4 MB]
Des:8 http://ftp.cica.es/mirrors/Linux/raspbian/raspbian buster/main armhf libpam-systemd armhf 241-7~deb10u8+rpi1 [194 kB]
Des:9 http://ftp.cica.es/mirrors/Linux/raspbian/raspbian buster/main armhf libudev-dev armhf 241-7~deb10u8+rpi1 [110 kB]                
Des:10 http://ftp.cica.es/mirrors/Linux/raspbian/raspbian buster/main armhf libsystemd0 armhf 241-7~deb10u8+rpi1 [306 kB]           
Des:11 http://ftp.cica.es/mirrors/Linux/raspbian/raspbian buster/main armhf systemd armhf 241-7~deb10u8+rpi1 [3.258 kB]              
Des:12 http://ftp.cica.es/mirrors/Linux/raspbian/raspbian buster/main armhf udev armhf 241-7~deb10u8+rpi1 [1.248 kB]              
Des:13 http://ftp.cica.es/mirrors/Linux/raspbian/raspbian buster/main armhf libudev1 armhf 241-7~deb10u8+rpi1 [144 kB]          
Des:14 http://ftp.cica.es/mirrors/Linux/raspbian/raspbian buster/main armhf aspell armhf 0.60.7~20110707-6+deb10u1 [214 kB]       
Des:15 http://ftp.cica.es/mirrors/Linux/raspbian/raspbian buster/main armhf libaspell15 armhf 0.60.7~20110707-6+deb10u1 [275 kB]
Des:16 http://archive.raspberrypi.org/debian buster/main armhf libraspberrypi-dev armhf 1:1.20210831-1 [400 kB]                       
Des:17 http://archive.raspberrypi.org/debian buster/main armhf raspberrypi-kernel armhf 1:1.20210831-1 [79,6 MB]
Des:18 http://ftp.cica.es/mirrors/Linux/raspbian/raspbian buster/main armhf firefox-esr-l10n-es-es all 78.13.0esr-1~deb10u1+rpi1 [547 kB]
Des:19 http://ftp.cica.es/mirrors/Linux/raspbian/raspbian buster/main armhf firefox-esr armhf 78.13.0esr-1~deb10u1+rpi1 [49,1 MB]                
Des:20 http://ftp.cica.es/mirrors/Linux/raspbian/raspbian buster/main armhf libgssapi-krb5-2 armhf 1.17-3+deb10u2 [136 kB]              
Des:21 http://ftp.cica.es/mirrors/Linux/raspbian/raspbian buster/main armhf libkrb5-3 armhf 1.17-3+deb10u2 [319 kB]
Des:22 http://ftp.cica.es/mirrors/Linux/raspbian/raspbian buster/main armhf libkrb5support0 armhf 1.17-3+deb10u2 [62,4 kB]
Des:23 http://ftp.cica.es/mirrors/Linux/raspbian/raspbian buster/main armhf libk5crypto3 armhf 1.17-3+deb10u2 [118 kB]
Des:24 http://ftp.cica.es/mirrors/Linux/raspbian/raspbian buster/main armhf libwebkit2gtk-4.0-37 armhf 2.32.3-1~deb10u1+rpi1 [11,2 MB]
[... cut ...]
Procesando disparadores para gnome-menus (3.31.4-3) ...
Procesando disparadores para libc-bin (2.28-10+rpi1) ...
Procesando disparadores para man-db (2.8.5-2) ...
Procesando disparadores para dbus (1.12.20-0+deb10u1) ...
Procesando disparadores para desktop-file-utils (0.23-4) ...
Procesando disparadores para initramfs-tools (0.133+deb10u1) ...
Procesando disparadores para dictionaries-common (1.28.1) ...
pi@raspberrypi:~ $ 

```

after upgrade without restart rp4 works fine, after restar the same problem, too slowly.

thanks in advanced

Koenkk, 2021-09-04:
Can you try to:

1. Start from the working situation
1. `sudo apt-get update`
1. `sudo apt-mark hold raspberrypi*`
1. `sudo apt-get upgrade -y`
1. `reboot`
1. now check if z2m still works correctly
1. `sudo apt-mark unhold raspberrypi*`
1. `reboot`
1. check if z2m works again, if it worked at step 6 I would expect it to fail now.

kitopopo, 2021-09-05:
> Can you try to:
> 
> 1. Start from the working situation
> 2. `sudo apt-get update`
> 3. `sudo apt-mark hold raspberrypi*`
> 4. `sudo apt-get upgrade -y`
> 5. `reboot`
> 6. now check if z2m still works correctly
> 7. `sudo apt-mark unhold raspberrypi*`
> 8. `reboot`
> 9. check if z2m works again, if it worked at step 6 I would expect it to fail now.

Dear @Koenkk , thanks again by your dedication and time.
In the step 6 the z2m works correctly. the following packeds were retained:

```
raspberrypi-bootloader  retenido.
raspberrypi-sys-mods retenido.
raspberrypi-ui-mods  retenido.
raspberrypi-artwork  retenido.
raspberrypi-kernel  retenido.
raspberrypi-net-mods  retenido.
raspberrypi-archive-keyring  retenido.
raspberrypi-kernel-headers fijado como retenido.
```

After unhold raspberrypi* and after make the upgrade and reset fails again. it seems that one of these 8 packages is affecting z2m. each of the 8 packages can be retained?

Thanks in advanced

kitopopo, 2021-09-05:
Dear @koenkk,

I have holded only the packets : raspberrypi-kernel and raspberrypi-kernel-headers and after upgrade and reset works correctly. 
one or both packages have the problem. What would be the next step? Thanks again

Koenkk, 2021-09-05:
It seems the latest kernel has some issues, looks the same as e.g. #8647, I'm not sure how to further investigate this

Koenkk, 2021-09-05:
@kitopopo what is the output of `uname -r` in both the working and non-working situation?

kitopopo, 2021-09-05:
> @kitopopo what is the output of `uname -r` in both the working and non-working situation?

Dear @Koenkk , before upgrade 5.10.17-v7l+ . After upgrade 5.10.60-v7l+.
Thanks

pedrolamas, 2021-09-05:
As I've already discussed with @Koenkk, I can also confirm the findings here on both my NanoPi M4V2 (that I use as my production IoT docker stack) and my old NanoPi NEO2.

I took the NanoPi NEO2 and installed the latest armbian available that has a kernel 5.10.60, and I could repro the issue 100%: during a period of a few seconds, I opened and closed the sensor a few times and nothing showed on the logs, but once I pressed the small button on the sensor, all the events came at once!

I then used `sudo armbian-config` to downgrade the kernel to 5.4.88, rebooted, ran exact same test and everything now worked fine, so I could see all the events as they happened, no need to press the button.

I then updatred the kernel back to 5.10.60, and once again I could repro the issue.

I also tried updating the kernel to edge version 5.13.12, and once again could see the issue...

Here are the kernel I tried, and the results (red cross for repro of issue, green tick for working as expected)

[image]

Koenkk, 2021-09-05:
@puddly @Hedda are you seeing similar issues with ZHA?

Koenkk, 2021-09-05:
I believe this issue is caused by https://github.com/torvalds/linux/commit/3c18e9baee0ef97510dcda78c82285f52626764b#diff-3fa83c321e18090436b2da607d2d415ef62caf9be421c2e3bc13414fcbbee988 and has been fixed in https://github.com/torvalds/linux/commit/df7b16d1c00ecb3da3a30c999cdb39f273c99a2f#diff-3fa83c321e18090436b2da607d2d415ef62caf9be421c2e3bc13414fcbbee988 (look at the description of this commit)

puddly, 2021-09-05:
> are you seeing similar issues with ZHA?

My production network isn't on a machine with a 5.x kernel but it looks like Home Assistant OS may be building affected kernels for some hardware combinations. I've not seen any GitHub issues regarding this problem yet.

kitopopo, 2021-09-05:
> I believe this issue is caused by [torvalds/linux@3c18e9b#diff-3fa83c321e18090436b2da607d2d415ef62caf9be421c2e3bc13414fcbbee988](https://github.com/torvalds/linux/commit/3c18e9baee0ef97510dcda78c82285f52626764b#diff-3fa83c321e18090436b2da607d2d415ef62caf9be421c2e3bc13414fcbbee988) and has been fixed in [torvalds/linux@df7b16d#diff-3fa83c321e18090436b2da607d2d415ef62caf9be421c2e3bc13414fcbbee988](https://github.com/torvalds/linux/commit/df7b16d1c00ecb3da3a30c999cdb39f273c99a2f#diff-3fa83c321e18090436b2da607d2d415ef62caf9be421c2e3bc13414fcbbee988) (look at the description of this commit)

Dear @Koenkk ,

on raspberry pi i can't find any folder (drivers/usb/serial/ch341.c ) to be able to edit the file.  I do not know the process, what would be the solution to fix this problem? Will we have to keep kernel without updating? get rid of the CC2652R stick?

Thanks in advanced

hgelpke, 2021-09-05:
I'm having the same issue on HassOS Proxmox install.

Koenkk, 2021-09-06:
I suggest to keep running an older kernel version. If it is indeed caused by https://github.com/Koenkk/zigbee2mqtt/issues/8623#issuecomment-913202614 

In the 5.10.* range it seems that 5.10.58 - 5.10.61 are affected, in 5.10.62 the problem is fixed.

bossgos, 2021-09-06:
> I'm having the same issue on HassOS Proxmox install.

I`m having same... also Proxmox with Hass OS

Koenkk, 2021-09-06:
For hass os users, this issue will be fixed in 6.4 (https://github.com/home-assistant/operating-system/issues/1532#issuecomment-913440970), for now downgrade to 6.2 with:

```
ha os update --version 6.2
```

kitopopo, 2021-09-06:
> I suggest to keep running an older kernel version (if it is indeed caused by [#8623 (comment)](https://github.com/Koenkk/zigbee2mqtt/issues/8623#issuecomment-913202614) )
> 
> In the 5.10.* range it seems that 5.10.58 till 5.10.61 are affected, in 5.10.62 the problem is fixed.

Thanks dear @Koenkk, i wait to the 5.10.62 kernel upgrade. Best regards

Koenkk, 2021-09-06:
Since this is not something that can be fixed from z2m i've created a pinned issue with the affected kernel version. https://github.com/Koenkk/zigbee2mqtt/issues/8663 

Let's close this one.

closed 2021-09-06 by Koenkk

bossgos, 2021-09-07:
> For hass os users, this issue will be fixed in 6.4 ([home-assistant/operating-system#1532 (comment)](https://github.com/home-assistant/operating-system/issues/1532#issuecomment-913440970)), for now downgrade to 6.2 with:
> 
> ```
> ha os update --version 6.2
> ```

Thanks it`s working :)