Koenkk/zigbee2mqtt #8625: Issue while publishing a message

adriannowaq, 2021-09-03:
Hi, I have trouble with setting state while many devices are connected

Log: 

> Sep 03 15:53:51 raspberrypi npm[1333]: Zigbee2MQTT:debug 2021-09-03 15:53:51: Publishing 'set' 'state' to 'plug-2'
> Sep 03 15:54:00 raspberrypi npm[1333]: Zigbee2MQTT:error 2021-09-03 15:54:00: Publish 'set' 'state' to 'plug-2' failed: 'Error: Command 0x60a423fffe7b55ec/1 genOnOff.toggle({}, {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (SRSP - AF - dataRequest after 6000ms)'
> Sep 03 15:54:00 raspberrypi npm[1333]: Zigbee2MQTT:debug 2021-09-03 15:54:00: Error: Command 0x60a423fffe7b55ec/1 genOnOff.toggle({}, {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (SRSP - AF - dataRequest after 6000ms)
> Sep 03 15:54:00 raspberrypi npm[1333]:     at Timeout._onTimeout (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35)
> Sep 03 15:54:00 raspberrypi npm[1333]:     at listOnTimeout (internal/timers.js:557:17)
> Sep 03 15:54:00 raspberrypi npm[1333]:     at processTimers (internal/timers.js:500:7)
> Sep 03 15:54:00 raspberrypi npm[1333]: Zigbee2MQTT:info  2021-09-03 15:54:00: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'state' to 'plug-2' failed: 'Error: Command 0x60a423fffe7b55ec/1 genOnOff.toggle({}, {\"sendWhenActive\":false,\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (SRSP - AF - dataRequest after 6000ms)'","meta":{"friendly_name":"plug-2"},"type":"zigbee_publish_error"}'
> 

configuration:
```

advanced:
  log_level: debug
  cache_state: false
  cache_state_persistent: false
homeassistant: false
permit_join: true
mqtt:
  base_topic: zigbee2mqtt
  server: mqtt://192.168.4.70:1883
  force_disable_retain: true
serial:
  port: /dev/ttyACM0
external_converters:
  - temp-humidity.js
devices:
  '0x60a423fffe7b571f':
    friendly_name: plug-1
  '0x60a423fffe7b55ec':
    friendly_name: plug-2
  '0x60a423fffe7eda5c':
    friendly_name: plug-3
  '0x60a423fffe7df0f6':
    friendly_name: plug-4
  '0x60a423fffe7ecdd6':
    friendly_name: plug-5
  '0x60a423fffe6ecd93':
    friendly_name: plug-6
  '0x60a423fffe7b5598':
    friendly_name: plug-7
  '0x60a423fffe7df033':
    friendly_name: plug-8
  '0x60a423fffe7b56e2':
    friendly_name: plug-9
  '0x60a423fffeae6f16':
    friendly_name: plug-10
  '0x14b457fffedf1a7a':
    friendly_name: temp-1
  '0x60a423fffe44d3f3':
    friendly_name: temp-2
```

masterd89, 2021-09-03:
Hello, 
i seem to have a similar problem: when turning off my ikea bulb i got this:

> Zigbee2MQTT:error 2021-09-03 19:23:35: Publish 'set' 'state' to 'IKEA_BULB' failed: 'Error: Command 0x5c0272fffe539374/1 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received)'
> Zigbee2MQTT:info  2021-09-03 19:23:35: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'state' to 'IKEA_BULB' failed: 'Error: Command 0x5c0272fffe539374/1 genOnOff.off({}, {\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (no response received)'","meta":{"friendly_name":"IKEA_BULB"},"type":"zigbee_publish_error"}'

then, after about a minute, it seems to have reestablished the connection (and set the wrong state btw):

Zigbee2MQTT:info  2021-09-03 19:24:25: MQTT publish: topic 'zigbee2mqtt/IKEA_BULB', payload '{"brightness":254,"linkquality":81,"state":"ON","update":{"state":"idle"},"update_available":false}'

Tonio16, 2021-09-20:
Heĺlo
`I have also this error for ikea devices, or I think so:
Publish 'get' 'state' to '0x086bd7fffe0134d3' failed: 'Error: Read 0x086bd7fffe0134d3/1 genOnOff(["onOff"], {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 28440 - 1 - 233 - 6 - 1 after 10000ms)'
`
I am running z2m with jeedom and zigbeelinker (= I don’t know how to have herdsman logs) on a pi3+ and a zigate v1 (firmware 3.1d).

Version Zigbee2MQTT 1.21.1 commit: 4a51e0c0 
Type de Coordinateur zigate 
Révision du Coordinateur 1d03 
Version Interface 0.6.0

Antoine