Koenkk/zigbee2mqtt #8630: 1.21.0 breaks half functionality for XHK1-TC Xfinity Alarm security keypad

ikke-zelf, 2021-09-03:
### Discussed in https://github.com/Koenkk/zigbee2mqtt/discussions/8629

<div type='discussions-op-text'>

<sup>Originally posted by **ikke-zelf** September  3, 2021</sup>
Hi,
Since a little time ( I expect this to be as of release 1.21.0 as this is where I notice an entry in the release notes for the XHK1-TC Xfinity Alarm security keypad) the functionality for the keypad icw zigbee2mqtt has been stripped and changed.

Prior to this we had the following modes available:
disarm, arming_night, arming_away, arming_stay, entry_delay, exit_delay and in_alarm.
When one of the 'arming' modes was set the keypad would lit up the corresponding button light in red.
When the entry or exit_delay was set, the keypad would starts beeping until a different mode is set.

Now the supported modes are: disarm, arm_day_zones, arm_night_zones, arm_all_zones, invalid_code, not_ready, already_disarmed.
The buttons (top three) do not lit anymore, and the beeping is gone when setting (trying to) the exit_delay.

I use this functionality with node-red and home assistant and am not into this coding enough to put my finger on what to change in git to restore the functionality.

any help would be appreciated</div>

Koenkk, 2021-09-03:
@robertalexa  could this because of https://github.com/Koenkk/zigbee-herdsman-converters/commit/439187782c62c6f73c78e51dfdb7019641d33209?

robertalexa, 2021-09-03:
Hi @ikke-zelf 

I would suggest you start with the updated documentation for your device https://www.zigbee2mqtt.io/devices/XHK1-TC.html

The last update brought changes to all the alarm keypads and their flow of operation. In the previous version of z2m, the commands and values used were erroneous and not in accordance with the ZCL specs.

This has been corrected in this update and under the release notes you can see the following:
`Improve handling of keypad alarm arming, check your alarm documentation for updated procedure`

For a full explanation of the reasoning please read https://github.com/Koenkk/zigbee-herdsman-converters/pull/2967

In short, you just need to update the logic of operation in your Node-Red flow, by following the documentation. Such a logical flow is also described here: https://github.com/Koenkk/zigbee-herdsman-converters/pull/2967#issuecomment-909225362

PS: make sure to backup your current node-red just in case. Either export it and keep it safe, or make a duplicate, disable the current one and edit the new one.

For any questions, issues or just general help let me know, I will help you :)

ikke-zelf, 2021-09-03:
mmm, i'm trying to understand what is changed.
armmode vs armnotification is something I see.

What I don't understand is that I'm using arm_mode with transaction for a keypad and now I can't use entry and exit_delay anymore. even the buttons don't lit up anymore.

I have read the links you provided, but still I get figure out what a change in my node-red scheme would solve the reduction in functionality I encounter.
You talk about the zigbee design rules, but strange it is that this reduces flexibility and functionality.

I would like to get your help in understanding what needs to be done to 'fix' this. as in my case this seems broken now.

My flow in node-red:
mqtt-in -> keypad filter -> if a button is pressed with the correct 4 digit code -> set the corresponding arm_mode by sending payload message with transaction and armo_mode through mqqt-out to keypad.

result: normally the corresponding button turns red. and within Home Assistant a alarm state is set which can be referred to.

mqtt-in-> motion sensor/door sensor triggered -> if arm_mode is in a predefined 'On' state the alarm/siren will sound.

exit delay/entry delay:
when a door sensor is triggered when the alarm is set to 'arm_all_zones' or 'arm_day_zones' or 'arm_night_zones', the node-red will send an arm_mode 'entry_delay' towards the keypad which in his turn starts beeping.
when the time delay has passed (within the node-red) the node-red itself sends the command to the siren to sound the alarm and the node-red send the arm_mode 'in_alarm' to the keypads.

some additional flow settings have been set in my node-red, but I expect you get the idea how it (should ) function.

in my turn I don't understand why this had to be changed.

please explain

robertalexa, 2021-09-03:
> in my turn I don't understand why this had to be changed.

https://github.com/Koenkk/zigbee-herdsman-converters/pull/2967#issuecomment-909431577

> mqtt-in -> keypad filter -> if a button is pressed with the correct 4 digit code -> set the corresponding arm_mode by sending payload message with transaction and armo_mode through mqqt-out to keypad.

After the keypad sends a request to the server, the server MUST either accept or decline the action with the correct transaction number. 

If the pin is invalid the server MUST respond with "invalid_code"
If the pin is valid the server MUST respond with the SAME arm_mode as the keypad asked for.

For example:
- keypad, with correct pin, asks to `arm_all_zones`
- node-red confirms the action with transaction number and armNotification `arm_all_zones`
- node-red sets the arm_mode to `exit_delay`. this updates the global variable. If the keypad is pressed now, the armed button will be flashing see step 2 onwards
- server waits the exit_delay time e.g. 30 seconds then sets the arm_mode to `arm_all_zones`

All you need is basically an extra action. You respond to the notification with the requested `arm_mode`, then you send a new command to set the `arm_mode`

PS: all the "MUST" is decided by the ZCL standard, that means that this is how all the zigbee devices in this category should operate. Hence the code was brought in line with the standard. Unfortunately you have set your automation when this was incorrect. It is a big annoying but it is for the best for all future users, because things will be consistent and according to documentation.

Let me know how it goes

robertalexa, 2021-09-03:
This is the documentation I am refering to:
[image]

robertalexa, 2021-09-03:
> armmode vs armnotification is something I see.

arm_mode is this - server SETS the state of the alarm
```
{
    "arm_mode": {
        "mode": "arm_all_zones"
    }
}
```

arm_notifications is this - server responds to keypad request
```
{
    "arm_mode": {
        "transaction": 99, // Transaction number (this must be the same as the keypad request `action_transaction`)
        "mode": "arm_all_zones" // Mode (this must be the same as the keypad request `action`)
    }
}
```

ikke-zelf, 2021-09-03:
Explanation is good. I think I understand. It's a two step 'command' towards the keypad now.
i will test tomorrow as testing with the beeps will get the house awake ;-)

Will let you know when done testing and share the outcome

robertalexa, 2021-09-04:
That is exactly it. Functionality and capability is identical.

Just a separate command to respond to the keypad followed by another command to set the required alarm arm mode.

Let me know. I would be more than happy to help you sort it out

ikke-zelf, 2021-09-04:
hi,

Just implemented most of the stuff. Until now it seems to work.
One remark however, it seems that it does not matter which transaction_id I use. I receive the id from the keypad, do nothing with it, send mode command with transaction _id=1 and it works every time.

Can you elaborate on what the usecase is of the id?

robertalexa, 2021-09-04:
According to the specification, the server responds to the alarm keypad's command including the transaction id of that command. This way the keypad "reacts". The keypad (at least mine) sends the same command 3 times in a row (3 attempts) with incrementing transaction numbers e.g. 55, 56, 57. If it does not receive a response then the "network error" light comes on.

I do not know enough, but i think the keypad has no idea about the transaction id, i do not think it keeps track of it. I think the transaction id benefits only the server, so that (if you want) you can create a log of all the actions that happened.

So it kinda makes sense that the keypad will accept any transaction id, but it definitely isn't best practice, not if you intend to keep a log, because then you will not know what commands match what keypad requests.

I am glad to hear that everything is working :) . If you are happy with everything, could you please Close this issue?

ikke-zelf, 2021-09-04:
just trying to figure out the 'entry_delay' flow setup.

guess; notice breach, send arm_mode=entry_delay ?, wait eg 30 seconds, send arm_mode=in_alarm ?

any toughts about this?

robertalexa, 2021-09-04:
That is exactly how i would do it too :)

A sensor triggers node-red
node-red changes arm_mode to entry_delay. This change should also notify the keypad and the keypad should start beeping (if your device can do that)
node-red starts a 30seconds timer. cancel this timer if the users successfully disarms.
if not, arm_mode in_alarm after the timer is over.

ikke-zelf, 2021-09-04:
{
    "arm_mode": {
        "mode": "entry_delay"
    }
}

then wait for eg 30 seconds, which during this time the keypad starts beeping.

Check if Alarm is not disarmed during this 30 seconds time frame; if not then:
{
    "arm_mode": {
        "mode": "in_alarm",
        "transaction": 23
    }
}

ikke-zelf, 2021-09-04:
had to remove the transaction_id in the first arm_mode command (entry_delay).

now it's working.

Thanks for all the help. will close the issue now

closed 2021-09-04 by ikke-zelf

robertalexa, 2021-09-04:
No, only this:
{
  "arm_mode": {
       "mode": "in_alarm",
  }
}

No need for transaction, because it is not a response to the keypad. Only use "transaction" to respond to keypad

ikke-zelf, 2021-09-05:
I notice that the keypad does not provide a single beep when I disarm the alarm.
Prior to the changes the keypad sounded a single beep when the alarm was armed or disarmed.
any thoughts about that?

robertalexa, 2021-09-05:
Can you please paste here exactly the mqqt commands you use in these scenarios. I want to validate them.

The keypad would also have beeped if the arm notification you were sending contained an invalid arm mode. for example exit delay would have been invalid, and the keypad could have been exactly because of that.

rindlerblabla, 2021-09-05:
Maybe I am just making thinks messier, but I am on 1.21.0 as well and for me the difference between sound and no sound when arming depends on if I send the transaction code or not, similar to what is described in this old post. https://github.com/Koenkk/zigbee2mqtt/issues/4613#issuecomment-782933457

robertalexa, 2021-09-05:
> Maybe I am just making thinks messier, but I am on 1.21.0 as well and for me the difference between sound and no sound when arming depends on if I send the transaction code or not, similar to what is described in this old post. [#4613 (comment)](https://github.com/Koenkk/zigbee2mqtt/issues/4613#issuecomment-782933457)

This is what I was thinking. That is why i asked OP for his mqtt commands. You need to send both of them, with transaction and without, because the 2 commands have different functions :)

Alarm keypad with correct pin asks for `disarm`
Server responds with:
```
{
    "arm_mode": {
        "transaction": 99, // Transaction number (this must be the same as the keypad request `action_transaction`)
        "mode": "disarm"
    }
}
```
This is to confirm to the keypad the transaction, but DOES NOT set the arm_mode

Server sends another transaction, immediatelly:
```
{
    "arm_mode": {
        "mode": "disarm"
    }
}
```

This now sets the arm_mode.

ikke-zelf, 2021-09-05:
gathering the commands to post, reveald some mistakes in flow on my end.
Let me recheck/modify and test again on my side before taking more of your time.

ikke-zelf, 2021-09-11:
Just ti let you know, everything is working as expected. Thanks for the help !

robertalexa, 2021-09-11:
Happy to hear :)

nielsfaber, 2021-09-19:
I was wondering where I can buy this product (or the Centralite version) in the Netherlands or EU. Anyone kind enough to share this?

rindlerblabla, 2021-09-19:
No, have only seen people from outside EU selling them on eBay. Am using it with your Alarmo integation and it really adds an extra touch to it!

thecobra666, 2021-09-28:
I still have an issue that arming night mode does not stop the exit_delay. When arming all zones it works like expected, the only difference in the code is arm_night_zones vs arm_all_zones. 

To sketch the process:

push zone button and enter code
respond back with requested zone and transaction id = keypad beeps once
send message exit_delay => keypad starts beeping
after 15 seconds respond again with requested zone but without transaction id = zone lights up on keypad = ok
exit_delay keeps beeping.

This only happens when using arm_night_zones instead of arm_all_zones.

robertalexa, 2021-09-28:
@thecobra666 your logic is flawed, hence it not behaving as you expect. Arming night zone (home) does not imply an exit, so you don't need to reply with exit delay. You are at home and you are not going anywhere, so the alarm system doesn't not need to give you a waiting time. You just reapond with night mode

thecobra666, 2021-09-28:
> @thecobra666 your logic is flawed, hence it not behaving as you expect. Arming night zone (home) does not imply an exit, so you don't need to reply with exit delay. You are at home and you are not [...]

I suspected as much. That's why I already hard coded the exit_delay into the arm_all_zones mode instead of generic when the alarm goes into "arming" which would set the keypad to exit_delay. 

No problem, will use it like this then.

normanr, 2021-10-17:
It might be nice to actually split the armMode enum, and add a panelStatus enum. If possible also rename/split "arm_mode", so that instead of using transaction to switch between armNotification and panelStatus, there are two different named messages (one for notification and another for status).

eg: "receive armMode, reply with armNotification, and send new panelStatus" might be easier to understand.

thecobra666, 2024-02-19:
So the entry or exit delay is still not working for the XHK1-TC. It's working fine for the XHK1-UE but not for the TC. All other functions are working on the TC but the sounds of the entry/delay not. Is there an error in the code somewhere?

thecobra666, 2024-02-24:
I've noticed the following:

I want to add a little here there is indeed a difference with how the keypad is responding.

The UE model does this when arming the exit_delay / entry_delay etc..

`Debug ug 2024-02-24 09:18:56Received MQTT message on 'zigbee2mqtt/0x000d6f0014a6fe0a/set' with data '{ "arm_mode": { "mode": "entry_delay" } }'
Debug 2024-02-24 09:18:56Publishing 'set' 'arm_mode' to '0x000d6f0014a6fe0a'
Debug 2024-02-24 09:19:07Received MQTT message on 'zigbee2mqtt/0x000d6f0014a6fe0a/set' with data '{ "arm_mode": { "mode": "disarm" } }'
Debug 2024-02-24 09:19:07Publishing 'set' 'arm_mode' to '0x000d6f0014a6fe0a'

`
The TC does this:

Debug 2024-02-24 09:19:57Received MQTT message on 'zigbee2mqtt/keypadboven/set' with data '{ "arm_mode": { "mode": "exit_delay" } }'
Debug 2024-02-24 09:19:57Publishing 'set' 'arm_mode' to 'keypadboven'
Debug 2024-02-24 09:19:58Received Zigbee message from 'keypadboven', type 'commandGetPanelStatus', cluster 'ssIasAce', data '{}' from endpoint 1 with groupID null
Info 2024-02-24 09:19:58MQTT publish: topic 'zigbee2mqtt/keypadboven', payload '{"action":null,"action_code":null,"action_transaction":null,"action_zone":null,"battery":100,"battery_low":false,"contact":true,"last_seen":"2024-02-24T08:19:33.313Z","linkquality":255,"occupancy":true,"presence":null,"tamper":true,"temperature":17,"voltage":6700}'
Info 2024-02-24 09:21:28MQTT publish: topic 'zigbee2mqtt/keypadboven', payload '{"action":null,"action_code":null,"action_transaction":null,"action_zone":null,"battery":100,"battery_low":false,"contact":true,"last_seen":"2024-02-24T08:19:33.313Z","linkquality":255,"occupancy":false,"presence":null,"tamper":true,"temperature":17,"voltage":6700}'

They keypad responds back, so I'm guessing it's asking for a response and that's why it's not working.