Koenkk/zigbee2mqtt #8635: devices disappear even though they are still listed in raw section

petertmooney, 2021-09-03:
Bug report?
- if I remove the zzh adapter or change port, upon rebooting all devices disappear and need re-paied even though the are still showing in raw.
- Provide the herdsman debug logging: https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

{
    "commit": "e050a76\n",
    "config": {
        "advanced": {
            "adapter_concurrent": null,
            "adapter_delay": null,
            "availability_blacklist": [],
            "availability_blocklist": [],
            "availability_passlist": [],
            "availability_timeout": 0,
            "availability_whitelist": [],
            "cache_state": true,
            "cache_state_persistent": true,
            "cache_state_send_on_startup": true,
            "channel": 11,
            "elapsed": true,
            "ext_pan_id": [
                221,
                221,
                221,
                221,
                221,
                221,
                221,
                221
            ],
            "homeassistant_discovery_topic": "homeassistant",
            "homeassistant_legacy_entity_attributes": true,
            "homeassistant_legacy_triggers": true,
            "homeassistant_status_topic": "homeassistant/status",
            "ikea_ota_use_test_url": true,
            "last_seen": "disable",
            "legacy_api": true,
            "log_directory": "/app/data/log/%TIMESTAMP%",
            "log_file": "log.txt",
            "log_level": "debug",
            "log_output": [
                "console",
                "file"
            ],
            "log_rotation": true,
            "log_symlink_current": true,
            "log_syslog": {
                "app_name": "Zigbee2MQTT",
                "eol": "/n",
                "host": "localhost",
                "localhost": "localhost",
                "path": "/dev/log",
                "pid": "process.pid",
                "port": 123,
                "protocol": "tcp4",
                "type": "5424"
            },
            "pan_id": 6754,
            "report": true,
            "rtscts": false,
            "soft_reset_timeout": 0,
            "timestamp_format": "YYYY-MM-DD HH:mm:ss"
        },
        "ban": [],
        "blocklist": [],
        "device_options": {},
        "devices": {
            "0x00124b0022676c64": {
                "friendly_name": "Cooker Hood"
            },
[... 1069 lines cut ...]
}

-->

# Bug Report
## devices disappear after repowering or removing device

## What did you expect to happen
all my paired devices to remain

## How to reproduce it (minimal and precise)
turn off server or restart docker after unplugging the coordinator or changing it to a new USB port. 

## Debug Info
Zigbee2MQTT version
1.21.1-dev commit: e050a76
Coordinator type
zStack3x0
Coordinator revision
20210120
Frontend version
0.6.0
Zigbee2MQTT version:
Adapter hardware: /dev/serial/by-id/usb-Silicon_Labs_slae.sh_cc2652rb_stick_-_slaesh_s_iot_stuff_00_12_4B_00_23_93_2B_60-if00-port0
Adapter firmware version:

petertmooney, 2021-09-04:
so in the raw log all devices are still there, but they are all unpaired. I have to repair them each time. then they assume their previous names and identities. its so frustrating. and I cannot find why it happens.

closed 2021-10-12 by github-actions[bot]

svenf01, 2024-08-08:
I have the same topic with current updated version of zigbee2mqtt.
Device in raw list but not in device list or dashboard.