Koenkk/zigbee2mqtt #8638: Develco Frient ZHEMI101 Energy Meter - Pulse Rate Adjust

lanky8804, 2021-09-04:
With the new addition of the energy monitoring dashboard, how can I set the pulse rate of my electric meter for the Frient zhemi101?

My pulse rate is 3200 imp/kWh

robertalexa, 2021-09-04:
@lanky8804 This is a duplicate of https://github.com/Koenkk/zigbee2mqtt/discussions/8405

Can you please close this issue and let's talk there. I don't have the device but I can try to help you guys out.

robertalexa, 2021-09-04:
[image]

Needed attribute

robertalexa, 2021-09-04:
Create PR to zigbee-herdsman to add support for the attribute:
https://github.com/Koenkk/zigbee-herdsman/pull/413

I will look at commands soon and get you to help with testing please. As mentioned before, i do not own the device so I am working blindly here.

grudyjsy, 2021-09-04:
> Create PR to zigbee-herdsman to add support for the attribute:
> [Koenkk/zigbee-herdsman#413](https://github.com/Koenkk/zigbee-herdsman/pull/413)
> 
> I will look at commands soon and get you to help with testing please. As mentioned before, i do not own the device so I am working blindly here.

Thank you, happy to test when ever you need me to.

robertalexa, 2021-09-04:
Will continue as soon as time allows. I have a few bits started already that i need to finish before, but I will keep you updated.

lanky8804, 2021-09-07:
Thank you, also happy to test anything

grudyjsy, 2021-09-14:
Just in case you want to have some sensors that are more accurate in the meantime I have created these:

template:
  - sensor:
      - name: "Electricity Live Usage"
        unit_of_measurement: W
        state_class: measurement
        device_class: power
        state: "{{ (states('sensor.develco_energy_meter_power')|float * 1000 / 3200) | round(2) }}"

      - name: "Electricity Usage Total"
        unit_of_measurement: kWh
        state_class: measurement
        device_class: energy
        state: "{{ (states('sensor.develco_energy_meter_energy')|float * 1000 / 3200) | round(2) }}"

I can't seem to get them to work in the energy module but the are ok for a dashboard.

pascalschwientek, 2021-09-15:
Hello! Just got this device and I need to set it to 10000 per kwh.
I saw that the PR from @robertalexa in zigbee-herdsman is merged. I installed the edge version of zigbee2mqtt, as the dev branch already includes the updated zigbee-herdsman library.

I am still new to Home Assistant, ZigBee and zigbee2mqtt. But I would expect 'develcoPulseConfiguration' to show up in the dev console for the device under 'seMetering'. But it did not show up. So I am guessing this new config still needs to be mentioned in the zigbee2mqtt codebase. I tried looking through the codebase, but didn't see anything obvious where this would make sense.

@robertalexa thanks for the help on this even though you don't even own the device. Do you maybe have the time to point me in the right direction so that I could try my luck on it?

robertalexa, 2021-09-15:
Hey guys, sorry I have been absent, I have not forgot about this, life has just kept me busier than anticipated.

You will basically need to try this with an external converter. In your converter you will will to provide the functions to handle those commands, just like fz. and tz. do, only that you first try it as functions in your converter.

One good example of custom commands can be found for the Danfoss Ally thermostat (that is the first device that comes to mind). Have a look at that device, see if you can figure it out. I will try to find some time ASAP to provide an external converter for you, in case you don't figure it out before me.

pascalschwientek, 2021-09-15:
Thanks! I tried, and I think I am on the right track. Either the data regarding the pulse configuration is not there or I am not actually on the right track.

```javascript
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const constants = require('zigbee-herdsman-converters/lib/constants');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const options = {manufacturerCode: 0x1015};

dpcTZ = {
    key: ['pulse_configuration'],
    convertSet: async (entity, key, value, meta) => {
        await entity.write('seMetering', {'develcoPulseConfiguration': value}, options);
        return {readAfterWriteTime: 200, state: {'thermostat_vertical_orientation': value}};
    },
    convertGet: async (entity, key, meta) => {
        await entity.read('seMetering', ['develcoPulseConfiguration'], options);
    },
};

const definition = {
    zigbeeModel: ['ZHEMI101'],
    model: 'ZHEMI101',
    vendor: 'Develco',
        description: 'Energy meter',
    fromZigbee: [fz.metering, dpcTZ],
    toZigbee: [dpcTZ],
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(2);

        await reporting.bind(endpoint, coordinatorEndpoint, ['seMetering']);
        await reporting.instantaneousDemand(endpoint);
        await reporting.readMeteringMultiplierDivisor(endpoint);
    },
    exposes: [e.power(), e.energy(),
        exposes.numeric('pulse_configuration', ea.ALL)
        .withDescription('Pulses per kwh')
    ],
};

module.exports = definition;
```

Log message:
`Error Publish 'get' 'pulse_configuration' to 'Energy meter' failed: 'Error: Read 0x0015bc001b021f57/1 seMetering(["develcoPulseConfiguration"], {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4117,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 41423 - 1 - 5 - 1794 - 1 after 10000ms)'`

I have also seen this debug message, which makes me believe that there is no pulse in the data yet.

`Received Zigbee message from 'Energy meter', type 'attributeReport', cluster 'seMetering', data '{"currentSummDelivered":[0,85789],"currentTier1SummDelivered":[65535,4294967295],"currentTier2SummDelivered":[65535,4294967295],"instantaneousDemand":3798,"status":0}' from endpoint 2 with groupID 0`

lanky8804, 2021-09-15:
> Just in case you want to have some sensors that are more accurate in the meantime I have created these:
> 
> template:
> 
> * sensor:
>   
>   * name: "Electricity Live Usage"
>     unit_of_measurement: W
>     state_class: measurement
>     device_class: power
>     state: "{{ (states('sensor.develco_energy_meter_power')|float * 1000 / 3200) | round(2) }}"
>   * name: "Electricity Usage Total"
>     unit_of_measurement: kWh
>     state_class: measurement
>     device_class: energy
>     state: "{{ (states('sensor.develco_energy_meter_energy')|float * 1000 / 3200) | round(2) }}"
> 
> I can't seem to get them to work in the energy module but they are ok for a dashboard.

Thank you this is a very nice temp fix

robertalexa, 2021-09-15:
@pascalschwientek Hey, i just tried to put sth together, but I have no way of testing. Try to use this code and tell me what happens:

```javascript
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const fzLocal = {
    develco_metering: {
        cluster: 'seMetering',
        type: ['attributeReport', 'readResponse'],
        convert: (model, msg, publish, options, meta) => {
            const result = {};
            if (msg.data.hasOwnProperty('develcoPulseConfiguration')) {
                result[postfixWithEndpointName('pulse_configuration', msg, model)] =
                    msg.data['develcoPulseConfiguration'];
            }
            return result;
        },
    },
}

const tzLocal = {
    develco_pulse_configuration: {
        key: ['pulse_configuration'],
        convertSet: async (entity, key, value, meta) => {
            await entity.write('seMetering', {'develcoPulseConfiguration': value}, 0x1015);
            return {readAfterWriteTime: 200, state: {'pulse_configuration': value}};
        },
        convertGet: async (entity, key, meta) => {
            await entity.read('seMetering', ['develcoPulseConfiguration'], 0x1015);
        },
    },
}

const definition = {
    zigbeeModel: ['ZHEMI101'],
    model: 'ZHEMI101',
    vendor: 'Develco',
    description: 'Energy meter',
    fromZigbee: [fz.metering, fzLocal.develco_metering],
    toZigbee: [tzLocal.develco_pulse_configuration],
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(2);
        await reporting.bind(endpoint, coordinatorEndpoint, ['seMetering']);
        await reporting.instantaneousDemand(endpoint);
        await reporting.readMeteringMultiplierDivisor(endpoint);
    },
    exposes: [e.power(), e.energy(),
    exposes.numeric('pulse_configuration', ea.ALL)
        .withDescription('Pulses per kwh')],
};

module.exports = definition;
```

pascalschwientek, 2021-09-15:
Same Error message:

`Received MQTT message on 'zigbee2mqtt/Energy meter/set' with data '{"pulse_configuration":10000}'`

`Publish 'set' 'pulse_configuration' to 'Energy meter' failed: 'Error: Write 0x0015bc001b021f57/1 seMetering({"develcoPulseConfiguration":10000}, {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 41423 - 1 - 124 - 1794 - 4 after 10000ms)'`

I am running version 1.21.1-dev commit: eca5492
which has the develcoPulseConfiguration definition included.

robertalexa, 2021-09-15:
my bad with the manufacturer code!

Can you replace `0x1015` with `{manufacturerCode: 0x1015}` on both lines that use it

[image]

Let me know how it goes

pascalschwientek, 2021-09-15:
No success, same error messages. 

I am wondering, this is the data payload we get from the device according to the logs:
`{"currentSummDelivered":[0,103522],"currentTier1SummDelivered":[65535,4294967295],"currentTier2SummDelivered":[65535,4294967295],"instantaneousDemand":6930,"status":0}`

shouldn't the object contain develcoPulseConfiguration?

pascalschwientek, 2021-09-15:
Btw, the device got re-branded. It's not called "frient". Might this have also caused a new manufacturer code? But then it wouldn't make sense that we get all other data perfectly fine from the device.

https://zigbee.blakadder.com/Develco_ZHEMI101.html

robertalexa, 2021-09-15:
Just to double check sth @pascalschwientek 

You are setting this up using an external converter? If so, in your database.db file, edit your device model number. For example add `-test` to it. Then in the external converter change this line `model: 'ZHEMI101'` to the new name.

This is to make sure that only the external converter code is running and not the current implementation.

The values you are seeing are most likely coming from the device Reporting. For the pulse_configuration attribute there would not be any reporting, just manual GET and SET (when needed, probably just once in a household).

Once you do the above, can you run a GET command and also a SET command and post the error message please (if you get one.) I want to make sure the correct manufacturer is passed through etc.

robertalexa, 2021-09-15:
@pascalschwientek This has gotten me puzzled. I can't spot anything wrong in what we did.... From the other people following this thread, is anyone else able to test this using the dev branch? (so that the latest zigbe--herdsman is used)

@sjorge or @Koenkk Can you guys have a glance at this and tell me if I am missing something obvious? Thanks!

sjorge, 2021-09-15:
Timeout seems to indicate the device is not responding at all I'm guessing.

robertalexa, 2021-09-15:
Those were my thoughts exactly, but @pascalschwientek seems to be getting data back from the device about the power consumption, which made me even more confused.

pascalschwientek, 2021-09-15:
```
Info MQTT publish: topic 'zigbee2mqtt/Energy meter', payload '{"energy":116.43,"linkquality":132,"power":6669,"pulse_configuration":null}'
Info MQTT publish: topic 'zigbee2mqtt/Energy meter', payload '{"energy":116.44,"linkquality":132,"power":6974,"pulse_configuration":null}'
Info MQTT publish: topic 'zigbee2mqtt/Energy meter', payload '{"energy":116.45,"linkquality":132,"power":6925,"pulse_configuration":null}'
Info MQTT publish: topic 'zigbee2mqtt/Energy meter', payload '{"energy":116.46,"linkquality":33,"power":7102,"pulse_configuration":null}'
Info MQTT publish: topic 'zigbee2mqtt/Energy meter', payload '{"energy":116.47,"linkquality":39,"power":7206,"pulse_configuration":null}'
Info MQTT publish: topic 'zigbee2mqtt/Energy meter', payload '{"energy":116.48,"linkquality":36,"power":6956,"pulse_configuration":null}'
Error Publish 'set' 'pulse_configuration' to 'Energy meter' failed: 'Error: Write 0x0015bc001b021f57/1 seMetering({"develcoPulseConfiguration":19238}, {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4117,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 41423 - 1 - 16 - 1794 - 4 after 10000ms)'
Info MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'pulse_configuration' to 'Energy meter' failed: 'Error: Write 0x0015bc001b021f57/1 seMetering({\"develcoPulseConfiguration\":19238}, {\"sendWhenActive\":false,\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":true,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":4117,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (Timeout - 41423 - 1 - 16 - 1794 - 4 after 10000ms)'","meta":{"friendly_name":"Energy meter"},"type":"zigbee_publish_error"}'
Info MQTT publish: topic 'zigbee2mqtt/Energy meter', payload '{"energy":116.5,"linkquality":36,"power":7061,"pulse_configuration":null}'
Info MQTT publish: topic 'zigbee2mqtt/Energy meter', payload '{"energy":116.51,"linkquality":33,"power":6887,"pulse_configuration":null}'
Info MQTT publish: topic 'zigbee2mqtt/Energy meter', payload '{"energy":116.52,"linkquality":33,"power":7625,"pulse_configuration":null}'
```

sjorge, 2021-09-15:
Might be that coordinator -> device has a different route than device -> coordinator.

pascalschwientek, 2021-09-15:
How could I check if this might be a problem?

sjorge, 2021-09-15:
You could try power cycling a few nearby router devices and hope it picks a new route, although Koen might have a better answer

robertalexa, 2021-09-15:
@grudyjsy @lanky8804 Can you guys try to test this as well and see what results you are getting? Thanks

lanky8804, 2021-09-15:
@robertalexa If someone can give me instructions on how to do this then can test.

robertalexa, 2021-09-15:
@lanky8804 You will need to use the dev version of z2m, add an external converter (instruction in the documentation) and use the code provided above

pascalschwientek, 2021-09-15:
@lanky8804 how is your setup looking? If you are running HomeAssistant OS with the Hassio Add-On:

Backup config from: Supervisor → Dashboard → Zigbee2mqtt → Configuration
(Uninstall if already installed: Supervisor → Dashboard → Zigbee2mqtt Edge → Uninstall)
Refresh repo: Supervisor → Add-on store → ⋮ → Reload
Re-install: Supervisor → Add-on store → Zigbee2mqtt Edge → Install
Restore config to: Supervisor → Dashboard → Zigbee2mqtt Edge → Configuration

Then inside the zigbee2mqtt folder (where the configuration.yaml is) create a new file called: `ZHEMI101.js ` with the following content:

```javascript
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const fzLocal = {
    develco_metering: {
        cluster: 'seMetering',
        type: ['attributeReport', 'readResponse'],
        convert: (model, msg, publish, options, meta) => {
            const result = {};
            if (msg.data.hasOwnProperty('develcoPulseConfiguration')) {
                result[postfixWithEndpointName('pulse_configuration', msg, model)] =
                    msg.data['develcoPulseConfiguration'];
            }
            return result;
        },
    },
}

const tzLocal = {
    develco_pulse_configuration: {
        key: ['pulse_configuration'],
        convertSet: async (entity, key, value, meta) => {
            await entity.write('seMetering', {'develcoPulseConfiguration': value}, {manufacturerCode: 0x1015});
            return {readAfterWriteTime: 200, state: {'pulse_configuration': value}};
        },
        convertGet: async (entity, key, meta) => {
            await entity.read('seMetering', ['develcoPulseConfiguration'], {manufacturerCode: 0x1015});
        },
    },
}

const definition = {
    zigbeeModel: ['ZHEMI101'],
    model: 'ZHEMI101',
    vendor: 'Develco',
    description: 'Energy meter',
    fromZigbee: [fz.metering, fzLocal.develco_metering],
    toZigbee: [tzLocal.develco_pulse_configuration],
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(2);
        await reporting.bind(endpoint, coordinatorEndpoint, ['seMetering']);
        await reporting.instantaneousDemand(endpoint);
        await reporting.readMeteringMultiplierDivisor(endpoint);
    },
    exposes: [
        e.power(), 
        e.energy(),
        exposes.numeric('pulse_configuration', ea.ALL).withValueMin(0).withValueMax(65535)
            .withDescription('Pulses per kwh. Default 1000 imp/kWh')],
};

module.exports = definition;
```

Now add it to the configuration. Either edit the file or go back to Supervisor → Dashboard → Zigbee2mqtt Edge → Configuration 

```
external_converters:
  - ZHEMI101.js
```

now restart and that should be it.

lanky8804, 2021-09-15:
@pascalschwientek Running HA on a RasPi 4. Just installed the Edge version which is now up and running. Will now follow other instructions (thank you) and get back to everyone

lanky8804, 2021-09-15:
@pascalschwientek @robertalexa ok so done all above, thank you.

Getting this in the log:
`Info MQTT publish: topic 'zigbee2mqtt/Frient - Energy Monitor', payload '{"energy":974.16,"linkquality":26,"power":2431,"pulse_configuration":null}'
Error Publish 'get' 'pulse_configuration' to 'Frient - Energy Monitor' failed: 'Error: Read 0x0015bc001b02246d/1 seMetering(["develcoPulseConfiguration"], {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4117,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 7240 - 1 - 4 - 1794 - 1 after 10000ms)'
Info MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'get' 'pulse_configuration' to 'Frient - Energy Monitor' failed: 'Error: Read 0x0015bc001b02246d/1 seMetering([\"develcoPulseConfiguration\"], {\"sendWhenActive\":false,\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":true,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":4117,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (Timeout - 7240 - 1 - 4 - 1794 - 1 after 10000ms)'","meta":{"friendly_name":"Frient - Energy Monitor"},"type":"zigbee_publish_error"}'
Info MQTT publish: topic 'zigbee2mqtt/Frient - Energy Monitor', payload '{"energy":974.17,"linkquality":26,"power":2423,"pulse_configuration":null}'`

Also when I try and change pulse rate in the device exposes it doesnt save.

robertalexa, 2021-09-15:
@sjorge Same error as @pascalschwientek. So we know it is not connection.

sjorge, 2021-09-15:
Have you tried sending it without the manufacturerCode ? Some ubisys devices refuse some reds too if we specify it... hence there is a null variant too.

robertalexa, 2021-09-15:
Documentation specifically mentions manufacturer code:
[image]

By mistake we did run some commands with manufacturer code null and it still failed: 

> Same Error message:
>
> `Received MQTT message on 'zigbee2mqtt/Energy meter/set' with data '{"pulse_configuration":10000}'`
>
> `Publish 'set' 'pulse_configuration' to 'Energy meter' failed: 'Error: Write [...]

lanky8804, 2021-09-15:
@robertalexa Some of this goes beyond my understanding but could we test a hardcoded pulsecounter value to see if that worked? Rather than the variable one?

How do we find out the version the device is running on? Just thinking as this is rebranded as mentioned before might be an older version??

sjorge, 2021-09-15:
I did notice it is marked as Optional, maybe Friend didn't implement it?

robertalexa, 2021-09-15:
> @robertalexa Some of this goes beyond my understanding but could we test a hardcoded pulsecounter value to see if that worked? Rather than the variable one?

That is what we are trying to do. The value needs to be SET to the device, it does not live in z2m. But as you confirmed, the command fails.

> I did notice it is marked as Optional, maybe Friend didn't implement it?

The documentation manual is for the device, I think Friend just rebranded, but don;t know. Also don't own it :)

pascalschwientek, 2021-09-15:
In the docs they mention SW versions. (I guess software version?)
Is there a way to read the sw version of the hardware?

`The Basic clustercontains a MFG attributewith theSW version.MFG code: 0x1015 [Develco], Attribute id: 0x8000, Value [3 Byte]Example: 0x040601is SW4.6.1`

robertalexa, 2021-09-15:
> Is there a way to read the sw version of the hardware?

Only if you can edit a file on your zigbee-herdsman to include that attribute. That is also manufacturer specific and is currently not present in zigbee-herdsman.

You would needs to edit the cluster file and under `genBasic` you would need to add a command for the attribute, in a similar way to how I added the `develcoPulseConfiguration` attribute

robertalexa, 2021-09-15:
What I am finding strange is that the pulse_configuration attribute was successfully added to deconz and a lot of people already confirmed it working, so we are either doing sth wrong, or we are indeed unlucky and have problems with the device SW
https://github.com/dresden-elektronik/deconz-rest-plugin/pull/4426/commits/3a63eb6799e45b1f11fcdcd25839af4af6c6254d

pascalschwientek, 2021-09-15:
Will try to get the SW version later tonight when I am home.

Very strange. I have asked frient for some technical documentation. I hope they reply with something useful.

sjorge, 2021-09-15:
They are setting both in one payload if I read the code correctly?
```
                   sensor.addItem(DataTypeUInt8, RConfigInterfaceMode)->setValue(1);
                    sensor.addItem(DataTypeUInt16, RConfigPulseConfiguration)->setValue(1000);
```

Maybe they both need to be set or CfgIntfMode needs to be set first?

robertalexa, 2021-09-15:
I have just done a new PR to zigbee-herdsman for this other attribute, maybe they are indeed paired.

robertalexa, 2021-09-15:
[image]

It is also weird that Deconz are setting the value to 1 for interface mode, which would be `Pulse probe –Pulse counting on a gas meter`. Something doesn't add up.

pascalschwientek, 2021-09-15:
From the deconz PR that introduced pulse configuration:
`Also the interface mode had some blemishes... Btw, allowed interface modes are from 1 (should be default, in order according to documentation, to 8).`

robertalexa, 2021-09-15:
Yeah, it isn't actually 1 to 8, they are building a list and assigning values. These are the values as per docs:
[image]

pascalschwientek, 2021-09-16:
Because zigbee2mqtt is in docker, I didn't find a quick way to edit the files. I probably need to edit the docker file first to mount the app folder and then edit from the host machine.

But I also updated to the latest edge version today and tried the interface mode - same story:

`Publish 'get' 'interface_mode' to 'Energy meter' failed: 'Error: Read 0x0015bc001b021f57/1 seMetering(["develcoInterfaceMode"], {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4117,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 41423 - 1 - 49 - 1794 - 1 after 10000ms)'`

`Publish 'set' 'interface_mode' to 'Energy meter' failed: 'Error: Write 0x0015bc001b021f57/1 seMetering({"develcoInterfaceMode":0}, {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4117,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 41423 - 1 - 123 - 1794 - 4 after 10000ms)'`

I wanted to buy a product to just have it work plug and play - should have gone the DIY route with ESPHome 😂

robertalexa, 2021-09-16:
> I wanted to buy a product to just have it work plug and play - should have gone the DIY route with ESPHome

I can understand the frustration. Hopefully we will get to the bottom of it. I have asked Koen to have a look at this and see if he has any thoughts. I think in advance, can you provide the zigbee-herdsman log too when running the commands?

lanky8804, 2021-09-16:
> I wanted to buy a product to just have it work plug and play - should have gone the DIY route with ESPHome 😂

I also looked into this route befor purchasing but powering the device was going to be the issue for me, hense this all in one.

Appriciate the support, hopefully we can resolve the issue

pascalschwientek, 2021-09-16:
[logs.txt](https://github.com/Koenkk/zigbee2mqtt/files/7176930/logs.txt)

I don't know what to look for - so I didn't filter out anything, just the raw data. I did a set for both and a get for both. I hope there isn't too much spam in the log from my other devices.

The Converter:

```javascript
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const fzLocal = {
    develco_metering: {
        cluster: 'seMetering',
        type: ['attributeReport', 'readResponse'],
        convert: (model, msg, publish, options, meta) => {
            const result = {};
            if (msg.data.hasOwnProperty('develcoPulseConfiguration')) {
                result[postfixWithEndpointName('pulse_configuration', msg, model)] =
                    msg.data['develcoPulseConfiguration'];
            }
            if (msg.data.hasOwnProperty('develcoInterfaceMode')) {
                result[postfixWithEndpointName('interface_mode', msg, model)] =
                    msg.data['develcoInterfaceMode'];
            }
            return result;
        },
    },
}

const tzLocal = {
    develco_pulse_configuration: {
        key: ['pulse_configuration'],
        convertSet: async (entity, key, value, meta) => {
            await entity.write('seMetering', {'develcoPulseConfiguration': value}, {manufacturerCode: 0x1015});
            return {readAfterWriteTime: 200, state: {'pulse_configuration': value}};
        },
        convertGet: async (entity, key, meta) => {
            await entity.read('seMetering', ['develcoPulseConfiguration'], {manufacturerCode: 0x1015});
        },
    },
    develco_interface_mode: {
        key: ['interface_mode'],
        convertSet: async (entity, key, value, meta) => {
            await entity.write('seMetering', {'develcoInterfaceMode': value}, {manufacturerCode: 0x1015});
            return {readAfterWriteTime: 200, state: {'interface_mode': value}};
        },
        convertGet: async (entity, key, meta) => {
            await entity.read('seMetering', ['develcoInterfaceMode'], {manufacturerCode: 0x1015});
        },
    },
}

const definition = {
    zigbeeModel: ['ZHEMI101'],
    model: 'ZHEMI101',
    vendor: 'Develco',
    description: 'Energy meter',
    fromZigbee: [fz.metering, fzLocal.develco_metering],
    toZigbee: [tzLocal.develco_pulse_configuration, tzLocal.develco_interface_mode],
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(2);
        await reporting.bind(endpoint, coordinatorEndpoint, ['seMetering']);
        await reporting.instantaneousDemand(endpoint);
        await reporting.readMeteringMultiplierDivisor(endpoint);
    },
    exposes: [
        e.power(), 
        e.energy(),
        exposes.numeric('pulse_configuration', ea.ALL).withValueMin(0).withValueMax(65535)
            .withDescription('Pulses per kwh. Default 1000 imp/kWh'),
        exposes.numeric('interface_mode', ea.ALL).withValueMin(0).withValueMax(65535)
    ],
};

module.exports = definition;
```

pascalschwientek, 2021-09-17:
Has anyone got a new idea on this? Did Koen have time to look at this already? Sadly, I am not getting any response from Frient regarding a current technical documentation.

grudyjsy, 2021-09-17:
Hi, I have tried Frient via social media and email and heard nothing back. I wonder if Develco would be more help as it’s there products that Frient label up as thee own product, https://www.develcoproducts.com/products/meter-interfaces/external-meter-interface/

robertalexa, 2021-09-17:
The documentation we have been following is Develco's
https://www.develcoproducts.com/media/1782/zhemi101-technical-manual-external-meter-interface.pdf

@Koenkk Would you be kind to have a look and see if we are missing anything here? Thanks in advance

Koenkk, 2021-09-18:
Can you try with `await entity.write('seMetering', {'develcoInterfaceMode': value}, {manufacturerCode: 0x1015, sendWhenActive: true});` instead? This should send the write command once the device is awake (e.g. trigger the device to send a message to z2m)

pascalschwientek, 2021-09-19:
Tried it but sadly no change in behavior. Since Frient is not responding, maybe it's worth a shot to purchase the develco product and rule out any firmware change that Frient made that's specific to them.

`Publish 'get' 'pulse_configuration' to 'Energy meter' failed: 'Error: Read 0x0015bc001b021f57/1 seMetering(["develcoPulseConfiguration"], {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4117,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 41423 - 1 - 29 - 1794 - 1 after 10000ms)'
`
`Publish 'set' 'pulse_configuration' to 'Energy meter' failed: 'Error: Write 0x0015bc001b021f57/1 seMetering({"develcoPulseConfiguration":0}, {"sendWhenActive":true,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4117,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 41423 - 1 - 131 - 1794 - 4 after 10000ms)'`

pascalschwientek, 2021-09-20:
I got a reply from Develco:

_We are the owner of the Frient brand._

I would rule out any firmware change. Develco is the B2B whitelabel manufacturer and Frient is their consumer brand.
But they also won't sell a Develco product to me as it's only B2B

zuzuman, 2021-11-09:
Latest Zigbee2Mqtt:
[image]
Can read values, but when trying to change them:
[image]

zuzuman, 2021-11-09:
OK. Done this one

zuzuman, 2021-11-09:
`const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const fzLocal = {
    develco_metering: {
        cluster: 'seMetering',
        type: ['attributeReport', 'readResponse'],
        convert: (model, msg, publish, options, meta) => {
            const result = {};
            if (msg.data.hasOwnProperty('develcoPulseConfiguration')) {
                result[postfixWithEndpointName('Pulse Configuration', msg, model)] =
                    msg.data['develcoPulseConfiguration'];
            }
            return result;
        },
    },
}

const tzLocal = {
    develco_interface_mode: {
        key: ['Pulse Configuration'],
        convertSet: async (entity, key, value, meta) => {

            await entity.write('seMetering', {'develcoPulseConfiguration' : value}, {manufacturerCode: 0x1015});
            return {readAfterWriteTime: 200, state: {'develcoPulseConfiguration': value}};
        },
        convertGet: async (entity, key, meta) => {
            await entity.read('seMetering', ['develcoPulseConfiguration'], {manufacturerCode: 0x1015});
        },
    },
}

const definition = {
    zigbeeModel: ['ZHEMI101'],
    model: 'ZHEMI101',
    vendor: 'Develco',
    description: 'Energy meter',
    fromZigbee: [fz.metering, fzLocal.develco_metering],
    endpoint: (device) => {
    return {'default': 2};
    },
    toZigbee: [tzLocal.develco_interface_mode],
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(2);
        await reporting.bind(endpoint, coordinatorEndpoint, ['seMetering']);
        await reporting.instantaneousDemand(endpoint);
        await reporting.readMeteringMultiplierDivisor(endpoint);
    },
    exposes: [
        e.power(), 
        e.energy(),
        exposes.numeric('Pulse Configuration', ea.ALL).withDescription('Pulses per whole value (m³ or kWh)')
    ],
};

module.exports = definition;
`

Tested with electricity meter (2500 led impulses )
Tested with gas meter (replaced LED with magnetic sensor) (100 magnetic impulses per 1m³)

Do not use "refresh" button. Just provide value inside text box

zuzuman, 2021-11-10:
https://user-images.githubusercontent.com/3532812/141027383-4fa6b10b-934f-4841-927f-af4375fe776c.mov

lanky8804, 2021-11-10:
@zuzuman just tired this also with Edge version and 3200 pulse and seem to be working, thank you.

[29 more replies]

closed 2021-11-12 by lanky8804