Koenkk/zigbee2mqtt #8648: The sirene Neo NAS-AB02B0 reset the duration time to zero/null

SargonofAssyria, 2021-09-05:
This sirene https://www.zigbee2mqtt.io/devices/NAS-AB02B0.html#neo-nas-ab02b0 exposes the duration time.
With the latest beta update 1.21.0-dev commit: d92bdf1c the device resets the duration time to 0 (zero), null in the message
`info  2021-09-05 07:57:06: MQTT publish: topic 'zigbee2mqtt/Sirene', payload '{"alarm":false,"battery_low":false,"duration":10,"humidity":42,"humidity_alarm":false,"humidity_max":90,"humidity_min":10,"last_seen":"2021-09-05T07:57:06+02:00","linkquality":72,"melody":14,"power_type":"usb","temperature":"26.0","temperature_alarm":false,"temperature_max":30,"temperature_min":11,"volume":"low"}'
info  2021-09-05 07:57:11: MQTT publish: topic 'zigbee2mqtt/Sirene', payload '{"alarm":false,"battery_low":false,"duration":null,"humidity":42,"humidity_alarm":false,"humidity_max":90,"humidity_min":10,"last_seen":"2021-09-05T07:57:11+02:00","linkquality":66,"melody":14,"power_type":"usb","temperature":"26.0","temperature_alarm":false,"temperature_max":30,"temperature_min":11,"volume":"low"}'
`
Previous it used to stay for ever on the set value.

SargonofAssyria, 2021-11-25:
I am on:
Zigbee2MQTT version 1.22.0-dev commit: ad3ff7fa
Coordinator type zStack3x0
Coordinator revision 20210708
Frontend version 0.6.36

Problem still exists.

closed 2021-12-03 by github-actions[bot]

KameDomotics, 2023-08-11:
Hi.
I have same problem with this product Neo NAS-AB02B0.
How is it possible to solve the problem? Thank you