Koenkk/zigbee2mqtt #8654: Extremely bad signal with slae.sh CC2652RB stick

Marmoset-threat, 2021-09-05:
I'm getting very, very bad signal with with my CC2652RB stick from slae.sh and I'm beginning to suspect that the hardware is broken or I've done something fundamentally wrong. Does anyone have any tips?

By bad signal I mean that a temp/humidity sensor within 2 meters that has line-of-sight to the antenna on the stick gets a linkquality of like 20-30.  A plug-in-repeater _in the same room_ and only three meters away doesn't connect at all (or when it does, only shows up with a linkquality of zero). I've tried all sorts of things like:

- Changed all my Ubiquiti AP's to channels 1 and 6
- Tried different 2.4 ghz antennas on the stick (the standard 3dbi one and a 6dbi one I had laying about)
- Ran the stick into a different room with a USB extension cable
- Purchased more plug in routers (I have three ikea ones and four osram-offbrands) 
- Updated Zigbee2Mqtt to version 1.21.0

Yet nothing helps. Even with the repeaters, I can just _barely_ get the signal to cover the room that the coordinator antenna is in, let alone go through walls. 

Is my stick broken? Do I have some undiscovered yet extremely powerful source of 2.4 ghz noise coming from somewhere? Is my house simply cursed? Should a buy a heavily shielded USB extension cable and a CC2652P dongle?

Any help is greatly appreciated.

Koenkk, 2021-09-05:
Do you have another adapter to compare with? (e.g. a CC2531)

pro2call-nl, 2021-09-05:
I have the same issue. looks like it started after the update core from 2021-9.2 to 2021-9.3. 

in mqtt the command from HA is directly visibel. lights are sometimes unresponsive or very slow

Shred99, 2021-09-06:
Try a different aerial.  I've found some of the external aerials / antennas that I've accumulated over the years are no good.

Koenkk, 2021-09-06:
@pro2call-nl that looks like https://github.com/Koenkk/zigbee2mqtt/issues/8623 , see https://github.com/Koenkk/zigbee2mqtt/issues/8623#issuecomment-913376139

Marmoset-threat, 2021-09-06:
@Koenkk Unfortunately, no. This is the first stick I bought. Thinking about USB sticks though encouraged me to crack open the case and take a picture, and looking closer reveals that one of the pins on the SMA connector is missing solder. Could this be the cause? It looks like it's one of the ground pins, so I'm not sure. 

[image]

Koenkk, 2021-09-06:
Looks like it, mine has solder on all pins.

[image]

wizu, 2021-09-06:
I tried to change adapter from CC2538 to Slae.sh CC2652RB but had problems with it too.
On my adapter, that leg is also not soldered and I asked Slae.sh about it and he said that it is not neccessary to solder that leg.

mercenaruss, 2021-09-06:
> @Koenkk Unfortunately, no. This is the first stick I bought.
>
> Thinking about USB sticks though encouraged me to crack open the case and take a picture, and looking closer reveals that one of the [...]

Check for short circuit between ground and signal, but remove antenna when doing this. I had some issue like this recently. Used solder iron to fix it.

Marmoset-threat, 2021-09-06:
@mercenaruss Alright, I measured and there's definitely a short between the signal and ground. The multimeter give me about 0.2 ish ohms, so basically. 

I de-soldered the SMA plug (which was a bit tricky), and measured from the pads, but the problem's still there. How were you able to fix it?

Even looking under the microscope and washing the whole thing in alcohol hasn't made a difference. Here are some pictures, maybe someone can spot a bridge:

<details>
<summary>Click here to show images</summary>

[image]
[image]
[image]

</details>

mercenaruss, 2021-09-06:
@Marmoset-threat Unable to help you, i saw this on my own designed stick, had some short circuit on antenna connector. Fixed by removing excessive tin. Contact @slaesh

djtrue, 2021-09-06:
Could there be a problem with the stick power supply? When it works at the input is there a stable 5V?

Marmoset-threat, 2021-09-06:
@djtrue I'd assume so since I'm running it on a pretty decent Intel NUC which has never had any problems with power supply. 

I'm beginning to think that the stick is just broken somehow, and I'm going to order a different one to see.

wizu, 2021-09-07:
I had problems like youts with Slae.sh stick ordered few months ago. Yesterday I've installed CC2652P and it's working great. W ill try to use Slae.sh stick as router and then I will be able to tell if mine is broken too.

closed 2021-10-15 by github-actions[bot]