Koenkk/zigbee2mqtt #8655: Problem at restart of Zigbee2MQTT with LQI

SargonofAssyria, 2021-09-05:
I am using:
Zigbee2MQTT version 1.21.0-dev commit: d92bdf1c
Coordinator type zStack3x0 20210708
Coordinator revision 20210708
Frontend version 0.6.0
Adapter LAUNCHXL-CC1352P-2
Domoticz version 2021.1 build 13360

When ever I have to restart zigbee2mqqt a few (sometimes a lot) of the device have an LQI of null.
Those device are not present anymore in Domoticz and cause a lot of errors because they are used in dzVents.
If I manually or by the Zigbee2MQTT GUI activate they come to live and function normally again.
Some of them (like a doorswitch) have to be included again for them to function normally.
Am I doing something wrong? If so, what. Is there a general solution?

bossgos, 2021-09-06:
Same here.. LQI entity dissapiring

closed 2021-10-14 by github-actions[bot]

ChrisLane, 2022-01-12:
@SargonofAssyria I'm having the same issue and it really doesn't make my home feel very "smart" when nothing works after a power outage.
How can I activate the devices via the GUI at least?

SargonofAssyria, 2022-01-12:
If you click on the friendlyname and click on state the the devices is activated and gets its LQI.
For other devices you just have to wait until the device comes to live.