Koenkk/zigbee2mqtt #8656: Inability to pair and inadequate network behavior after firmware CC2652RB 20210708

djtrue, 2021-09-05:
Zigbee2MQTT version  1.21.1-dev commit: 4125ae0
Coordinator type zStack3x0 20210708
Coordinator revision 20210708
Frontend version 0.6.0
Adapter CC2652RB  slae.sh 
hassio z2m add-on -dev

Serious problems occurred after updating the CC2652RB stick firmware to the latest version. Guided by the comments to the previous release of z2m, I started reconnecting all devices ( 90 pieces about 20 are routers), changed the pan_id, key and folder for storing z2m files. I connected first routers from near to far and then end devices . As a result, at least 7 routers do not perform their function and it is impossible to connect to them. A day later, several more routers do not allow pairing through themselves. I noticed that after reconnecting the stack, the devices themselves can disappear from the list of added ones or change the LQI to 0 and stop responding. There are several routers that never refuse to pair.

I started again 4 times using different sequences when connecting . I noticed that when the xiaomi relay is paired, it starts blinking slowly, as if they lose connection at this moment.
It is difficult to understand the moment of router failure, so logging is problematic .

There are 2 devices that never allow pairing through themselves:
1. Zemismart ZM-CSW032-D
2. Xiaomi JTQJ-BF-01LM/BW

Koenkk, 2021-09-06:
Not sure about the ZM-CSW032-D, but I expect the JTQJ-BF-01LM/BW to be an end device (since it's battery powered) so it's expected that no devices can be paired through it.

djtrue, 2021-09-06:
@Koenkk, thank you for your attention .
Both of these devices are displayed as routers and have a constant power supply.
Let me share my observations.

1) I conducted an experiment - I removed and added JTQJ-BF-01LM / BW right at the coordinator's antenna . After that, I was able to connect through this router. But as soon as I returned it to the old place ( located at a distance through several routers that do not have connection problems) and the ability to pair again disappears.

2) Zemismart ZM-CSW032-D tried to connect through 5 working routers in the reception area . The pairing of this router is successful. But he himself is not able to accept other devices. ( previously, there were 20 end devices hanging on it)

3) Also, several times starting anew, moving in different directions, I noticed that some routers begin to lose the ability to pair, as if the limit on the number of routers has been exceeded ( there are 26 of them now ) or

4) Yesterday, after restarting z2m, one of the routers stopped accepting interfaces ,I deleted it and added a new one , and all the end devices that communicated through it received lqi 0 and stopped communicating .

5) I also noticed one more thing - I don't have a coordinator in the list of routers and I can't select only him for connection . Is this normal? I remember the coordinator stopped being mentioned in the lists as soon as a year ago I switched to the CC2652RB stick. I always thought that it was specially removed in some of the updates.

6) When deleting routers and connecting again, a notification is written in the z2m interface that the device has left the network . It warns that this message comes up to 4 times in a row.

7) When you try to connect the router to another router that can not pair, the message comes the device is out of the network, but there is no further connection.

The prehistory is this - everything worked perfectly on CC2652RB
then I updated the firmware to 20210120 . Problems began. The next day, the firmware 20210708 was released after installation, I reconnected all the routers. but they also began to lose the ability to mate. After that, I started with a clean slate , but came to the same result ( I completely deleted z2m with all its folders and flashed the coordinator)

Koenkk, 2021-09-07:
Could you provide a sniff when trying to pair a device through the JTQJ-BF-01LM/BW? https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

djtrue, 2021-09-07:
@Koenkk Unfortunately, I don't have a second adapter.
I have rebooted z2m several times today . I re-downloaded the adapter firmware. And many times I paired JTQJ-BF-01LM / BW next to the coordinator. As a result, we somehow managed to connect them and not lose the ability to mate at a distance .

With ZM-CSW032-D, all these manipulations did not help . Even if you create a new network with only one of these devices, pairing with another pan-id does not occur through it.

I still have about 7 different routers that behave similarly . Also, after the adapter is de-energized, about 10-15 different end devices lose the lqi 0 connection and no longer respond until the pairing.

The main question is I don't have a coordinator in the list of devices . How can I add it there? To try to reconnect all routers by specifying exactly the coordinator.

Koenkk, 2021-09-07:
To investigate such issues further we really need to sniff to figure out what exactly is happening and to identify where it fails.

> The main question is I don't have a coordinator in the list of devices . How can I add it there? To try to reconnect all routers by specifying exactly the coordinator.

Do you mean the permit joining list on the frontend? If so I see it is indeed not included there. You should be able to do this via  MQTT (https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttbridgerequestpermit_join), use `coordinator` as the `device` name

GeorgeCastanza, 2021-09-16:
I too am having serious difficulty (read: not able to) pairing end devices with same circumstances as OP.
Zigbee2MQTT version 1.21.1
Running on bare metal
Adapter CC2652RB slae.sh
Coordinator type zStack3x0 20210708

GeorgeCastanza, 2021-09-16:
I downgraded my zstack firmware, and was able to pair devices again (although it did take a number of attempts before pairing). 

Coordinator type zStack 20210112 works for me

nickw444, 2021-09-20:
I think I might be having this problem, and also might be related to https://github.com/Koenkk/zigbee2mqtt/issues/8430. I raised a ticket here, https://github.com/Koenkk/zigbee2mqtt/issues/8803, and included a zip file with a handful of packet sniffs whilst attempting to pair. 

I am running a CC2562P2 coordinator, with the latest firmware. Consistently (but intermittently) reproducible once the network gains complexity with a number of routers.

GeorgeCastanza, 2021-11-20:
Not stale. Still issues.

djtrue, 2021-12-09:
> Not stale. Still issues.

I seem to have solved the problem. For about a month now, all routers do not lose the ability to pair.
My actions:
1) removed all routers forcibly (yes yes again ...)
2) rebooted z2m
3) Updated the stick firmware to ver. 20211114 (DEV branch in coordinator's firmware)
4) Added all routers again + several end devices that stopped responding

closed 2021-12-09 by djtrue