Koenkk/zigbee2mqtt #8660: Network doesn’t respond when upgrading from 1.19.1 to 1.21.0

rcetrulo, 2021-09-06:
# Bug Report
## What happened
I had been using 1.19.1 with Docker and have a working network with 30+ devices and a slaeh zstack 3 coordinator. When I upgraded to 1.21.0 the devices stopped working. Downgrading to my previous version solves the issue. Maybe I just missed something obvious?

## What did you expect to happen
I expected my devices to still respond and work, since my data folder is still there. 

## How to reproduce it (minimal and precise)
Upgrade your working docker z2m from 1.19.1 to 1.21.0 and the fun begins. Downgrade and everything goes back to working state. 

## Debug Info
Zigbee2MQTT version:
Adapter hardware: CC2652RB
Adapter firmware version: 20210708

rcetrulo, 2021-09-06:
I kept getting LQI errors and messages not delivered errors. It’s pretty obvious something is broken with the zigbee network with the new version.

Koenkk, 2021-09-06:
Could you check if you are affected by https://github.com/Koenkk/zigbee2mqtt/issues/8663 ?

rcetrulo, 2021-09-06:
My slaeh cc2652rb coordinator uses a SiLabs UART, CP2102, not the CH341. Unless the driver issue also applies to SiLabs…

Koenkk, 2021-09-06:
could you provide the herdsman debug log of the issue?

See https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging on how to enable the herdsman debug logging. Note that this is only logged to STDOUT and not to log files.

spider7611, 2021-09-09:
For me have exactly the same problem. I use cc2530 coordinator. Using 1.19.1 with Docker and everything is working good, but if I upgrade later versions have problem. After a while it stopped working and must disconnect manually from the usb the coordinator and reconnect it. Before never see this problem. Tryd 3 times upgrade and downgrade again and agin the same problem. With 1.19.1 working good but with the newer version stopped working some hours after.

spider7611, 2021-09-10:
[Zigbee2MQTT (1).csv](https://github.com/Koenkk/zigbee2mqtt/files/7145226/Zigbee2MQTT.1.csv)
Here it is the debug

spider7611, 2021-09-16:
> could you provide the herdsman debug log of the issue?
> 
> See https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging on how to enable the herdsman debug logging. Note that this is only logged to STDOUT and not to log files.

Any news about this problem?

spider7611, 2021-09-17:
[Zigbee2MQTT (2).csv](https://github.com/Koenkk/zigbee2mqtt/files/7187200/Zigbee2MQTT.2.csv)

Another debug. Worked almost 1 day. Now time to move back to 1.19.1 again :(

Koenkk, 2021-09-18:
@spider7611 I believe it is connected to #8282 , I'm currently working on fixing it.