Koenkk/zigbee2mqtt #8661: TuYa TS0044 device being recognised as TS004F

lunzai, 2021-09-06:
I've purchased 2 4gang wireless switches, product advertise with support for single click, double click, hold click for each of the 4 buttons like TS0044. However, the interview process recognised both devices as TS004F.

[image]

Device on the left, brand is MOES, model is ESW-0ZAA-EU, which also matches the model for Haozee ([here](https://www.zigbee2mqtt.io/devices/TS0044.html)).

Device on the right, no brand, model is RSH-ZigBee-SC04. 

Is it possible for me to define / force the z2m to assign TS0044's converter config for my devices? Will this be considered for new feature? Also, what can I do on my end of fix this issue?

## data/database.db entry of the device
**For MOES Model ESW-0ZAA-EU**
`{"id":38,"type":"EndDevice","ieeeAddr":"0x60a423fffefde5d0","nwkAddr":11994,"manufId":4098,"manufName":"_TZ3000_xabckq1v","powerSource":"Battery","modelId":"TS004F","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":260,"inClusterList":[0,1,3,4,6,4096],"outClusterList":[25,10,3,4,5,6,8,4096],"clusters":{"genBasic":{"attributes":{"modelId":"TS004F","manufacturerName":"_TZ3000_xabckq1v","powerSource":3,"zclVersion":3,"appVersion":65,"stackVersion":0,"hwVersion":1,"dateCode":""}},"genPowerCfg":{"attributes":{"batteryPercentageRemaining":200}}},"binds":[{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x00124b00238d7f69","endpointID":1}],"configuredReportings":[{"cluster":1,"attrId":33,"minRepIntval":3600,"maxRepIntval":62000,"repChange":0}],"meta":{}}},"appVersion":65,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":-1217940983},"lastSeen":1630896990469}`

**For OEM no branding, Model RSH-ZigBee-SC04**
`{"id":33,"type":"EndDevice","ieeeAddr":"0x60a423fffedb2e16","nwkAddr":30442,"manufId":4098,"manufName":"_TZ3000_xabckq1v","powerSource":"Battery","modelId":"TS004F","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":260,"inClusterList":[0,1,3,4,6,4096],"outClusterList":[25,10,3,4,5,6,8,4096],"clusters":{"genBasic":{"attributes":{"modelId":"TS004F","manufacturerName":"_TZ3000_xabckq1v","powerSource":3,"zclVersion":3,"appVersion":65,"stackVersion":0,"hwVersion":1,"dateCode":""}},"genPowerCfg":{"attributes":{"batteryPercentageRemaining":172}}},"binds":[{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x00124b00238d7f69","endpointID":1}],"configuredReportings":[{"cluster":1,"attrId":33,"minRepIntval":3600,"maxRepIntval":62000,"repChange":0}],"meta":{}}},"appVersion":65,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":-1217940983},"lastSeen":1630897359848}`

danieledwardgeorgehitchcock, 2021-09-06:
The model id of both of these devices is being reported as TS004F so this is the correct behaviour for Zigbee2MQTT.

As for changing the device handler to get a different featureset, unfortunately, Zigbee doesn't work that way.

The actions types are being reported back from the device, and Zigbee2MQTT is just converting them to a human readable format.

If for example, double click isn't supported at the device level, it won't be supported at Zigbee2MQTT level either.

lunzai, 2021-09-06:
> The model id of both of these devices is being reported as TS004F at interview by the devices themselves so, this is the correct behaviour for Zigbee2MQTT.
> 
> As for changing the device handler to get a different featureset, unfortunately, Zigbee doesn't work that way.
> 
> The action types are being reported back from the device, and Zigbee2MQTT is just converting them to a human readable format.
> 
> If for example, double click isn't supported at the device level, it won't be supported at Zigbee2MQTT level either.

Is there anything I can do on my end to allow my devices to work like a TS0044?

danieledwardgeorgehitchcock, 2021-09-06:
Not really - you could possibly fudged together an automation in something like Home Assistant but ultimately, it would probably be easier to get the devices that behave as you require to begin with

frankroeske, 2021-09-14:
There is an ongoing discussion. The device seems to support two different modes, when using the tuya gateway:
https://github.com/Koenkk/zigbee2mqtt/discussions/7158

closed 2021-10-22 by github-actions[bot]