Koenkk/zigbee2mqtt #8663: CH341 driver bugged in recent Linux kernel versions

Koenkk, 2021-09-06:
Recent Linux kernel versions contain a bug in the [CH341 driver](https://bugzilla.kernel.org/show_bug.cgi?id=214131). The CH341 controller is used on various coordinators, e.g. the zzh. This causes strange delays when receiving messages from devices.

## How do I know I'm affected?
Your coordinator uses a CH340/CH341 controller (e.g. zzh) AND you are running one of the kernel following kernel version (run `uname -r` to check which):
- 5.13.*: 5.13.10 till 5.13.13 are affected, fixed in 5.13.14
- 5.10.*: 5.10.58 till 5.10.61 are affected, fixed in 5.10.62
- 5.4.*: 5.4.140 till 5.4.143 are affected, fixed in 5.4.144
- I did not check other kernel version

## Workaround
- Downgrade your kernel version
- For hassos users, this will be fixed in [6.4](https://github.com/home-assistant/operating-system/issues/1532#issuecomment-913440970), for now dowgrande to 6.2 using: `ha os update --version 6.2`

## Related issues:
- https://github.com/home-assistant/operating-system/issues/1532
- https://github.com/Koenkk/zigbee2mqtt/issues/8623
- https://github.com/home-assistant/operating-system/issues/1537
- https://github.com/Koenkk/zigbee2mqtt/issues/8552

bossgos, 2021-09-06:
Thanks :) it`s saves my network ;) I was made downgrade to OS v6.2

bossgos, 2021-09-07:
Today HASS OS 6.4 has been released. Is it still affected with this bug?

netweaver1970, 2021-09-07:
no, will be fixed.

Cippo2019, 2021-09-08:
> no, will be fixed.

only for information , do you know more or less the time windows ?
to understand if it is more like 1 months or 1 week , for example..

I know, this in not up to you but maybe you know something more :-)
wbr
Tz

bruvv, 2021-09-09:
Hassos 6.4 has been released and this is now fixed.

okastl, 2021-09-09:
@Koenkk are you sure, that only CH341 is affected? I have Slash's stick with a CP2102 and it behaved "sluggish" with the recent kernel (5.10.60) on a Raspberry PI 3B. E.g. OTA updates always failed. Sonoff ZBMINI switched with noticeable delay.
I downgraded the Kernel to 5.10.52, and OTA updates work now. ZBMINI behaves better.
Just a coincidence, or a related problem?

Koenkk, 2021-09-10:
@okastl CP2102 uses a different driver so its not affected by the CH341 driver bug for sure (but maybe the CP2102 also has a bug, will keep this in mind for slaesh users)

pedrolamas, 2021-09-11:
Quick update on this, I've just updated the kernel in my NanoPi M4V2 to 5.10.63 and the problem is not happening anymore, thus confirming that this was indeed a kernel issue.

barrymossel, 2021-09-19:
I just updated to 6.4 and Zigbee2MQTT wasn't working anymore at all...

okastl, 2021-09-19:
> I just updated to 6.4 and Zigbee2MQTT wasn't working anymore at all...
Which adapter are you using?

barrymossel, 2021-09-19:
ZZH. I just reverted to 6.2 and everything works as it should...

RaffoNin, 2021-09-24:
I can confirm on my raspberry pi 3 that 5.10.63 is still buggy. Issue was resolved when i downgraded all the way to 5.10.52.

ondrejbaranek, 2021-09-25:
rpi-update from commit b4e395b3e87dba4964f314e12871630cabb35f70 fixed the issue (5.10.63-v7+)

piit79, 2021-09-28:
Ah, and I was wondering what was going on when I upgraded to OpenWrt 21.02 (I'm using zzh! over TCP using ser2net) - suddenly messages from Xiaomi switches (and perhaps some other sensors) wouldn't be delivered instantly, but only with another message from a different device. OpenWrt 21.02.0 has kernel 5.4.143. Let's hope 21.02.1 is published soon, it will have at least 5.4.145.

daverobertson63, 2021-09-28:
I found same issue with latest rpi kernel so I tried to use the latest Linux drivers from vendor. 

https://github.com/daverobertson63/CH341SER

My solution was to replace the current driver ch341.ko with ch34x.Ko driver. I only tested on rpi and udoo quad - running Ubuntu 14.

Koenkk, 2021-09-30:
@daverobertson63 thanks, seems that this bug also got in the latest ubuntu 20.04 kernel (update OP with version numbers). With your driver everything works perfectly!

Arkadi85, 2021-10-02:
On hassio 6.4 the problem is solved, tested will all aqara sensors.

daverobertson63, 2021-10-15:
> > I found same issue with latest rpi kernel so I tried to use the latest Linux drivers from vendor.
> > https://github.com/daverobertson63/CH341SER
> > My solution was to replace the current driver ch341.ko with ch34x.Ko driver. I only tested on rpi and udoo quad - running Ubuntu 14.
> 
> Hi @daverobertson63 how do I update the driver in Ubuntu? things are not working for me and I really need to resolve this issue. Can you please help? thanks

I followed this to get the headers onto the Linux machine 

https://www.tecmint.com/install-kernel-headers-in-ubuntu-and-debian/

Thats the most important part - when you do a make and sudo make load in the driver code - it should just work - if it fails on the make - you probably dont have the headers or gcc installed

Then just follow the instructions.  I noted that some drivers are gzip and some not - you can figure that out by looking at whats there.   

Make sure you do the depmod - as that creates the kernal boot map for the devices - its more or less the same I did for the pi.  I only tested in Ubuntu 14 - so things may have changed -but the make and sudo make load should work on pretty much anything

bachoo786, 2021-10-15:
> Quick update on this, I've just updated the kernel in my NanoPi M4V2 to 5.10.63 and the problem is not happening anymore, thus confirming that this was indeed a kernel issue.

Hi @pedrolamas how did you update to 5.10.63? I am trying my best to find information online but I cannot find anything. Is there a such thing how to downgrade kernel in ubuntu?

pannal, 2021-10-18:
@daverobertson63 your driver works great with `5.11.0-37-generic #41~20.04.2-Ubuntu`. Would it be possible to make it a dkms module? Thanks!

matejdro, 2021-11-10:
Thanks, upgrading kernel helped!

For anyone on Ubuntu, to install new kernel (before it is released into mainline repo), you can follow instructions under `Install Howto` on this page: https://ubuntu.pkgs.org/20.04/ubuntu-proposed-main-amd64/linux-image-5.4.0-90-generic_5.4.0-90.101_amd64.deb.html

However, you might also want to install `sudo apt-get install linux-modules-extra-5.4.0-90-generic` before rebooting, otherwise some devices (such as network card) might not work.

pannal, 2021-12-13:
Anyone aware if this is fixed in 5.11.0-41? (hwe on Ubuntu 20.04.3 LTS)

pannal, 2021-12-17:
Can't confirm 5.11.0-41 to be much better. The delay seems to be gone, but at times, especially when you rely on multiple zigbee devices to react at the same time, it's still a mess.

Running the CH341X driver as DKMS module still works better.

klim123123, 2021-12-21:
I'm testing 5.10.83 for 24 hours already. Looks like problem has been fixed.
Also tried to use multiple zigbee devices at the same time, no delays, no problems.

closed 2022-01-28 by github-actions[bot]

joshuakoh1, 2022-03-23:
I'm on 20.04 LTS and 5.4.0-104-generic. My zzh stick keeps timing out after a day. Is this the same issue? I can't fix it without a complete reboot.

```
Mar 23 15:46:42 OpenHAB3 npm[895]: (node:895) UnhandledPromiseRejectionWarning: Error: SRSP - ZDO - mgmtPermitJoinReq after 6000ms
Mar 23 15:46:42 OpenHAB3 npm[895]:     at Timeout._onTimeout (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35)
Mar 23 15:46:42 OpenHAB3 npm[895]:     at listOnTimeout (internal/timers.js:554:17)
Mar 23 15:46:42 OpenHAB3 npm[895]:     at processTimers (internal/timers.js:497:7)
Mar 23 15:46:42 OpenHAB3 npm[895]: (node:895) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was>
Mar 23 15:50:02 OpenHAB3 npm[895]: (node:895) UnhandledPromiseRejectionWarning: Error: SRSP - ZDO - mgmtPermitJoinReq after 6000ms
Mar 23 15:50:02 OpenHAB3 npm[895]:     at Timeout._onTimeout (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35)
Mar 23 15:50:02 OpenHAB3 npm[895]:     at listOnTimeout (internal/timers.js:554:17)
Mar 23 15:50:02 OpenHAB3 npm[895]:     at processTimers (internal/timers.js:497:7)
Mar 23 15:50:02 OpenHAB3 npm[895]: (node:895) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was
```

Heronimonimo, 2022-04-03:
> I'm on 20.04 LTS and 5.4.0-104-generic. My zzh stick keeps timing out after a day. Is this the same issue? I can't fix it without a complete reboot and ejecting the stick.
> 
> ```
> Mar 23 15:46:42 OpenHAB3 npm[895]: (node:895) UnhandledPromiseRejectionWarning: Error: SRSP - ZDO - mgmtPermitJoinReq after 6000ms
> Mar 23 15:46:42 OpenHAB3 npm[895]:     at Timeout._onTimeout (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35)
> Mar 23 15:46:42 OpenHAB3 npm[895]:     at listOnTimeout (internal/timers.js:554:17)
> Mar 23 15:46:42 OpenHAB3 npm[895]:     at processTimers (internal/timers.js:497:7)
> Mar 23 15:46:42 OpenHAB3 npm[895]: (node:895) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was>
> Mar 23 15:50:02 OpenHAB3 npm[895]: (node:895) UnhandledPromiseRejectionWarning: Error: SRSP - ZDO - mgmtPermitJoinReq after 6000ms
> Mar 23 15:50:02 OpenHAB3 npm[895]:     at Timeout._onTimeout (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35)
> Mar 23 15:50:02 OpenHAB3 npm[895]:     at listOnTimeout (internal/timers.js:554:17)
> Mar 23 15:50:02 OpenHAB3 npm[895]:     at processTimers (internal/timers.js:497:7)
> Mar 23 15:50:02 OpenHAB3 npm[895]: (node:895) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was
> ```

I have the exact same problem since updating to 1.25.0 running the Home Assistent add-on on a Ubuntu machine.

RubenKelevra, 2022-04-07:
@Heronimonimo this looks like a different issue. Open a new issue for this :)

diagonali, 2022-05-17:
I'm having various issues with stability recently where before (6 months to a year ago) all worked well. No idea what the cause is, whether the zzh firmware a bug in zigbee2mqtt or in the driver etc. I'm currently running kernel 5.15.0-1006 on Ubuntu 22.04 on a Raspberry Pi 4 having changed from Raspberry Pi OS where same issues and latency is in region of 2-3 seconds. Tried all the usual - reinstall from scratch, change channel (I have much better signal on channel 25 now) etc. Really hope the root problem can be identified and it gets fixed soon, whatever it is.