Koenkk/zigbee2mqtt #8667: Zemismart M515EGB model

VikeDragon, 2021-09-06:
## Is your feature request related to a problem? Please describe

Greetings!
Is it possible for the Zemismart M515EGB model to provide an output of the battery charge level?
Is it possible to specify the real data of the vendor of the Zemismart device?

## Describe the solution you'd like
I expect to track the battery charge in the device Zemismart M515EGB
I expect that the picture that is already in the description will be beautifully displayed, and not the picture of another device. I also expect that the bucket data will be entered correctly.

## Describe alternatives you've considered
Track the battery discharge visually. Wait until the red LED lights up and the device stops working.
I will continue to look at information about other equipment :(

## Additional context
https://www.zigbee2mqtt.io/devices/TS0601_cover.html
https://www.zigbee2mqtt.io/images/devices/M515EGB.jpg

danieledwardgeorgehitchcock, 2021-09-07:
As the device is actually RF, with a USB powered Zigbee to RF bridge, this would not be possible.

RF is _usually_ a single way protocol. the way the bridge likely reports its states is by spamming the end device, and 'assuming' that the command has worked then reporting back a pseudo state through Zigbee.

VikeDragon, 2021-09-07:
Thank you for the answer! This is about the battery status, but what about other requests :
Is it possible to specify the real data of the vendor of the Zemismart device?
Is it possible to display custom images selected and displayed in the list on the Zigbee map?

danieledwardgeorgehitchcock, 2021-09-07:
At the moment, no - if the device is a white label of another device, a generic picture and description is used, and either the originating manufacturer name, or more generic TuYa name is used

VikeDragon, 2021-09-08:
Thank you for the answer! 
But in the device card there is a tab "Settings" and there is an "Icon" field at the very bottom, <<The user-defined device icon for the frontend. It can be a link to an image (not a path to a file) or base64 encoded data URL like: image/svg+xml;base64,PHN2ZyB3aW....R0aD)>>
 I filled in the "Icon" field and in the list I see the desired image of my device and I am very happy with this. But I see a general image on the map, is it possible to load the "Icon" field on the map as well as in the list ? Is it also possible to make the "Icon" field populated from a local file, and not just from the URL? About the description, vendor, etc., I understand that it is difficult to sort it out yet, but it would be possible to supplement the Settings department with fields for correction and everything, as with the "Icon" field. I understand that on the one hand these are small things, but on the other hand, the main impression about the project is formed from small things.

VikeDragon, 2021-11-02:
Thanks! The new version has custom icons on the zigbee map! Thank you very much!

closed 2021-11-02 by VikeDragon