Koenkk/zigbee2mqtt #8670: Cannot pair unless the router/enddevice is very close to the coordinator

ccorderod, 2021-09-06:
## What Happened

I did notice on the latest Z2M releases some difficulties to pair/repair devices/enddevices. Now it has become a real pitas as the only way to pair/repair is getting the device/enddevice extremely close to the coordinator.

Nothing shows up on the debug log/herdsman debug log. No sign of device/enddevice announcement when trying to pair/repair far from the coordinator.

This happen with hue lights/remotes as well as with xiaomi end devices.

I did try to pair hue lights/remote to a hue bridge (far from the lights) and it just took me a couple of minutes to do it.

## What did you expect to happen

Be able to pair/unpair a device/enddevice whenever it is located

## How to reproduce it (minimal and precise)

- Delete a hue light from the frontend, hence the hue light enters into pairing mode (tested with a hue bridge)
- Nothing is reported by Z2M

## Debug info
Zigbee2MQTT version: 1.21.1-dev commit: 02a50827
Adapter hardware: CC1352P-2
Adapter firmware version: 20210430
The mesh is big: 115 routers/enddevices, including 4 cc2530 routers, 50+ lights (hue and ikea - routers), 8 meazon meters (routers), 3 innr plugs (routers), 3 xiaomi smart plugs (routers), 2 tuya plugs (routers), and xiaomi and nyce enddevices

r-vit, 2021-09-06:
I'm experiencing the same problem recently. 

My coordinator is based on CC26X2R1, the Z2M version is 1.21.1 commit eb5750d9, firmware version is 20210708

The situation doesn't change whether I allow joining on all routers or on a specific one (the closest).

I tried to pair Tuya switch and Xiaomi power plug with the same result.

sjorge, 2021-09-06:
There was a new issue during that got fixed in 1.21 as mentioned in the release notes:
https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.21.0

> - This release fixes an issue in the backup/restore mechanism (#7680) which prevented devices to join via a router after reflashing the firmware of your adapter. Since this was caused by an incomplete backup, users who already reflashed their coordinators earlier have to repair the routers where joninig is not working for. This only applies to adapters with a Z-Stack_3.x.0 (e.g. zzh, CC2652R launchpad, Slaesh's) firmware. Important: before reflashing your adapter, make sure to start and stop this release at least once! This ensures your backup (coordinator_backup.json) is correct.
> - New firmwares are available for Texas Instruments CC2652R, CC2652RB, CC1352P-2 based adapters: 20210708 (changelog). This i.a. enables control for the LEDs (will be on when joining is permitted) and fixes an issue of devices not being able to join when joining is only enabled on a specific router. Important: before reflashing first read the point above.

sjorge, 2021-09-06:
@Koenkk maybe also worth making this a sticky for a while. I think it’s the 3rd or 4th issues about it.

ant-ds, 2021-09-07:
I seem to be having the same issue, despite having freshly flashed my Slaesh stick and made a new network entirely. I will try to unpair and repair everything today, but the coordinator is the only thing things want to join at the moment. Will update if it works.

djtrue, 2021-09-07:
I have renovated everything from scratch 4 times. The result is the same. There is clearly some kind of error.

sander816, 2021-09-07:
yes, same here, unable to pair, unless close to the coordinator

ccorderod, 2021-09-07:
> There was a new issue during that got fixed in 1.21 as mentioned in the release notes:
> https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.21.0
> 
> > * This release fixes an issue in the backup/restore mechanism ([Unable to control Zigbee2MQTT Covers in HA 2021.6.0 #7680](https://github.com/Koenkk/zigbee2mqtt/issues/7680)) which prevented devices to join via a router after reflashing the firmware of your adapter. Since this was caused by an incomplete backup, users who already reflashed their coordinators earlier have to repair the routers where joninig is not working for. This only applies to adapters with a Z-Stack_3.x.0 (e.g. zzh, CC2652R launchpad, Slaesh's) firmware. Important: before reflashing your adapter, make sure to start and stop this release at least once! This ensures your backup (coordinator_backup.json) is correct.
> > * New firmwares are available for Texas Instruments CC2652R, CC2652RB, CC1352P-2 based adapters: 20210708 (changelog). This i.a. enables control for the LEDs (will be on when joining is permitted) and fixes an issue of devices not being able to join when joining is only enabled on a specific router. Important: before reflashing first read the point above.
> 
> But since you're experiencing the issue, that means repairing all router devices. You should be able to enable joining and then tell a router close to the coordinator to leave and work your way out from there. I was able to avoid opening the wall sockets and switches to get at the devices behind it.

Thank you @sjorge. I miss this release note. My bad :(. I started to repair the routers starting with those near the coordinator. I notice a couple of things:
- Once deleted from z2m, some routers DO NOT rejoin. Painful. Have to reset the lights with the hue remote.
- Some others do rejoin but after the interview process got stuck (unable to power on/off the light/change the brightness. Again I have to reset those with the hue remote.

The networks (I run 2 Z2M networks as each support 100+ devices) are huge, and it will take some time to repair. Will keep it posted.
Thank you again. Stay safe.

sjorge, 2021-09-07:
>     * Once deleted from z2m, some routers DO NOT rejoin. Painful. Have to reset the lights with the hue remote.

Luckily all my in wall stuff did, I think i had like one tint bulb not reconnect immediately but cutting the power and restoring it kicked it in action.

>     * Some others do rejoin but after the interview process got stuck (unable to power on/off the light/change the brightness. Again I have to reset those with the hue remote.
Oops! This is because I forgot to mention you should use the join via router option! This happens when the device tries to join via a router that is still broken.

When moving out from the coordinator i always picked a router I already rejoined near the other device. So it would only be able to join via that one.

[image]

You can hit the V icon on the join button to select a specific router.

ccorderod, 2021-09-07:
"Oops! This is because I forgot to mention you should use the join via router option! This happens when the device tries to join via a router that is still broken."

No worries!. That's what I did, but even then some got stuck :(. Thank you Jorge.

ant-ds, 2021-09-07:
> I seem to be having the same issue, despite having freshly flashed my Slaesh stick and made a new network entirely. I will try to unpair and repair everything today, but the coordinator is the only thing things want to join at the moment. Will update if it works.

Update, I managed to get it to work with the latest version of Z2M (1.21.1-1) and with latest firmware (20210708). I started from scratch. One thing that seemed to be necessary was first pairing my routers (hue bulbs) to the coordinator (slaesh), then removing them in Z2M and letting them pair again. Afterwards they were joinable.

ccorderod, 2021-09-07:
Any of you has Ikea Tradfri bulbs in the network you are/you had rebuilt? How did you repair those bulbs? Thanks

ccorderod, 2021-09-08:
@sjorge I am struggling to rebuild both networks, and as it seems you went through this process, you may give me some light:
- Enabling joining on a previous repaired router (for example a bulb next to the bulb I want to repair) is NOT reliable. Sometimes it works, sometime it doesn't and I have to enable join on "All" for the bulb to join. Did this happen to you? Any thought?
- Ikea bulbs are almost impossible to rejoin unless you do it close (a few cms) to the coordinator, but sometimes it works. Any thought?
- Some hue lights are just gone, and do not rejoin at all. I "reset" those with a hue remote (extremely dangerous as it may reset a bulb located a few meters from the one you want to reset; this happen a lot with Ikea bulbs). Even after "resetting" the hue light, it does not rejoin. I have to setup a hue bridge, make the bulb join using serial number, then delete the bulb from the hue and enabling the permit join all on the z2m network. What a pain. Any thought?

Thank you. Have all a great day!.

sjorge, 2021-09-08:
> @sjorge I am struggling to rebuild both networks, and as it seems you went through this process, you may give me some light:
Sadly I went through it like 3 or so times while this was still being debugged.

>     * Enabling joining on a previous repaired router (for example a bulb next to the bulb I want to repair) is NOT reliable. Sometimes it works, sometime it doesn't and I have to enable join on "All" for the bulb to join. Did this happen to you? Any thought?
Didn't have many issues with this, I did the following:
1. enable join only on the closest router
2. tell the target bulb to leave

I think I had 2 that didn't come back immediately and required the 6-off-on cycle (I own mostly tradfri and a few innr)

> 
>     * Ikea bulbs are almost impossible to rejoin unless you do it close (a few cms) to the coordinator, but sometimes it works. Any thought?
Didn't experience any problems with this at all, I'm guessing they are still trying to join via a router that hasn't been rejoined yet. I guess you could try powering down those routers instead and use join via the default all?

> * Some hue lights are just gone, and do not rejoin at all. I "reset" those with a hue remote (extremely dangerous as it may reset a bulb located a few meters from the one you want to reset; this [...]

I didn't own any Hue lights at the time, I own 1 now but I did not have to reset it yet. Perhaps it's normal for hue to not imediatly try to rejoin after it is told to leave. I don't know.

RaffoNin, 2021-09-11:
I really would consider the possibility that the new update didn't fix the problem entirely. Personally I've done the instructions on the change log and re-paired 3 times but it's still not pairing to the router. Recently i wiped everything entirely and started from scratch. 

My network is stable now with 40 devices but I had to pair everything to the coordinator and move it out and pray it's not as sticky as my aqara sensors.

ccorderod, 2021-09-11:
Quick update. 
First is first: @sjorge Thanks for your support. It is really appreciated.
The issue is NOT fixed and/or there is a problem dealing with latest hue firmware.
I finally rebuild my two networks (i real huge pain!). 
Some background: hue and ikea firmware were updated in all lights to latest version. Some of my hue lights did not update from z2m (wrong firmware version); i moved them to hue bridge and updated firmware from there, When moved back to Z2M the old firware still shows up on the details of the bulb, but OTA checking does not raise any new update.
My findings:
- It is ONLY possible to pair through coordinator and ikea lights (but only lights on firmware 2.3.050/2.3.068; I do have a couple of bulbs on 1.3.X as no firmware is available for those which are not able to route pairing). Selecting any other router in my network (including hue, meazon, tuya, innr, xiaomi) won't work
- Remove (I strongly advise to do this) the router from Z2M from the frontend BEFORE repairing; not doing so will give you a hardtime sometimes repairing
- The removed router will repair quickly as far as permit join is enabled for an ikea router (on firmware 2.X)
- Ikea bulbs WONT repair automatically; remove the router through z2m frontend first, then RESET the router with the 6 steps on/off to reset the router. If you don't, it will simply NOT pair.
- Meazon routers won't repair after removing them; you will need to reset the router (push the button until you get the led flashing 3 times, wait 15 seconds, and push the button one more time).
- CCRouters, tuya, innr and xiaomi will repair automatically after removing
- You may loose some enddevices during the process; I think this repairing process mess a lot with the network and they loose their route. In my case (xiaomi, aqara enddevices), just pushing the button a few times will make them rejoin the network without the need to reset them all.

I do hope this will help someone. Have a great we.

MattWestb, 2021-09-12:
Philips HUE need being power cycled before its can being touch link retested then its blocked after some minutes for security reason.

The problem is having 2 networks up and running one with bad setting and one with good.
If one router is joining thru one with bad setting its getting bad setting.

I think its better changing some network parameter (network key, PAN and E-PAN ID) for building one new fresh network and not getting problem with old junk in the mesh.

Or shutting down all devices and rejoining them one by one (and no old shall being alive then doing that so retested can only finding the good network).

ccorderod, 2021-09-12:
@MattWestb thanks for the comment. There is no problem to have 2 networks or even more as far as permit join is only enabled on the one you want to rejoin your stuff ;-). No need to change network config or whatever.
I did not use touchlink at all as you need to have the bulb very close to the coordinator and all my lights are already in place, so this is not an option. But using the hue remote reset procedure works quite well as far as you take care of not resetting a light close to the one you want to reset :-). By the way, you also need to power cycle the light (hue, ikea, ...) before the reset procedure with the hue remote.

MattWestb, 2021-09-12:
I have not real  testing touch link reset of my IKEA lights but by accident with deCONZ 6 meter away in the kitchen then not looking so exact before hitting reset so some of IKEA 2 gen WS GU10 dont need re power for getting reseed but can needing it for entering pairing mode.

And the HUE remote is technically using touch link reset only without "GUI" ;-).

If you have more networks they **must** have different PAN and -PAN ID or you is getting very much and strange problems all over your networks with conflicts between them and jumping" devices with or without success !!

groenmarsmannetje, 2021-10-29:
Thanks for all recommendations. Finally I got one of my  Tradfri lights back in the network again. I used the IKEA repeater, which I repaired first to coordinator. By moving this repeater it was easier to do. 

But do I really need to repair all the router devices?

sjorge, 2021-10-29:
> But do I really need to repair all the router devices?

Sadly yes :( I had to repair over 40 devices.

groenmarsmannetje, 2021-10-30:
What a mess I have 44 routers and 56 end devices. It is not clear to me which change introduced this bug and if there is an option to rollback to a previous version. Was it a zigbee2mqtt release or a coordinator firmware relaase?

sjorge, 2021-10-30:
Rollback is not going to work once you’re in the situation.

Basically the coordinator backup was missing a field in one release of z2m. If you reflagged the coordinator while you were on the release you ended up with incomplete state being restored to the coordinator.

From that point onwards there is way to fix it unless to repair all routers.

~ sjorge

nickw444, 2021-10-30:
>  One of the devices that I repaired yesterday has dropped of the network already and I am not able to repair it again via the routers that I repaired yesterday.

I'm in this same position, and looks like it's a bug in the coordinator firmware: https://github.com/Koenkk/zigbee2mqtt/issues/8803#issuecomment-933772434

> > When TC settings do not mandate TCLK update, TC does not process APS packets w/ well-known TCLK encryption for remote multi-hop devices
> 
> 
> Looks likes TI might already have fixed this bug, however I didn't update to 5.20 yet since it gave issues (current fw is based on 5.10). I expect the next release soon (last was released in July and they release every quarter), will ping you when I have a fw ready.

pepp86, 2021-11-09:
Hi,

quick question, is there a way to check whether my network is affected or not? I'm currently not sure if that's my problem or if there is another. But repairing would take ages for me (nearly 100 devices)

groenmarsmannetje, 2021-11-09:
> Hi,
> 
> quick question, is there a way to check whether my network is affected or not? I'm currently not sure if that's my problem or if there is another. But repairing would take ages for me (nearly 100 devices)

I noticed when I tried to repair a sensor which was dropped from the network. I was only able to repair it when I was near the coordinator. Actually it took a while before I became aware that this is weird until I read this issue.

artist67, 2021-11-09:
I had the same issue with my 100 devices with the July firmware. I tried with sleash stich and sonoff+.
Furthermore, communication was bad, several events from sensors did not arrive sporadically.
I repaired all routers and several end devices but still face the same issues.

After I went back to the January firmware, all devices pair smoothly.

djtrue, 2021-12-09:
I seem to have solved the problem. For about a month now, all routers do not lose the ability to pair.
My actions:
1) removed all routers forcibly (yes yes again ...)
2) rebooted z2m
3) Updated the stick firmware to ver. 20211114 (DEV branch in coordinator's firmware)
4) Added all routers again + several end devices that stopped responding

Krystm, 2021-12-14:
> I seem to have solved the problem. For about a month now, all routers do not lose the ability to pair. My actions:
>
> 1. removed all routers forcibly (yes yes again ...)
> 2. rebooted z2m
> 3. Updated the [...]

When you upgraded was the flashing backup / upgrade / restore? This issue has been a nightmare trying to get devices in reliable locations when Conbee and My Hubitat never had issues with quality. For the record I was able to get a lot of devices semi-close, technically there are definitely some I feel should be on different routers, but a complete / wipe / fresh install I cannot get 1 of 8 different routers to allow joins. So figure will take a stab at this, since nothing to lose but want to make sure I make it as painless as possible.

Krystm, 2021-12-15:
Update:
Upgrading is amazing I will say, just flash and good to go. 

I updated to the 202111114 as well, all the routers fell off by themselves, forced deleted them, restarted z2m, re-paired them and some of the end devices as well. So far everything is balancing out it looks like and working great. I feel much better about this and WAF is much higher so that's always a plus. Did have to restart a few times as there are some other separate issues looks like with Z2M where devices were showing left network even though they were deleted etc. But I did get them all on pretty painlessly, devices I was having issues with joining are now joining their nearest routers.

swiergot, 2022-01-08:
I'm experiencing the same problem. Today I installed SONOFF Zigbee 3.0 USB Dongle Plus, it's my third coordinator CC2531 and CC2538, and I hoped to solve all my problems with SONOFF. Unfortunately, I didn't.

I have the coordinator on the 2nd floor, one router on the 2nd floor, two routers on the 1st floor and three routers on the ground floor. Routers on the 1st and 2nd floor paired just fine, but I could not pair anything on the ground floor.

It had a July firmware, so I flashed the latest one, removed 1st floor routers and paired them again, but it didn't help with paring ground floor routers.

So I moved my RPi to the ground floor and paired everything with no issues. Then I moved RPi back to the 2nd floor where it belongs and now nothing on the ground floor is working.

Any ideas?

closed 2022-02-15 by github-actions[bot]