Koenkk/zigbee2mqtt #8676: Ikuu Tuya single gang switch does not show on/off status when toggled locally

Asleep-at-the-wheel, 2021-09-07:
## What happened
The following device has been installed and paired ok with Z2mQtt just fine.
https://www.ikuu.com.au/product/single-switch/

Control of the device over the Zigbee network is fine and the device identifies ok. The device however does not report an on/off status change when toggled on the switch itself using the button built onto the front of it. When I press on/off on the device itself, I see no log messages in Zigbee2mqtt, no status change anywhere and nothing to indicate anything at all has happened.

I found that pairing the switch with ZHA using a CC2531 I had spare produced expected results. A friend of mine used his Conbee2 to pair with Deconz add on in HA and it also worked as expected too.

I tried making an external converter to learn about the device but every time I try zigbee2mqtt just fails to load, I guess I am doing that wrong too.

I am led to believe that the way these Ikuu branded devices are interpreted by zigbee2mqtt is incorrect some how based on the strange behavior I also see here and with the dimmer version of this device here https://github.com/Koenkk/zigbee2mqtt/issues/8556. 

After a couple of solid days plugging away at these issues I got no where unfortunately. I am happy to try and find, learn and iron out these issues myself with this device and the dimmer version if I am provided some eli5 tips on where to look and what to start with.

Is there maybe something that needs to be "turned on" in the firmware of these tuya devices that maybe Deconz or ZHA do? I find it odd that I see literally zero log messages from this device when pressed locally, I saw someone else report the same thing using Zigbee2Mqtt on the HA forums, so I know it isn't just my unit.

## What did you expect to happen
On/Off status updates reflected in HA when the device is toggled locally.

## How to reproduce it (minimal and precise)
1. Connect device
2. Pair with Zigbee2Mqtt
3. Press button locally, nothing observed.

## Debug info
Zigbee2MQTT version: Hassio 1.21.1-1 (Zigbee2MQTT version 1.21.1)
Adapter hardware: Conbee II
Adapter firmware version: What ever it came with.

**About device:**
`Friendly name:   0x60a423fffeb7e49b
Last seen:    N/A
Device type:    EndDevice
Zigbee Model:    TS0011
Zigbee Manufacturer:    _TZ3000_l8fsgo6p
Description:    Smart light switch - 1 gang
Support status:    Supported
IEEE Address:    0x60a423fffeb7e49b
Network address:    0xA4AB
Manufacturer:    TuYa
Model:    TS0011
Power:
Interview completed:    True`

**Database.db**
`{"id":14,"type":"EndDevice","ieeeAddr":"0x60a423fffeb7e49b","nwkAddr":42155,"manufId":4098,"manufName":"_TZ3000_l8fsgo6p","powerSource":"Mains (single phase)","modelId":"TS0011","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[0,4,5,6,3],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS0011","manufacturerName":"_TZ3000_l8fsgo6p","powerSource":3,"zclVersion":3,"appVersion":67,"stackVersion":0,"hwVersion":1,"dateCode":""}},"genOnOff":{"attributes":{"onOff":1}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":67,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":2},"lastSeen":1631013983438}`

**Log entry when controlling over network:** (works as expected)
`info  2021-09-07 19:56:21: MQTT publish: topic 'zigbee2mqtt/0x60a423fffeb7e49b', payload '{"linkquality":255,"state":"OFF"}'
debug 2021-09-07 19:56:22: Received MQTT message on 'zigbee2mqtt/0x60a423fffeb7e49b/set' with data '{"state":"ON"}'
debug 2021-09-07 19:56:22: Publishing 'set' 'state' to '0x60a423fffeb7e49b'
info  2021-09-07 19:56:22: MQTT publish: topic 'zigbee2mqtt/0x60a423fffeb7e49b', payload '{"linkquality":255,"state":"ON"}'
debug 2021-09-07 19:56:24: Received MQTT message on 'zigbee2mqtt/0x60a423fffeb7e49b/set' with data '{"state":"OFF"}'
debug 2021-09-07 19:56:24: Publishing 'set' 'state' to '0x60a423fffeb7e49b'
info  2021-09-07 19:56:24: MQTT publish: topic 'zigbee2mqtt/0x60a423fffeb7e49b', payload '{"linkquality":255,"state":"OFF"}'
`

No log message produced when pressed locally.

Asleep-at-the-wheel, 2021-09-14:
I have done lots of testing and I am confused.

Works as expected with:
- ZZH and ZHA
- Conbee 2 and Deconz

Does not work as expected with:
- ZZH and Zibee2MQTT
- Conbee 2 and Zigbee2MQTT
- CC2351 and Zigbee2MQTT

I set up an external converter with tuya dump from Zigbee and saw literally nothing (no dump seen) when pressing the button or sending on/off over the network.

Discussion here shows that the device works for one user after joining ZHA network then moving to Zigbee2MQTT using ZZH.
https://community.home-assistant.io/t/mains-light-switch-with-button-press-state-change-push-notification/325222/24

None of this makes any sense to me. Are there any tips anyone can give me on where to look next?

Koenkk, 2021-09-14:
Changing the definition to below might fix the problem:

```js
{
    zigbeeModel: ['TS0011'],
    model: 'TS0011',
    vendor: 'TuYa',
    description: 'Smart light switch - 1 gang',
    extend: extend.switch(),
    whiteLabel: [{vendor: 'Vrey', model: 'VR-X712U-0013'}, {vendor: 'TUYATEC', model: 'GDKES-01TZXD'},
        {vendor: 'Lonsonho', model: 'QS-Zigbee-S05-L', description: '1 gang smart switch module without neutral wire'}],
    configure: async (device, coordinatorEndpoint, logger) => {
        await reporting.bind(device.getEndpoint(1), coordinatorEndpoint, ['genOnOff']);
        // Reports itself as battery which is not correct: https://github.com/Koenkk/zigbee2mqtt/issues/6190
        device.powerSource = 'Mains (single phase)';
        device.save();
    },
},
```

Asleep-at-the-wheel, 2021-09-15:
Thank you for your suggestion. Your support is invaluable as usual. My issue is I just lack a real in depth understanding of Zigbee and how to fault find it.

I did a quick and dirty test of the following:
`    {                  
        fingerprint: [{modelID: 'TS0011', manufacturerName: '_TZ3000_l8fsgo6p'}],
        zigbeeModel: ['TS0011'],                          
        model: 'TS0011',
        vendor: 'TuYa',                                       
        description: 'Smart light switch - 1 gang - test',
        extend: extend.switch(),
        whiteLabel: [{vendor: 'Vrey', model: 'VR-X712U-0013'},                  
        {vendor: 'TUYATEC', model: 'GDKES-01TZXD'},                    
        {vendor: 'Lonsonho', model: 'QS-Zigbee-S05-L', description: '1 gang smart switch module without neutral wire'},
        ],             
        configure: async (device, coordinatorEndpoint, logger) => {
            // Reports itself as battery which is not correct: https://github.com/Koenkk/zigbee2mqtt/issues/6190
            device.powerSource = 'Mains (single phase)';
            device.save();                                                          
        },
    },`

Found it made no difference.

What I am super confused about, is how can the tuya data point dump not show me anything? Unless the device literally doesn't output that data? But then how on earth does it work with ZHA?

Is there some other way (other than the tuya dump) to see what this device outputs? Or is there something that gets "set" in the device when its connected to ZHA that some how Z2MQTT doesn't "set"?

Asleep-at-the-wheel, 2021-09-15:
Got the following logs out of ZHA when the button was being pressed lots. I'll try to learn what most of this stuff means =)

```
2021-09-15 14:15:06 DEBUG (MainThread) [zigpy_znp.api] Sending request: SYS.Ping.Req()

2021-09-15 14:15:06 DEBUG (MainThread) [zigpy_znp.api] Received command: SYS.Ping.Rsp(Capabilities=<MTCapabilities.APP_CNF|GP|UTIL|ZDO|AF|SYS: 1625>)

2021-09-15 14:15:20 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.IncomingMsg.Callback(GroupId=0x0000, ClusterId=6, SrcAddr=0xC60A, SrcEndpoint=1, DstEndpoint=1, WasBroadcast=<Bool.false: 0>, LQI=93, SecurityUse=<Bool.false: 0>, TimeStamp=14872706, TSN=0, Data=b'\x18\x7B\x0A\x00\x00\x10\x01', MacSrcAddr=0xC60A, MsgResultRadius=29)
2021-09-15 14:15:20 DEBUG (MainThread) [zigpy.zcl] [0xc60a:1:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=123 command_id=Command.Report_Attributes>

2021-09-15 14:15:20 DEBUG (MainThread) [zigpy.zcl] [0xc60a:1:0x0006] ZCL request 0x000a: [[Attribute(attrid=0, value=<TypeValue type=Bool, value=Bool.true>)]]

2021-09-15 14:15:20 DEBUG (MainThread) [zigpy.zcl] [0xc60a:1:0x0006] Attribute report received: on_off=1

2021-09-15 14:15:22 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.IncomingMsg.Callback(GroupId=0x0000, ClusterId=6, SrcAddr=0xC60A, SrcEndpoint=1, DstEndpoint=1, WasBroadcast=<Bool.false: 0>, LQI=87, SecurityUse=<Bool.false: 0>, TimeStamp=14945582, TSN=0, Data=b'\x18\x7C\x0A\x00\x00\x10\x00', MacSrcAddr=0xC60A, MsgResultRadius=29)

2021-09-15 14:15:22 DEBUG (MainThread) [zigpy.zcl] [0xc60a:1:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=124 command_id=Command.Report_Attributes>

2021-09-15 14:15:22 DEBUG (MainThread) [zigpy.zcl] [0xc60a:1:0x0006] ZCL request 0x000a: [[Attribute(attrid=0, value=<TypeValue type=Bool, value=Bool.false>)]]

2021-09-15 14:15:22 DEBUG (MainThread) [zigpy.zcl] [0xc60a:1:0x0006] Attribute report received: on_off=0

2021-09-15 14:15:31 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.IncomingMsg.Callback(GroupId=0x0000, ClusterId=6, SrcAddr=0xC60A, SrcEndpoint=1, DstEndpoint=1, WasBroadcast=<Bool.false: 0>, LQI=93, SecurityUse=<Bool.false: 0>, TimeStamp=15516734, TSN=0, Data=b'\x18\x7D\x0A\x00\x00\x10\x01', MacSrcAddr=0xC60A, MsgResultRadius=29)

2021-09-15 14:15:31 DEBUG (MainThread) [zigpy.zcl] [0xc60a:1:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=125 command_id=Command.Report_Attributes>

2021-09-15 14:15:31 DEBUG (MainThread) [zigpy.zcl] [0xc60a:1:0x0006] ZCL request 0x000a: [[Attribute(attrid=0, value=<TypeValue type=Bool, value=Bool.true>)]]

2021-09-15 14:15:31 DEBUG (MainThread) [zigpy.zcl] [0xc60a:1:0x0006] Attribute report received: on_off=1

2021-09-15 14:15:36 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.IncomingMsg.Callback(GroupId=0x0000, ClusterId=6, SrcAddr=0xC60A, SrcEndpoint=1, DstEndpoint=1, WasBroadcast=<Bool.false: 0>, LQI=90, SecurityUse=<Bool.false: 0>, TimeStamp=120244, TSN=0, Data=b'\x18\x7E\x0A\x00\x00\x10\x00', MacSrcAddr=0xC60A, MsgResultRadius=29)

2021-09-15 14:15:36 DEBUG (MainThread) [zigpy.zcl] [0xc60a:1:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=126 command_id=Command.Report_Attributes>

2021-09-15 14:15:36 DEBUG (MainThread) [zigpy.zcl] [0xc60a:1:0x0006] ZCL request 0x000a: [[Attribute(attrid=0, value=<TypeValue type=Bool, value=Bool.false>)]]

2021-09-15 14:15:36 DEBUG (MainThread) [zigpy.zcl] [0xc60a:1:0x0006] Attribute report received: on_off=0

2021-09-15 14:15:36 DEBUG (MainThread) [zigpy_znp.api] Sending request: SYS.Ping.Req()

2021-09-15 14:15:36 DEBUG (MainThread) [zigpy_znp.api] Received command: SYS.Ping.Rsp(Capabilities=<MTCapabilities.APP_CNF|GP|UTIL|ZDO|AF|SYS: 1625>)

2021-09-15 14:15:40 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.IncomingMsg.Callback(GroupId=0x0000, ClusterId=6, SrcAddr=0xC60A, SrcEndpoint=1, DstEndpoint=1, WasBroadcast=<Bool.false: 0>, LQI=96, SecurityUse=<Bool.false: 0>, TimeStamp=385724, TSN=0, Data=b'\x18\x7F\x0A\x00\x00\x10\x01', MacSrcAddr=0xC60A, MsgResultRadius=29)

2021-09-15 14:15:40 DEBUG (MainThread) [zigpy.zcl] [0xc60a:1:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=127 command_id=Command.Report_Attributes>

2021-09-15 14:15:40 DEBUG (MainThread) [zigpy.zcl] [0xc60a:1:0x0006] ZCL request 0x000a: [[Attribute(attrid=0, value=<TypeValue type=Bool, value=Bool.true>)]]

2021-09-15 14:15:40 DEBUG (MainThread) [zigpy.zcl] [0xc60a:1:0x0006] Attribute report received: on_off=1

2021-09-15 14:15:44 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.IncomingMsg.Callback(GroupId=0x0000, ClusterId=6, SrcAddr=0xC60A, SrcEndpoint=1, DstEndpoint=1, WasBroadcast=<Bool.false: 0>, LQI=90, SecurityUse=<Bool.false: 0>, TimeStamp=645939, TSN=0, Data=b'\x18\x00\x0A\x00\x00\x10\x00', MacSrcAddr=0xC60A, MsgResultRadius=29)

[... cut ...]
2021-09-15 14:16:19 DEBUG (MainThread) [zigpy.zcl] [0xc60a:1:0x0006] ZCL request 0x000a: [[Attribute(attrid=0, value=<TypeValue type=Bool, value=Bool.false>)]]

2021-09-15 14:16:19 DEBUG (MainThread) [zigpy.zcl] [0xc60a:1:0x0006] Attribute report received: on_off=0
2021-09-15 14:16:36 DEBUG (MainThread) [zigpy_znp.api] Sending request: SYS.Ping.Req()
2021-09-15 14:16:36 DEBUG (MainThread) [zigpy_znp.api] Received command: SYS.Ping.Rsp(Capabilities=<MTCapabilities.APP_CNF|GP|UTIL|ZDO|AF|SYS: 1625>)
```

Asleep-at-the-wheel, 2021-09-15:
Actually, I am an idiot.
Looking back at what I tried compared to what Koenkk suggested, I made a mistake.

The following does work:
```java
    {                                                         
        fingerprint: [{modelID: 'TS0011', manufacturerName: '_TZ3000_l8fsgo6p'}],
        zigbeeModel: ['TS0011'],
        model: 'TS0011',                                                        
        vendor: 'TuYa',                                                
        description: 'Smart light switch - 1 gang new converter',
        extend: extend.switch(),
        whiteLabel: [{vendor: 'Vrey', model: 'VR-X712U-0013'}, {vendor: 'TUYATEC', model: 'GDKES-01TZXD'},
            {vendor: 'Lonsonho', model: 'QS-Zigbee-S05-L', description: '1 gang smart switch module without neutral wire'}],
        configure: async (device, coordinatorEndpoint, logger) => {
                await reporting.bind(device.getEndpoint(1), coordinatorEndpoint, ['genOnOff']);
            // Reports itself as battery which is not correct: https://github.com/Koenkk/zigbee2mqtt/issues/6190
            device.powerSource = 'Mains (single phase)';
            device.save();                                                       
        },                         
    },                 
```
I will get a pull request sorted and have this issue fixed then close this issue!
Thanks again!

Koenkk, 2021-09-15:
> Still super confused about why the Tuya dump didn't show me anything though.

TuYa dump is for the TuYa specific clusters, the `genOnOff` cluster is just a standard Zigbee cluster.

What `await reporting.bind(device.getEndpoint(1), coordinatorEndpoint, ['genOnOff']);` does is bind the `genOnOff` cluster of the device to the coordinator, this will make the device send updates of any attributes in the `genOnOff` cluster to the coordinator. Assuming this can be closed now.

closed 2021-09-15 by Koenkk