Koenkk/zigbee2mqtt #8677: please add legrand celiane netatmo cable outlet

Pepeuch, 2021-09-07:
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
zigbeeModel: ['0 648 47/51/63/79/82/83/98'],
model: ' 0 648 47/51/63/79/82/83/98',
vendor: 'Legrand',
description: 'Sortie de cable 3000W avec fil pilote et consommation',
fromZigbee: [],
toZigbee: [],
[image]

data/database.db entry of the device:

{"id":26,"type":"Router","ieeeAddr":"0x00047400001202f1","nwkAddr":60051,"manufId":4129,"manufName":" Legrand\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000","powerSource":"Mains (single phase)","modelId":" Cable outlet\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":1,"inClusterList":[0,3,4,6,5,64513,2820,64576],"outClusterList":[0,64513,25],"clusters":{"genBasic":{"attributes":{"modelId":" Cable outlet\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000","manufacturerName":" Legrand\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000","powerSource":1,"zclVersion":2,"appVersion":0,"stackVersion":66,"hwVersion":1,"dateCode":" \u000020210622\u0000\u0000\u0000\u0000\u0000","swBuildId":"0008"}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"epId":242,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":0,"stackVersion":66,"hwVersion":1,"dateCode":" \u000020210622\u0000\u0000\u0000\u0000\u0000","swBuildId":"0008","zclVersion":2,"interviewCompleted":false,"meta":{},"lastSeen":1631022197513}

## Information about the device + link
..

## data/database.db entry of the device
..

Koenkk, 2021-09-07:
Can you provide a link to this device?

Pepeuch, 2021-09-08:
https://www.legrand.fr/pro/catalogue/43076-sortie-de-cable/sortie-de-cable-connectee-pour-installation-with-netatmo-compatible-fil-pilote-3000w-on-off-et-mesure-conso-blanc

Koenkk, 2021-09-10:
I don't see any `fromZigbee` or `toZigbee` converters in the definition, please follow https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html

Pepeuch, 2021-09-14:
I'm looking for how to program the pilot wire output but I don't have any example in the converter yet, I feel that it takes a lot of time. I would have to do some tests with deconz, they have already integrated the module.

Pepeuch, 2021-09-15:
hello, i have looked a little more carefully in the list of devices which are already recognized, and the schneider eclectric devices already have an integration, i am trying to do some tests with the configuration of the wiser.

Do you think it can work?
I still have errors in the code I will try to finalize it.
do not hesitate to tell me if you see the errors (you are probably better than me):

cable_outlet.js:
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['0 648 47/51/63/79/82/83/98'], // The model ID from: Device with modelID 'lumi.sens' is not supported.
    model: 'Cable outlet', // Vendor model number, look on the device for a model number
    vendor: 'Legrand', // Vendor of the device (only used for documentation and startup logging)
    description: 'cable outlet 3000W with consumption and pilot wire 4 or 6 order', // Description of the device, copy from vendor site. (only used for documentation and startup logging)
    fromZigbee: [fz.thermostat, fz.metering, fz.legrand_scenes, fz.identify, fz.on_off, fz.legrand_pilot_mode, fz.electrical_measurement, fz.legrand_device_mode, fz.ignore_basic_report, fz.ignore_genOta], // We will add this later
    toZigbee: [tz.legrand_deviceMode, tz.on_off, tz.legrand_identify, tz.electrical_measurement_power, tz.thermostat_system_mode, tz.thermostat_running_state,
            tz.thermostat_local_temperature, tz.thermostat_occupied_heating_setpoint, tz.thermostat_control_sequence_of_operation, tz.legrand_pilot_mode], // Should be empty, unless device can be controlled (e.g. lights, switches).
    exposes: [e.power(), e.energy(),
            exposes.enum('legrand_pilot_mode', ea.ALL, ['contactor', 'pilot']).withDescription('Controls piloting mode'),
            exposes.climate().withSetpoint('occupied_heating_setpoint', 4, 30, 0.5).withLocalTemperature()
                .withSystemMode(['off', 'auto', 'heat']).withPiHeatingDemand()],
     configure: async (device, coordinatorEndpoint, logger) => {
             const endpoint1 = device.getEndpoint(1);
             const endpoint2 = device.getEndpoint(2);
             await reporting.bind(endpoint1, coordinatorEndpoint, ['hvacThermostat']);
             await reporting.thermostatOccupiedHeatingSetpoint(endpoint1, {min: 0, max: 60, change: 1});
             await reporting.thermostatPIHeatingDemand(endpoint1, {min: 0, max: 60, change: 1});
             await reporting.bind(endpoint2, coordinatorEndpoint, ['seMetering']);
             await reporting.instantaneousDemand(endpoint2, {min: 0, max: 60, change: 1});
             await reporting.currentSummDelivered(endpoint2, {min: 0, max: 60, change: 1});
         },
}

Pepeuch, 2021-09-15:
good evening, after a little research on the site "works with legrand" I found some interesting documentation on this page:

https://developer.legrand.com/documentation/product-cluster-list/

closed 2021-10-23 by github-actions[bot]

Pepeuch, 2021-10-31:
> 
> 
> Hello @Pepeuch, Have you succeeded in including your LEGRAND cable outlet. I would like to buy 4 of them coupled with Zigbee2mqtt. Regards.

hi, no I have not yet succeeded in integrating these cable outputs for lack of time, I will try again in a few weeks.