Koenkk/zigbee2mqtt #8678: Xiaomi MCCGQ01LM taking very long to send first update, second update received instantly

LouisMT, 2021-09-07:
## What happened
I have two Xiaomi MCCGQ01LM contact sensors. One is right next to the coordinator (and connected directly), the other one is at the other side of the house (and connected through a router). They both worked fine before. But recently I expanded my network and I had to change the channel from 11 to 25. Now it takes very long for the first update to pop up. However, whenever I trigger a second update, the first and second update are received instantly at the same time.

For example:
* I open the door: no update for a long time (eventually I receive the update, but it takes multiple seconds to minutes)
* I open the door and close it: I receive both updates instantly at the same time

I can reproduce this every single time. Tried repairing and waiting for the network to settle, but no luck. I doubt it has something to do with battery level or signal strength. One sensor is a few days old. And when triggering a second update it's always popping up instantly.

## What did you expect to happen
To receive the first update instantly as well.

## How to reproduce it (minimal and precise)
1. Pair a MCCGQ01LM
2. Trigger an update

## Debug info
Zigbee2MQTT version: 1.21.1
Adapter hardware: CC2652R
Adapter firmware version: 20210708

Koenkk, 2021-09-07:
Probably because of https://github.com/Koenkk/zigbee2mqtt/issues/8663

LouisMT, 2021-09-07:
Oops, totally overlooked that one! Seems right, I'm running 5.10.60 and installed the kernel update around the same time as my network expansion. Probably didn't experience WiFi interference too, then. :sweat_smile:

Thanks! I'll reopen this if a kernel update doesn't fix this.

closed 2021-09-07 by LouisMT