Koenkk/zigbee2mqtt #8682: My DOOR Sensor exposes TEMPERATURE through Apple HomeKit ?!

groovy75, 2021-09-07:
Hello, 

I have a problem with my Aqara Door Sensor ([MCCGQ11LM](https://www.zigbee2mqtt.io/devices/MCCGQ11LM.html#xiaomi-mccgq11lm)) 

## What happened

on apple homekit i can see the temperature of the sensor. I don't want to see it because it is about the temperature of the battery.

## What did you expect to happen

I don't want to see the temperature value in HomeKit. I just want to use my product as a simple door sensor

## How to reproduce it (minimal and precise)

**1.  Buy MCCGQ11LM (Aqara Door Sensor)  on Aliexpress**
**2. Add your MCCGQ11LM to Z2mqtt**

[image]

[image]

[image]
**3. Use homebridge to expose Z2mqtt devices to Apple Home App.**
**4. Here we go, you have a temperature value, i don't want it.** (see picture below, i want to get rid of 32°C)

[image]

## Debug Info
Zigbee2MQTT version: 1.21.0 commit: 70891eec
Coordinator type: ConBee2/RaspBee2
Coordinator revision :0x26580700
Frontend version: 0.5.18

Shred99, 2021-09-08:
Take a look at the filtered_attributes configuration option.

[https://www.zigbee2mqtt.io/information/configuration.html](https://www.zigbee2mqtt.io/information/configuration.html)

sjorge, 2021-09-08:
> Take a look at the filtered_attributes configuration option.
> 
> https://www.zigbee2mqtt.io/information/configuration.html

That won't filter the expose data though, it will still expose it but probably as null.

We'd need something like `filtered_exposes` but I don't know how technically feasible something like that is. cc: @Koenkk

groovy75, 2021-09-08:
> Take a look at the filtered_attributes configuration option.
> 
> https://www.zigbee2mqtt.io/information/configuration.html

> That won't filter the expose data though, it will still expose it but probably as null.
>
> We'd need something like `filtered_exposes` but I don't know how technically feasible something like that is. [...]

Hello,

Thank you for your help.
This is what I did , but nothing change after saving new config file and restarted my Raspberry Pi: 

[url=https://postimg.cc/qgvN5Jvp][img]https://i.postimg.cc/qgvN5Jvp/Screenshot-2021-09-08-at-10-24-56.png[/img][/url]

[url=https://postimg.cc/m1khzw5P][img]https://i.postimg.cc/m1khzw5P/Screenshot-2021-09-08-at-10-25-10.png[/img][/url]

sjorge, 2021-09-08:
The first entry is correct, but as I mentioned above, it won't do what you want.

groovy75, 2021-09-08:
I also noticed that when I put my mouse on "temperature" . Maybe this can help you guys ? 

[image]

Is there a way to erase the "temperature" line from Z2mqtt ? I don't want it anymore :(

sjorge, 2021-09-08:
Currently, no there is no way to filter those out.

groovy75, 2021-09-08:
Oh, okay, thank you.. if someone has a work-around please tell me...

puijken, 2021-09-10:
Why not do this in Homebridge? In the Z2M plugin for Homebridge you can exlude devices or device properties .. looks something like this:
[image]

In your case you just need to exclude "temperature".

groovy75, 2021-09-10:
> Why not do this in Homebridge? In the Z2M plugin for Homebridge you can exlude devices or device properties .. looks something like this:
> [image]
> 
> In your case you just need to exclude "temperature".

WORKING ! Hopefully I have Homebridge... THANK YOU! :)

AbeltjeNL, 2021-09-11:
Just disable the temperature entity for the device in HA. Then it won't show up in HomeKit. Did this in my setup and also for some motion sensors because the readings are crap.

closed 2021-10-20 by github-actions[bot]