Koenkk/zigbee2mqtt #8686: How to add a the device BSEED Zigbee Switch, whiche is a clone of an existing (ZM-L03E-Z)?

bmaehr, 2021-09-07:
How to add a new device, which is a clone of an existing?

https://www.aliexpress.com/item/1005002637917367.html is (according to zigbee2mqtt) a clone of ZM-L03E-Z but because of the nice design, the good price, the different types and the availability I think it is worth to mention it.

Koenkk, 2021-09-08:
When pairing this device, does it show up as supported?

bmaehr, 2021-09-08:
```
info  2021-08-31 00:58:18: Successfully interviewed '0x60a423fffe459632', device has successfully been paired
info  2021-08-31 00:58:18: Device '0x60a423fffe459632' is supported, identified as: Zemismart Smart light switch - 3 gang with neutral wire (ZM-L03E-Z)
info  2021-08-31 00:58:18: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"Smart light switch - 3 gang with neutral wire","exposes":[{"endpoint":"left","features":[{"access":7,"description":"On/off state of the switch","endpoint":"left","name":"state","property":"state_left","type":"binary","value_off":"OFF","value_on":"ON","value_toggle":"TOGGLE"}],"type":"switch"},{"endpoint":"center","features":[{"access":7,"description":"On/off state of the switch","endpoint":"center","name":"state","property":"state_center","type":"binary","value_off":"OFF","value_on":"ON","value_toggle":"TOGGLE"}],"type":"switch"},{"endpoint":"right","features":[{"access":7,"description":"On/off state of the switch","endpoint":"right","name":"state","property":"state_right","type":"binary","value_off":"OFF","value_on":"ON","value_toggle":"TOGGLE"}],"type":"switch"},{"access":1,"description":"Link quality (signal strength)","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"ZM-L03E-Z","supports_ota":false,"vendor":"Zemismart"},"friendly_name":"0x60a423fffe459632","ieee_address":"0x60a423fffe459632","status":"successful","supported":true},"type":"device_interview"}'
info  2021-08-31 00:58:18: Configuring '0x60a423fffe459632'
info  2021-08-31 00:58:18: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_successful","meta":{"description":"Smart light switch - 3 gang with neutral wire","friendly_name":"0x60a423fffe459632","model":"ZM-L03E-Z","supported":true,"vendor":"Zemismart"},"type":"pairing"}'
info  2021-08-31 00:58:18: MQTT publish: topic 'homeassistant/switch/0x60a423fffe459632/switch_left/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"command_topic":"zigbee2mqtt/0x60a423fffe459632/left/set","device":{"identifiers":["zigbee2mqtt_0x60a423fffe459632"],"manufacturer":"Zemismart","model":"Smart light switch - 3 gang with neutral wire (ZM-L03E-Z)","name":"0x60a423fffe459632","sw_version":"Zigbee2MQTT 1.21.0"},"json_attributes_topic":"zigbee2mqtt/0x60a423fffe459632","name":"0x60a423fffe459632 left","payload_off":"OFF","payload_on":"ON","state_topic":"zigbee2mqtt/0x60a423fffe459632","unique_id":"0x60a423fffe459632_switch_left_zigbee2mqtt","value_template":"{{ value_json.state_left }}"}'
info  2021-08-31 00:58:18: MQTT publish: topic 'homeassistant/switch/0x60a423fffe459632/switch_center/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"command_topic":"zigbee2mqtt/0x60a423fffe459632/center/set","device":{"identifiers":["zigbee2mqtt_0x60a423fffe459632"],"manufacturer":"Zemismart","model":"Smart light switch - 3 gang with neutral wire (ZM-L03E-Z)","name":"0x60a423fffe459632","sw_version":"Zigbee2MQTT 1.21.0"},"json_attributes_topic":"zigbee2mqtt/0x60a423fffe459632","name":"0x60a423fffe459632 center","payload_off":"OFF","payload_on":"ON","state_topic":"zigbee2mqtt/0x60a423fffe459632","unique_id":"0x60a423fffe459632_switch_center_zigbee2mqtt","value_template":"{{ value_json.state_center }}"}'
info  2021-08-31 00:58:18: MQTT publish: topic 'homeassistant/switch/0x60a423fffe459632/switch_right/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"command_topic":"zigbee2mqtt/0x60a423fffe459632/right/set","device":{"identifiers":["zigbee2mqtt_0x60a423fffe459632"],"manufacturer":"Zemismart","model":"Smart light switch - 3 gang with neutral wire (ZM-L03E-Z)","name":"0x60a423fffe459632","sw_version":"Zigbee2MQTT 1.21.0"},"json_attributes_topic":"zigbee2mqtt/0x60a423fffe459632","name":"0x60a423fffe459632 right","payload_off":"OFF","payload_on":"ON","state_topic":"zigbee2mqtt/0x60a423fffe459632","unique_id":"0x60a423fffe459632_switch_right_zigbee2mqtt","value_template":"{{ value_json.state_right }}"}'
info  2021-08-31 00:58:18: MQTT publish: topic 'homeassistant/sensor/0x60a423fffe459632/linkquality/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0x60a423fffe459632"],"manufacturer":"Zemismart","model":"Smart light switch - 3 gang with neutral wire (ZM-L03E-Z)","name":"0x60a423fffe459632","sw_version":"Zigbee2MQTT 1.21.0"},"enabled_by_default":false,"icon":"mdi:signal","json_attributes_topic":"zigbee2mqtt/0x60a423fffe459632","name":"0x60a423fffe459632 linkquality","state_class":"measurement","state_topic":"zigbee2mqtt/0x60a423fffe459632","unique_id":"0x60a423fffe459632_linkquality_zigbee2mqtt","unit_of_measurement":"lqi","value_template":"{{ value_json.linkquality }}"}'
info  2021-08-31 00:58:18: Successfully configured '0x60a423fffe459632'
```

Koenkk, 2021-09-10:
Thanks, could you also provide the data/database.db entry of this device?

bmaehr, 2021-09-10:
Of course. Thank you for your work!

```
{
  "id": 7,
  "type": "Router",
  "ieeeAddr": "0x60a423fffe459632",
  "nwkAddr": 34993,
  "manufId": 4098,
  "manufName": "_TZ3000_kflqsrse",
  "powerSource": "Mains (single phase)",
  "modelId": "TS0003",
  "epList": [1, 2, 242, 3],
  "endpoints": {
    "1": {
      "profId": 260,
      "epId": 1,
      "devId": 256,
      "inClusterList": [0, 3, 4, 5, 6, 57344, 57345],
      "outClusterList": [25, 10],
      "clusters": {
        "genOnOff": {
          "attributes": {
            "onOff": 0
          }
        },
        "genBasic": {
          "attributes": {
            "modelId": "TS0003",
            "manufacturerName": "_TZ3000_kflqsrse",
            "powerSource": 1,
            "zclVersion": 3,
            "appVersion": 71,
            "stackVersion": 0,
            "hwVersion": 1,
            "dateCode": ""
          }
        }
      },
      "binds": [{
          "cluster": 6,
          "type": "endpoint",
          "deviceIeeeAddress": "0x00124b0021b77edf",
          "endpointID": 1
        }
      ],
      "configuredReportings": [],
      "meta": {}
    },
    "2": {
      "profId": 260,
      "epId": 2,
      "devId": 256,
      "inClusterList": [3, 4, 5, 6, 57344, 57345],
      "outClusterList": [],
      "clusters": {
        "genOnOff": {
          "attributes": {
            "onOff": 1
          }
        }
      },
      "binds": [{
          "cluster": 6,
          "type": "endpoint",
          "deviceIeeeAddress": "0x00124b0021b77edf",
          "endpointID": 1
        }
      ],
      "configuredReportings": [],
      "meta": {}
    },
    "3": {
      "profId": 260,
      "epId": 3,
      "devId": 256,
      "inClusterList": [3, 4, 5, 6, 57344, 57345],
      "outClusterList": [],
      "clusters": {
        "genOnOff": {
          "attributes": {
            "onOff": 0
          }
        }
      },
      "binds": [{
          "cluster": 6,
          "type": "endpoint",
          "deviceIeeeAddress": "0x00124b0021b77edf",
          "endpointID": 1
        }
      ],
      "configuredReportings": [],
      "meta": {}
    },
    "242": {
      "profId": 41440,
      "epId": 242,
      "devId": 97,
      "inClusterList": [],
      "outClusterList": [33],
      "clusters": {},
      "binds": [],
      "configuredReportings": [],
      "meta": {}
    }
  },
  "appVersion": 71,
  "stackVersion": 0,
  "hwVersion": 1,
  "dateCode": "",
  "zclVersion": 3,
  "interviewCompleted": true,
  "meta": {
    "configured": 1
  },
  "lastSeen": 1631297673973
}
```
I could see any attribut beside the `0x60a423fffe459632` matching to the data in the log above so hopfully this is the right entry.

Koenkk, 2021-09-11:
Thanks, added. Will be automatically added to the docs on the next release.

closed 2021-09-11 by Koenkk