Koenkk/zigbee2mqtt #8689: Please add this device : NodOn SIN-4-2-20

XLDom, 2021-09-08:
Hello,

Can you integrate this device please. Thanks :-)

## Information about the device + link
2 Relays Light switch 2 x 690W
https://nodon.fr/nodon/module-eclairage-on-off-zigbee/
[image]

## data/database.db entry of the device
{"id":2,"type":"Router","ieeeAddr":"0x5c0272fffe7d3675","nwkAddr":12021,"manufId":4747,"manufName":"NodOn","powerSource":"Mains (single phase)","modelId":"SIN-4-2-20","epList":[1,2,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[0,3,4,5,6,7,8,4096,64599],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"SIN-4-2-20","manufacturerName":"NodOn","powerSource":1,"zclVersion":3,"appVersion":17,"stackVersion":103,"hwVersion":0,"dateCode":"2020"}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0022982996","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":3600,"repChange":0}],"meta":{}},"2":{"profId":260,"epId":2,"devId":256,"inClusterList":[0,3,4,5,6,7,8],"outClusterList":[3],"clusters":{"genBasic":{"attributes":{"hwVersion":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0022982996","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":3600,"repChange":0}],"meta":{}},"242":{"profId":41440,"epId":242,"devId":102,"inClusterList":[33],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":17,"stackVersion":103,"hwVersion":0,"dateCode":"2020","zclVersion":3,"interviewCompleted":true,"meta":{"configured":1620506091},"lastSeen":1631045670295}

sjorge, 2021-09-08:
```node
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
        zigbeeModel: ['SIN-4-2-20'],
        model: 'SIN-4-2-20',
        vendor: 'NodOn',
        description: 'Double LED Relay',
        exposes: [
            e.switch().withEndpoint('l1'), e.switch().withEndpoint('l2'),
        ],
        fromZigbee: [fz.on_off],
        toZigbee: [tz.on_off],
        endpoint: (device) => {
            return {'l1': 1, 'l2': 2};
        },
        meta: {multiEndpoint: true},
        configure: async (device, coordinatorEndpoint, logger) => {
            const ep1 = device.getEndpoint(1);
            const ep2 = device.getEndpoint(2);
            await reporting.bind(ep1, coordinatorEndpoint, ['genOnOff']);
            await reporting.bind(ep2, coordinatorEndpoint, ['genOnOff']);
        },
};

module.exports = definition;

```

Try with this, see https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html on how to use the file.

XLDom, 2021-09-08:
> Try with this, see https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html on how to use the file.
Yes, it's ok. Thanks !

Do you know how to do to import options like "On with timed off" ?
This module have many options

sjorge, 2021-09-08:
> > Try with this, see https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html on how to use the file.
> > Yes, it's ok. Thanks !
> 
> Do you know how to do to import options like "On with timed off" ?
> This module have many options
If the device supports it, it should already working https://github.com/Koenkk/zigbee-herdsman-converters/blob/715b05f2dfcc25c6d94308780fd88155906562ad/converters/toZigbee.js#L258

Well assuming that the definition I posted earlier today works.

XLDom, 2021-09-08:
Ho other thing, the definition work on direction z2m -> switch : z2m can change the state of the two switch.
but not on direction switch -> z2m : 
When i activate the relay with S1 input. The state is not update on z2m.
Do you know why ?

sjorge, 2021-09-08:
Try with this one instead
```node
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const e = exposes.presets;

const definition = {
        zigbeeModel: ['SIN-4-2-20'],
        model: 'SIN-4-2-20',
        vendor: 'NodOn',
        description: 'Double LED Relay',
        exposes: [
            e.switch().withEndpoint('l1'), e.switch().withEndpoint('l2'),
        ],
        fromZigbee: [fz.on_off],
        toZigbee: [tz.on_off],
        endpoint: (device) => {
            return {'l1': 1, 'l2': 2};
        },
        meta: {multiEndpoint: true},
        configure: async (device, coordinatorEndpoint, logger) => {
            const ep1 = device.getEndpoint(1);
            const ep2 = device.getEndpoint(2);
            await reporting.bind(ep1, coordinatorEndpoint, ['genOnOff']);
            await reporting.bind(ep2, coordinatorEndpoint, ['genOnOff']);
            await reporting.onOff(ep1);
            await reporting.onOff(ep2);
        },
};

module.exports = definition;
```

XLDom, 2021-09-08:
don't work to.
If i click on refresh : 
[image]
The state is updated on z2m.

Other think, there is no log entry on z2m (debug mode) when i push, or release my button on S1.

sjorge, 2021-09-08:
Did you hit the reconfigure button for the device after updating the file?

Can you also grab a screenshot of the cluster tab in the frontend.

XLDom, 2021-09-08:
ok, i've try to reconfigure : same.
I've try to reset the node, and do a new join : 
Error Failed to configure '0x5c0272fffe7d3675', attempt 1 (Error: Bind 0x5c0272fffe7d3675/1 genOnOff from '0x00124b0022982996/1' failed (AREQ - ZDO - bindRsp after 10000ms) at Timeout._onTimeout (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35) at listOnTimeout (internal/timers.js:557:17) at processTimers (internal/timers.js:500:7))

Here is the cluster tab : 
[image]
[image]
[image]

sjorge, 2021-09-09:
There doesn't seem to be any output cluster on ep1 and ep2 aside from genIdentity so I don't think we can get 'actions' for when the buttons are pressed.

Odd that it's throwing timeouts, what if you add the reporting for onOff manually using the reporting tab?

[image]

Nivek-domotic, 2021-09-09:
Hi sjorge, i read something in the user manual of this device

----------------------------------------------------------------
3.2.5. On/Off
The Device supports the On/Off Cluster:
- As Server to allow the control of the Output 1 through these commands:
o Off
o On
o Toggle
o Off with effect
o On with recall global scene
o On with timed off

...
-------------------------------------------------------------
I'm all right with this on/off/toogle, but it's the first time i seen the other option

So how can we use the other options?

sjorge, 2021-09-09:
By sending the extra property in the payload as mentioned https://github.com/Koenkk/zigbee2mqtt/issues/8689#issuecomment-915510685

sjorge, 2021-09-09:
```bash
mosquitto_pub -t zigbee2mqtt/0x0123456789/set -m '{"state": "ON", "on_time": 60}'
```

Should turn it on for 1 minute

XLDom, 2021-09-09:
> ```shell
> mosquitto_pub -t zigbee2mqtt/0x0123456789/set -m '{"state": "ON", "on_time": 60}'
> ```
Hello,

ok this is working. The relay stay ON and go to OFF after the on_time.

I don't understand why there is no report on direction node -> z2m.
I've change max rep on 900. (i don't know what it's for) : no change for report :)

sjorge, 2021-09-09:
It could be the device doesn’t support reporting.

~ sjorge

XLDom, 2021-09-10:
Hello,
we have test with the node model SIN-4-1-20 and there is the reporting when we activate the S button....
Probable a firmware problem on my node (i will to test).

For information, i've test the SIN-4-1-20 (1 relay node) with this (and it's ok): 
```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const e = exposes.presets;

const definition = {
        zigbeeModel: ['SIN-4-1-20'],
        model: 'SIN-4-1-20',
        vendor: 'NodOn',
        description: 'Simple LED Relay',
        exposes: [
            e.switch().withEndpoint('l1'),
        ],
        fromZigbee: [fz.on_off],
        toZigbee: [tz.on_off],
        endpoint: (device) => {
            return {'l1': 1};
        },
        meta: {multiEndpoint: true},
        configure: async (device, coordinatorEndpoint, logger) => {
            const ep = device.getEndpoint(1);
            await reporting.bind(ep, coordinatorEndpoint, ['genOnOff']);
            await reporting.onOff(ep);
        },
};

module.exports = definition;
```

Do you see something to change on the definition file ?

How to do a request now ? to improve the 2 nodes definition on z2m ? please

XLDom, 2021-09-10:
Other question : and how to give the OTA firmware update files to z2m ?
thanks

sjorge, 2021-09-10:
>Other question : and how to give the OTA firmware update files to z2m ?
thanks

If you have the firmware files, they need to be added here https://github.com/Koenkk/zigbee-OTA

> we have test with the node model SIN-4-1-20 and there is the reporting when we activate the S button....
> Probably a firmware problem on my node (i will to test to upgrade him this week-end).

Good news! I'll try and get a PR open then against zhc to add support, hopefully you can sort the non responding one.

XLDom, 2021-09-19:
Hello, sorry for that, but since last upgrade, the NodOn is unsupported, and detected like that : 
[image]

In database, i've got : 
`{"id":2,"type":"Router","ieeeAddr":"0x5c0272fffe7d3675","nwkAddr":51949,"manufId":4747,"manufName":"NodOn","powerSource":"Mains (single phase)","modelId":"SIN-4-RS-20\u0000\u0000\u0000\u0000\u0014\u0000\u0000\u0000ï¿½ï¿½\u0000 9\u0002\u0001\u0000ï¿½\u0000\u0000\u0000ï¿½","epList":[1,2,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[0,3,4,5,6,7,8,4096,64599],"outClusterList":[3,6,25],"clusters":{"genBasic":{"attributes":{"modelId":"SIN-4-RS-20\u0000\u0000\u0000\u0000\u0014\u0000\u0000\u0000ï¿½ï¿½\u0000 9\u0002\u0001\u0000ï¿½\u0000\u0000\u0000ï¿½","manufacturerName":"NodOn","powerSource":1,"zclVersion":3,"appVersion":35,"stackVersion":103,"hwVersion":0,"dateCode":"2020"}}},"binds":[],"configuredReportings":[],"meta":{}},"2":{"profId":260,"epId":2,"devId":256,"inClusterList":[0,3,4,5,6,7,8],"outClusterList":[3,6],"clusters":{"genBasic":{"attributes":{"hwVersion":0}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":102,"inClusterList":[33],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":35,"stackVersion":103,"hwVersion":0,"dateCode":"2020","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1632084060087}`

do you have any idea what the problem is ?

sjorge, 2021-09-19:
You’re running on the dev branch? It hasn’t made it into a release yet.

XLDom, 2021-09-19:
> You’re running on the dev branch? It hasn’t made it into a release yet.

yes in dev branch.
My problem is the appearance of these characters:
before : "modelId":"SIN-4-RS-20"
after upgrade of NodOn firmware (or z2m ? i don't know) : `"modelId":"SIN-4-RS-20\u0000\u0000\u0000\u0000\u0014\u0000\u0000\u0000ï¿½ï¿½\u0000 9\u0002\u0001\u0000ï¿½\u0000\u0000\u0000ï¿½"`

sjorge, 2021-09-19:
Hmmm yeah that not going to March the device code then. How did you update the firmware? Seems like there is some random data behind the model now.

XLDom, 2021-09-19:
Last firmware and is push with android EFR with bluetooth.
You confirm to me that this data comes from the equipment (the node) and not from zigbee2mqtt? (if is the case, i will ask to NodOn)

sjorge, 2021-09-19:
The data is send (well, actually read) during interview from the genBasic cluster so z2m can identify it.

XLDom, 2021-09-19:
> You can also read it via the dev console, e.g. for a random ikea bulb I have: `Read result of 'genBasic': {"modelId":"TRADFRI bulb GU10 WS 400lm","manufacturerName":"IKEA of Sweden"}`

yes, i've try
but the answer is : 
[image]

(lol)

i will ask to NodOn if they have an idea

sjorge, 2021-09-19:
@Koenkk can we allow the raw read converters for unsupported devices?

XLDom, 2021-10-12:
Hello,

the NodOn Developper don't find solution for the `\u0000` caracters after the modelId.

Into zigbee2mqtt, can you do to stop the modelId before the first `\u0000` find ?

i that can help, on the legrand.js there is similar witout random : 
```
module.exports = [
    {
        zigbeeModel: [' Contactor\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000'+
            '\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000'],
        model: 'FC80CC',
```

XLDom, 2021-10-12:
Hello @Koenkk

i've try, i'm on last dev, and the result is : 
[image]

mrgreenfr, 2021-10-14:
> Is this the roller shutter module? Your device has modelId `SIN-4-RS-20` which matches with https://shop.smarthome-europe.com/en/peripheriques/4923-nodon-zigbee-30-roller-shutter-module-3700313925201.html

God... my bad. The roller shutter module is not in the definitions files.
Yes it is. Can you add him into the converter please ?
Construtor link : https://nodon.fr/nodon/module-volet-roulant-zigbee/
Image link : https://nodon.fr/wp-content/uploads/2021/04/SIN-4-RS-20_ISO-600x600.jpg

Thanks

closed 2021-11-22 by github-actions[bot]

titou4307, 2022-12-25:
Hi...
Is the firmware of the Nodon products can be updated by the zigbee2mqtt network ??

Best regard

furieuxjk, 2023-04-13:
@XLDom - did you managed to get SIN-4-1-20 device reporting the state if changed by external switch (without refreshing Z2M exposes page)?

Another way how to manually refresh is by just sending json payload {"state": ""} to the MQTT topic zigbee2mqtt/FRIENDLY_NAME/get - then topic "state" is updated.

Regards