Koenkk/zigbee2mqtt #8690: zigbee2mqtt freezes from time to time

kalatos86, 2021-09-08:
Hi there!

Since the last update of the HA OS, Ha Core and zigbee2mqtt I am struggling with the issue of stability. I mean that from time to time (randomly) it seems for some reasons I am not able to control the devices. After some period of time everything back to the normal and stable work.

Here is the set of the versions I am currently using:

- Zigbee2mqtt Current version: 1.21.1-1 (Changelog)
- HA  core-2021.9.4 
- HA supervisor-2021.09.0
- Home Assistant OS 6.3
- Conbee II with the newest firmware (August 2021)

I have a lot of such logs as shared below consist of all the devices I am trying to call:

_Zigbee2MQTT:error 2021-09-08 09:16:27: Publish 'set' 'state' to 'WLWentylacjaWCDol' failed: 'Error: Command 0x84fd27fffe80c42d/1 genOnOff.off({}, {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received)'_

What is also new here and might be useful is that. Since I upgraded the HA to the one of the latest version is the new integration visible with Conbee II 

[image]

After trying to configure it:

[image]

[image]

sjorge, 2021-09-08:
If you want to use zigbee2mqtt you definitely want to hit ignore on the 'Zigbee Home Automation' that is the ha's native zigbee integration, which will then grab the USB device before zigbee2mqtt can. Or if zigbee2mqtt already is running zha will fail to grab it.

It shows up due to the recent added USB discovery.

kalatos86, 2021-09-08:
I did exactly as you wrote. I mentioned about it because I through i might influence stable way of working of zigbee devices. Even though I did ignore on  'Zigbee Home Automation' from time to time I am struggling with stability of zigbee2mqtt. Do you have idea why?

Koenkk, 2021-09-08:
I don't know what usb-to-serial chip the Conbee II uses, but hassos 6.3 is affected by https://github.com/Koenkk/zigbee2mqtt/issues/8663, I would suggest to first update to hassos 6.4 or downgrade to 6.2

kalatos86, 2021-09-08:
Thank you so mych for info. For me the hasos is not yet available:

[image]

dekamilv, 2021-09-12:
> I don't know what usb-to-serial chip the Conbee II uses, but hassos 6.3 is affected by #8663, I would suggest to first update to hassos 6.4 or downgrade to 6.2

On 6.4 same problems with freezes .... Thinking to downgrade to 6.2.

closed 2021-10-20 by github-actions[bot]