Koenkk/zigbee2mqtt #8691: [zStackEraseAllNvMem] UnhandledPromiseRejectionWarning: Error: SRSP - SYS - nvDelete after 0,9ms

bruvv, 2021-09-08:
## What happened
I upgraded the firmware on my `cc2652p_E72-2G4M20S1E` stick which did work out:
```
Invoking bootloader by method 1 (default)
Opening port /dev/ttyUSB0, baud 500000
Reading data from CC1352P2_CC2652P_other_coordinator_20210708.hex
Your firmware looks like an Intel Hex file
Connecting to target...
CC1350 PG2.0 (7x7mm): 352KB Flash, 20KB SRAM, CCFG.BL_CONFIG at 0x00057FD8
Primary IEEE Address: 00:12:4B:00:22:AB:78:56
    Performing mass erase
Erasing all main bank flash sectors
    Erase done
Writing 360448 bytes starting at address 0x00000000
Write 104 bytes at 0x00057F980
    Write done
Verifying by comparing CRC32 calculations.
    Verified (match: 0x39d15b77)
```

After I ran the zStackEraseAllNvMem.js in a freshly created docker container that I created as followed:
`docker run -t -i --device=/dev/ttyUSB0 ubuntu bash`

But I get the following error:
```
node zStackEraseAllNvMem.js /dev/ttyUSB0
Detected znp version 'zStack3x0' ({"transportrev":2,"product":1,"majorrel":2,"minorrel":7,"maintrel":1,"revision":20210708})
Clearing all NVMEM items, from 0 to 815
NVMEM item #4 - deleting, size: 20
(node:25) UnhandledPromiseRejectionWarning: Error: SRSP - SYS - nvDelete after 0,9ms
    at Timeout.waiter.timer.setTimeout [as _onTimeout] (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/utils/waitress.js:37:35)
    at ontimeout (timers.js:436:11)
    at tryOnTimeout (timers.js:300:5)
    at listOnTimeout (timers.js:263:5)
    at Timer.processTimers (timers.js:223:10)
(node:25) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). (rejection id: 1)
(node:25) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.
```

## What did you expect to happen
That it finishes the script with a message similar to `succes`

## How to reproduce it (minimal and precise)
1: install hassos on any client ( I am running it on a HP T620)
2: enable SSH to get access to the underlying OS (port 2222)
3: install ubuntu: `docker run -t -i --device=/dev/ttyUSB0 ubuntu bash`
4: within ubuntu container follow: https://www.zigbee2mqtt.io/getting_started/running_zigbee2mqtt.html#2-installing
5: run: `node /opt/zigbee2mqtt/scripts/zStackEraseAllNvMem.js /dev/ttyUSB0

## Debug info
Zigbee2MQTT version: https://github.com/Koenkk/zigbee2mqtt/commit/4a51e0c0c99dfeb4b319048db270c9eb69db419e
Adapter hardware:  `cc2652p_E72-2G4M20S1E`
Adapter firmware version: ` CC1352P2_CC2652P_other_coordinator_20210708.zip `

bruvv, 2021-09-30:
@Koenkk any update on this?

Koenkk, 2021-09-30:
@Hacker-CB can you check this?