Koenkk/zigbee2mqtt #8697: Neo NAS-PD07 sensor

Masterz69, 2021-09-09:
Have device not yet included in Z2M: Neo NAS-PD07 motion, temp&humidity sensor.
I'm using HA OS installation. I've created .js file for it and included fz.tuya_data_point_dump converter.
Unfortunately it crashing HA Z2M Add-in with:
```
Error: ENOENT: no such file or directory, open 'data/tuya.dump.txt'
[21:11:38] INFO: Handing over control to Zigbee2mqtt Core ...
```
Reported that issue in [HA add-ons GitHub](https://github.com/home-assistant/addons/issues/2175).

From logs I've extracted: 
```
debug 2021-09-09 08:41:29: No converter available for 'NAS-PD07' with cluster 'manuSpecificTuya' and type 'commandSetTimeRequest' and data '{"payloadSize":6912}'

[dp 104] - temperature x10 IMHO

debug 2021-09-09 13:14:41: Received Zigbee message from '0x842e14fffee351a0', type 'commandSetDataResponse', cluster 'manuSpecificTuya', 
data '{"data":{"data":[0,0,0,189],"type":"Buffer"},"datatype":2,"dp":104,"fn":0,"status":0,"transid":0}' from endpoint 1 with groupID 0

debug 2021-09-09 13:14:41: No converter available for 'NAS-PD07' with cluster 'manuSpecificTuya' and type 'commandSetDataResponse' and 
data '{"data":{"data":[0,0,0,189],"type":"Buffer"},"datatype":2,"dp":104,"fn":0,"status":0,"transid":0}'

[dp 105] - humidity IMHO

debug 2021-09-09 13:18:30: Received Zigbee message from '0x842e14fffee351a0', type 'commandSetDataResponse', cluster 'manuSpecificTuya', 
data '{"data":{"data":[0,0,0,64],"type":"Buffer"},"datatype":2,"dp":105,"fn":0,"status":0,"transid":0}' from endpoint 1 with groupID 0

debug 2021-09-09 13:18:30: No converter available for 'NAS-PD07' with cluster 'manuSpecificTuya' and type 'commandSetDataResponse' and 
data '{"data":{"data":[0,0,0,64],"type":"Buffer"},"datatype":2,"dp":105,"fn":0,"status":0,"transid":0}'

[dp 101] - definetely motion : 1 - detected, 0 - no

debug 2021-09-09 13:18:29: Received Zigbee message from '0x842e14fffee351a0', type 'commandSetDataResponse', cluster 'manuSpecificTuya', 
data '{"data":{"data":[1],"type":"Buffer"},"datatype":1,"dp":101,"fn":0,"status":0,"transid":0}' from endpoint 1 with groupID 0

debug 2021-09-09 13:18:29: No converter available for 'NAS-PD07' with cluster 'manuSpecificTuya' and type 'commandSetDataResponse' and 
data '{"data":{"data":[1],"type":"Buffer"},"datatype":1,"dp":101,"fn":0,"status":0,"transid":0}'

debug 2021-09-09 13:19:42: Received Zigbee message from '0x842e14fffee351a0', type 'commandSetDataResponse', cluster 'manuSpecificTuya', 
data '{"data":{"data":[0],"type":"Buffer"},"datatype":1,"dp":101,"fn":0,"status":0,"transid":0}' from endpoint 1 with groupID 0

debug 2021-09-09 13:19:42: No converter available for 'NAS-PD07' with cluster 'manuSpecificTuya' and type 'commandSetDataResponse' and 
data '{"data":{"data":[0],"type":"Buffer"},"datatype":1,"dp":101,"fn":0,"status":0,"transid":0}'
```

Not sure about possible commands to sensor or deice-specific settings.
If this information can help to add device support - thank You very much.

NAS-PD07.js:
```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    fingerprint: [
        { modelID: 'TS0601', manufacturerName: '_TZE200_7hfcudw5' },
    ],
    model: 'NAS-PD07',
    vendor: 'Neo',
    description: 'Motion, temperature & humidity sensor',
    fromZigbee: [ fz.tuya_data_point_dump ],
    toZigbee: [],
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genBasic']);
    },
    exposes: [],
};

module.exports = definition;
```

[Device pictures available here](https://photos.app.goo.gl/TccFb6eqMzgtSfo2A).
[Seller information about device](https://aliexpress.ru/item/1005002339649035.html).

mrTuomoK, 2021-09-14:
I guess this is the same issue --> https://github.com/Koenkk/zigbee2mqtt/issues/7915

Masterz69, 2021-09-14:
> I guess this is the same issue --> #7915 (Closed though)

Yes, looks same device.
Unfortunately it still not added to Z2M :( I'm not so experienced to do it by myself.

Masterz69, 2021-09-14:
database.db entry:
```txt
{"id":25,"type":"EndDevice","ieeeAddr":"0x842e14fffee351a0","nwkAddr":40834,"manufId":4098,"manufName":"_TZE200_7hfcudw5","modelId":"TS0601",
"epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,4,5,61184],"outClusterList":[25,10],
"clusters":{"genBasic":{"attributes":{"65503":"9D�(\u0019�D�(\u0019)E�(\u0019�E�(\u0019\u0019F�(\u0019�F�(\u0019\tG�(\u0019�G�(\u0019","65506":31,
  "65508":0,"modelId":"TS0601","manufacturerName":"_TZE200_7hfcudw5","appVersion":85,"zclVersion":3,"stackVersion":0,"hwVersion":1,"dateCode":""}}},
"binds":[{"cluster":0,"type":"endpoint","deviceIeeeAddress":"0x00124b0022812732","endpointID":1}],"configuredReportings":[],"meta":{}}},
"appVersion":85,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":821693351},"lastSeen":1631614399122}
```

plysdyret, 2021-09-21:
I don't really know how to add it either but I hobbled together a pull request [1], based on this issue as well as info picked up from #7915 and github/zigbee-herdsman-converters#1179. 

[1] https://github.com/Koenkk/zigbee-herdsman-converters/pull/3064

closed 2021-10-11 by Masterz69