Koenkk/zigbee2mqtt #8698: Moes BHT-002 Thermostat not showing running_state

KameDomotics, 2021-09-09:
## What happened
In HomeAssistant thermostat and Zigbee2mqtt is not showing me the running_state

## What did you expect to happen
running_state: The current running state. Possible values are: idle, heat, cool.

## How to reproduce it (minimal and precise)
Send a message to zigbee2mqtt/FRIENDLY_NAME/get 
with payload {"running_state": ""}
## Debug info
Zigbee2MQTT version: 1.21.1-1
Adapter hardware: CC2538
Adapter firmware version:

KameDomotics, 2021-09-09:
[image]
[image]
[image]

Plawasan, 2021-09-17:
That would probably explain this error that keeps showing up in my logs:

Logger: homeassistant.helpers.template
Source: helpers/template.py:1485
First occurred: 8:47:01 (24 occurrences)
Last logged: 9:06:17

Template variable warning: dict object has no element Undefined when rendering '{% set values = {'idle':'off','heat':'heating','cool':'cooling','fan only':'fan'} %}{{ values[value_json.running_state] }}'

KameDomotics, 2021-09-17:
Is there any way to fix this?

Safs, 2021-09-20:
Same problem for me too...

rizer1980, 2021-09-20:
same problem. log file grow up hundred megabytes...
5 BHT-002-GCLZB thermostat
with debounce:
Logger: homeassistant.helpers.template
Source: helpers/template.py:1485
First occurred: 1:26:56 (3682 occurrences)
Last logged: 10:41:51

Template variable warning: dict object has no element Undefined when rendering '{% set values = {'idle':'off','heat':'heating','cool':'cooling','fan only':'fan'} %}{{ values[value_json.running_state] }}'

KameDomotics, 2021-09-25:
> same problem. log file grow up hundred megabytes...
> 5 BHT-002-GCLZB thermostat
> with debounce:
> Logger: homeassistant.helpers.template
> Source: helpers/template.py:1485
> First occurred: 1:26:56 (3682 [...]

that's right, it gives me the same warning

kocsishome, 2021-09-25:
That's right.  Is there a solution to this?  Thanks

jinnerbichler, 2021-09-25:
Same for me.

The state send over MQTT has a field with the name `head`, which has the value 'ON' or 'OFF'.
So the data for `running_state` in available, but is not sent via `running_state`

```
{
  "child_lock": "UNLOCK",
  "current_heating_setpoint": 12,
  "deadzone_temperature": 1,
  "heat": "OFF",
  "linkquality": 135,
  "local_temperature": 26,
  "local_temperature_calibration": -3,
  "max_temperature": 35,
  "max_temperature_limit": 45,
  "preset": "hold",
  "preset_mode": "hold",
  "program": [
    {
      "weekdays_p1": "6h:0m 20°C"
    },
    {
      "weekdays_p2": "11h:30m 21°C"
    },
    {
      "weekdays_p3": "13h:30m 22°C"
    },
    {
      "weekdays_p4": "17h:30m 23°C"
    },
    {
      "saturday_p1": "6h:0m 24°C"
    },
    {
      "saturday_p2": "12h:0m 23°C"
    },
    {
      "saturday_p3": "14h:30m 22°C"
    },
    {
      "saturday_p4": "17h:30m 21°C"
    },
    {
      "sunday_p1": "6h:0m 19°C"
    },
    {
      "sunday_p2": "12h:30m 20°C"
    },
    {
      "sunday_p3": "14h:30m 21°C"
    },
    {
      "sunday_p4": "18h:30m 20°C"
    }
  ],
  "sensor": "AL",
  "system_mode": "heat"
}
```

KameDomotics, 2021-09-25:
> Same for me.
>
> The state send over MQTT has a field with the name `head`, which has the value 'ON' or 'OFF'.
> So the data for `running_state` in available, but is not sent via `running_state`
>
> ```
> { [...]

In "developer tools" also appears the status `running_state: idle / heat`, so I don't understand why it is not displayed in the card and generates the warning

KameDomotics, 2021-09-25:
> Same for me.
>
> The state send over MQTT has a field with the name `head`, which has the value 'ON' or 'OFF'.
> So the data for `running_state` in available, but is not sent via `running_state`
>
> ```
> { [...]

In "developer tools" also appears the status `running_state: idle / heat`, so I don't understand why it is not displayed in the card and generates the warning

[image]

KameDomotics, 2021-09-25:
[image]

rizer1980, 2021-09-26:
to disable this flood warning
in /config/configuration.yaml add this:

logger:
  default: warning
  logs:
    homeassistant.helpers.template: error

jinnerbichler, 2021-09-26:
I think this issue is addressed in this merge: https://github.com/Koenkk/zigbee-herdsman-converters/pull/3057

KameDomotics, 2021-09-27:
> I think this issue might be addressed in this merge: [Koenkk/zigbee-herdsman-converters#3057](https://github.com/Koenkk/zigbee-herdsman-converters/pull/3057)
> I guess we just have to wait for next release.

That would be really cool!  Thank you very much 😃

KameDomotics, 2021-10-02:
Problem solved with Zigbee2mqtt v1.21.2-1

Thanks!

Plawasan, 2021-10-03:
> Problem solved with Zigbee2mqtt v1.21.2-1
> 
> Thanks!

You mean the dev branch?

closed 2021-10-03 by Koenkk