Koenkk/zigbee2mqtt #8700: Update model Hue Impress outdoor Pedestal

nocodeman55, 2021-09-09:
I managed to add the Hue Impress outdoor Pedestal and it does show up on my device list but it claims to be unsupported. 

I think this is because there might be a new or different version of this product. 

My paired product model: 1745430P7

zigbee2mqtt supported model: 1743130P7 

I did try to add this device myself in a .js file and added to my configuration.yaml as external_cnverter. I could not get zigbee2mqtt to restart so i need some help.

## Information about the device + link

https://www.zigbee2mqtt.io/devices/1743130P7.html

## data/database.db entry of the device

{"id":8,"type":"Router","ieeeAddr":"0x001788010922eef1","nwkAddr":58775,"manufId":4107,"manufName":"Philips","powerSource":"Mains (single phase)","modelId":"1745430P7","epList":[11,242],"endpoints":{"11":{"profId":260,"epId":11,"devId":269,"inClusterList":[0,3,4,5,6,8,4096,768,64513],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"1745430P7","manufacturerName":"Philips","powerSource":1,"zclVersion":2,"appVersion":2,"stackVersion":1,"hwVersion":0,"dateCode":"20190529","swBuildId":"1.56.7_r29434"}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":2,"stackVersion":1,"hwVersion":0,"dateCode":"20190529","swBuildId":"1.56.7_r29434","zclVersion":2,"interviewCompleted":true,"meta":{},"lastSeen":1631218332859}

danieledwardgeorgehitchcock, 2021-09-10:
Could you post up your external converter file?

nocodeman55, 2021-09-10:
Thank you for your reply.

This is the code i have tested:

const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['1745430P7'],
    model: '1745430P7',
    vendor: 'Philips',
    description: 'Hue Impress outdoor Pedestal',
    meta: {turnsOffAtBrightness1: true},
    extend: hueExtend.light_onoff_brightness_colortemp_color(),
    ota: ota.zigbeeOTA,
};

module.exports = definition;

danieledwardgeorgehitchcock, 2021-09-10:
Two things I have noticed from the above.. 

1. You don't have hueExtend defined, and it is not a part of the standard extend library
2. You don't have ota defined

Try the following code:

```javascript
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');

const hueExtend = {
    light_onoff_brightness: (options={}) => ({
        ...extend.light_onoff_brightness(options),
        toZigbee: extend.light_onoff_brightness(options).toZigbee.concat([tz.hue_power_on_behavior, tz.hue_power_on_error]),
    }),
    light_onoff_brightness_colortemp: (options={}) => ({
        ...extend.light_onoff_brightness_colortemp(options),
        toZigbee: extend.light_onoff_brightness_colortemp(options).toZigbee.concat([tz.hue_power_on_behavior, tz.hue_power_on_error]),
    }),
    light_onoff_brightness_color: (options={}) => ({
        ...extend.light_onoff_brightness_color({supportsHS: true, ...options}),
        toZigbee: extend.light_onoff_brightness_color({supportsHS: true, ...options}).toZigbee
            .concat([tz.hue_power_on_behavior, tz.hue_power_on_error]),
    }),
    light_onoff_brightness_colortemp_color: (options={}) => ({
        ...extend.light_onoff_brightness_colortemp_color({supportsHS: true, ...options}),
        toZigbee: extend.light_onoff_brightness_colortemp_color({supportsHS: true, ...options})
            .toZigbee.concat([tz.hue_power_on_behavior, tz.hue_power_on_error]),
    }),
};

const definition = {
    zigbeeModel: ['1745430P7'],
    model: '1745430P7',
    vendor: 'Philips',
    description: 'Hue Impress outdoor Pedestal',
    meta: {turnsOffAtBrightness1: true},
    extend: hueExtend.light_onoff_brightness_colortemp_color(),
    ota: ota.zigbeeOTA,
};

module.exports = definition;
```

nocodeman55, 2021-09-10:
Thank you for your guidance. I have made these changes and all the new Hue lights now works.

danieledwardgeorgehitchcock, 2021-09-10:
Brilliant!

The final step is to find out what the supported colour temperatures are for your device and add it to your device handler. See the guide [here](https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html#31-retrieving-color-temperature-range-only-required-for-lights-which-support-color-temperature) for that

Finally, please contribute your device back into the Zigbee2MQTT codebase. If you need a hand with that, or wish for me to do it, just let me know!

nocodeman55, 2021-09-10:
I will work further on this and i will ofcourse contribute this update to the Zigbee2MQTT database.

closed 2021-09-27 by Koenkk