Koenkk/zigbee2mqtt #8705: 2ch zigbee curtain module - second 'device' not available (TS130F)

tomidix, 2021-09-10:
## What happened
I've bought the 2 channel (dual) zb curtain module which is available to connect and control to 2 different motors. Device is recognized as TS130F but is available of controlling only first curtain. (Hardwire switch works fine and can control both motors through the same module)
https://www.aliexpress.com/item/1005003044227983.html

Here is the difference in clusters for normal TS130F switch and dual module:
[image]

## What did you expect to happen
To see both curtains as separate devices and be able to control both

## How to reproduce it (minimal and precise)
Just connecting the specified module to zigbee network and see that second curtain cant be controlled

## Debug info
Zigbee2MQTT version: 1.21.1-1 (home assistant)
Adapter hardware: CC2531
Adapter firmware version: /

I would at least need a hint on what should I do for solution of the problem? Do I create a support for new device, or should I expand existing one, how?

danieledwardgeorgehitchcock, 2021-09-10:
Can you share your database.db entry for this device?

My guess is it will need its own device handler

tomidix, 2021-09-10:
This looks like the entry:

`{"id":23,"type":"Router","ieeeAddr":"0xa4c138365b8c8367","nwkAddr":27480,"manufId":4417,"manufName":"_TZ3000_j1xl73iw","powerSource":"Mains (single phase)","modelId":"TS130F","epList":[1,2,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":514,"inClusterList":[4,5,6,258,0],"outClusterList":[25,10],"clusters":{"genOnOff":{"attributes":{"tuyaBacklightMode":0}},"closuresWindowCovering":{"attributes":{"32768":0,"currentPositionLiftPercentage":78,"tuyaMovingState":1,"tuyaCalibration":1,"tuyaMotorReversal":0,"moesCalibrationTime":100}},"genBasic":{"attributes":{"modelId":"TS130F","manufacturerName":"_TZ3000_j1xl73iw","powerSource":1,"zclVersion":3,"appVersion":66,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}},"2":{"profId":260,"epId":2,"devId":514,"inClusterList":[4,5,6,258],"outClusterList":[],"clusters":{"closuresWindowCovering":{"attributes":{"32768":0,"currentPositionLiftPercentage":0,"tuyaMovingState":1,"tuyaCalibration":1,"tuyaMotorReversal":0,"moesCalibrationTime":100}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":66,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1631271971062}`

danieledwardgeorgehitchcock, 2021-09-10:
Could you try this external converter? - I've just thrown it together so, may need a bit of work...

```javascript
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    fingerprint: [{modelID: 'TS130F', manufacturerName: '_TZ3000_j1xl73iw'}],
    model: 'TS130F',
    vendor: 'TuYa',
    description: 'Dual urtain/blind switch',
    fromZigbee: [fz.cover_position_tilt, fz.tuya_backlight_mode, fz.tuya_cover_options],
    toZigbee: [tz.cover_state, tz.cover_position_tilt, tz.tuya_cover_calibration, tz.tuya_cover_reversal, tz.tuya_backlight_mode],
    meta: {multiEndpoint: true},
    endpoint: (device) => {
        return {'l1': 1, 'l2': 1};
    },
    exposes: [
        e.cover_position().withEndpoint('l1'),
        exposes.enum('moving', ea.STATE, ['UP', 'STOP', 'DOWN']).withEndpoint('l1'),
        e.cover_position().withEndpoint('l2'),
        exposes.enum('moving', ea.STATE, ['UP', 'STOP', 'DOWN']).withEndpoint('l2'),
        exposes.binary('calibration', ea.ALL, 'ON', 'OFF'), exposes.binary('motor_reversal', ea.ALL, 'ON', 'OFF'),
        exposes.enum('backlight_mode', ea.ALL, ['LOW', 'MEDIUM', 'HIGH']),
        exposes.numeric('calibration_time', ea.STATE).withUnit('S').withDescription('Calibration time')],
 };
 
module.exports = definition;
```

tomidix, 2021-09-10:
Thank you very much for the help, couldn't try it earlier.

I did the steps you have mentioned but zigbee2mqtt fails to startup correctly with the following error:
```
(node:1073) UnhandledPromiseRejectionWarning: AssertionError [ERR_ASSERTION]: Endpoint not supported for cover type
    at HomeAssistant.exposeToConfig (/app/lib/extension/homeassistant.js:283:13)
    at HomeAssistant.populateMapping (/app/lib/extension/homeassistant.js:591:54)
    at new HomeAssistant (/app/lib/extension/homeassistant.js:64:14)
    at new Controller (/app/lib/controller.js:83:34)
    at start (/app/index.js:94:18)
...
```

Seems that zigbee2mqtt/homeassistant does not expect second endpoint on a cover device.. 😔

danieledwardgeorgehitchcock, 2021-09-10:
> Thank you very much for the help, couldn't try it earlier.
>
> I did the steps you have mentioned but zigbee2mqtt fails to startup correctly with the following error:
>
> ```
> (node:1073) [...]

D’oh! If nobody jumps on that new issue, I’ll have a look at it on Monday (I’m away for the weekend)

tomidix, 2021-09-11:
@Koenkk jumped in and I was able to add your converter.
I have been experimenting with it because of course, like we expected, it doesn't fully work.

First I saw that motor reversal and calibration on off works as it should for both endpoints.

Then I saw that position for endpoints does not update if it is changed by button in interface (open, close, stop) or by real hardwired switch. Only when it is set via position, it will 'remember' the value in zigbee2mqtt. 

Then I checked at the device state and even if you have exposed only l1 and l2 in converter.. there are also states which are available on ordinary, 1ch switch; state looked like this (I've removed the backlight part as it's a module and doesn't have it):
```json
{
    "calibration": "OFF",
    "calibration_l1": "OFF",
    "calibration_l2": "OFF",
    "calibration_time": 12.8,
    "calibration_time_l1": null,
    "calibration_time_l2": null,
    "linkquality": 36,
    "motor_reversal": "OFF",
    "motor_reversal_l1": "OFF",
    "motor_reversal_l2": "OFF",
    "moving": "STOP",
    "moving_l1": null,
    "moving_l2": null,
    "position": 6,
    "position_l1": 6,
    "position_l2": 28
}
```

by playing with the module I've found that all the values without sufix (_l1 or _l2) are the last value updated from either endpoint.

If I change position (doesn't matter how it's changed) of first cover, then it will write the value in 'position' and update the 'moving' accordingly; meanwhile its _l1 fields values stay 'null'.
If I do that for second cover afterwards, the same fields will be overwritten while it's fields will also stay 'null'.

On the other hand setting all of the values works fine and does the job.

It all looks like this currently (I've exposed moving and calibration_time without endpoints so I can track what happens more easily):
[image]

I'm really new at this but it seems to me that some behavior which is used from imported converters dictates this read behavior and some code should be written specifically for this device. I will look more into this but not sure if I'll be able to do it on my own.

Here is my current converter:
<details>
<summary>Code</summary>
<p>

```javascript
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    fingerprint: [{modelID: 'TS130F', manufacturerName: '_TZ3000_j1xl73iw'}],
    model: 'TS130F',
    vendor: 'TuYa',
    description: 'Dual curtain/blind module',
    fromZigbee: [fz.cover_position_tilt, fz.tuya_cover_options],
    toZigbee: [tz.cover_state, tz.cover_position_tilt, tz.tuya_cover_calibration, tz.tuya_cover_reversal],
    meta: {multiEndpoint: true},
    endpoint: (device) => {
        return {'l1': 1, 'l2': 2};
    },
    exposes: [
        exposes.enum('moving', ea.STATE, ['UP', 'STOP', 'DOWN']),
        exposes.numeric('calibration_time', ea.STATE).withUnit('S').withDescription('Calibration time'),

        e.cover_position().withEndpoint('l1'),
        exposes.enum('moving', ea.STATE, ['UP', 'STOP', 'DOWN']).withEndpoint('l1'),
        exposes.binary('calibration', ea.ALL, 'ON', 'OFF').withEndpoint('l1'),
        exposes.binary('motor_reversal', ea.ALL, 'ON', 'OFF').withEndpoint('l1'),
        exposes.numeric('calibration_time', ea.STATE).withUnit('S').withDescription('Calibration time').withEndpoint('l1'),
        
        e.cover_position().withEndpoint('l2'),
        exposes.enum('moving', ea.STATE, ['UP', 'STOP', 'DOWN']).withEndpoint('l2'),
        exposes.binary('calibration', ea.ALL, 'ON', 'OFF').withEndpoint('l2'),
        exposes.binary('motor_reversal', ea.ALL, 'ON', 'OFF').withEndpoint('l2'),
        exposes.numeric('calibration_time', ea.STATE).withUnit('S').withDescription('Calibration time').withEndpoint('l2')
    ],
 };

module.exports = definition;
```

</p>
</details>

danieledwardgeorgehitchcock, 2021-09-11:
Sounds like the fromZigbee converters for some of the clusters need to be suffixed with the endpoint. There is a function to do this and this would need to be added to the ones that only report the last action of either of the endpoints

tomidix, 2021-09-13:
Tried the converter, interesting error pops up in log (and the cover hasn't moved at all):
 _Publish 'set' 'state' to 'DualTuya' failed: 'Error: Command 0xa4c138365b8c8367/1 closuresWindowCovering.downClose({}, {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'No network route' (205))'_

tomidix, 2021-09-13:
Yup, I repaired it again.. Now it seems to send the command but still doesn't update the position. (I've edited my previous comment in a same time when you have posted yours)

tomidix, 2021-10-04:
@Koenkk is there any possibility to update fromZigbee.js so 'tuya_cover_options' and 'cover_position_tilt' support endpoints?
I still can't get the 'get' part to work properly and I think the problem is that it won't read the data from the endpoints.

I'm willing to play it by myself if I could be able to fork and publish my version to home assistant easily (not the dev one)?

closed 2021-11-12 by github-actions[bot]

hopsor, 2021-12-24:
hi @dominicusmento @danieledwardgeorgehitchcock,

I recently got this device too. Has any progress been made? If not I'd like to help. I'll investigate how the external adapters work and come up with my own tests. If I'm able to achieve something I'll share it here

dominicusmento, 2021-12-27:
Hi @hopsor, after some time I've tried the external converter again, (with fully updated zigbee2mqqt on my home assistant) and it seems that it works now.. Status can be set successfully and is correctly updated after the cover position is changed by physical switch.

Below is external converter which I am using currently.. The problem for me is that I can't map the cover to ordinary dashboard cover cards. Because it has its endpoints and not standard 'current_position' attribute. I manage it via zigbee2mqtt interface directly for now, without automations..

<details>
<summary>Code</summary>
<p>
```javascript
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    fingerprint: [{modelID: 'TS130F', manufacturerName: '_TZ3000_j1xl73iw'}],
    model: 'TS130F',
    vendor: 'TuYa',
    description: 'Dual curtain/blind module',
    fromZigbee: [fz.cover_position_tilt, fz.tuya_cover_options],
    toZigbee: [tz.cover_state, tz.cover_position_tilt, tz.tuya_cover_calibration, tz.tuya_cover_reversal],
    meta: {multiEndpoint: true},
    endpoint: (device) => {
        return {'left': 1, 'right': 2};
    },
    exposes: [
        exposes.enum('moving', ea.STATE, ['UP', 'STOP', 'DOWN']),
        exposes.numeric('calibration_time', ea.STATE).withUnit('S').withDescription('Calibration time'),

        e.cover_position().withEndpoint('left'),
        exposes.enum('moving', ea.STATE, ['UP', 'STOP', 'DOWN']).withEndpoint('left'),
        exposes.binary('calibration', ea.ALL, 'ON', 'OFF').withEndpoint('left'),
        exposes.binary('motor_reversal', ea.ALL, 'ON', 'OFF').withEndpoint('left'),
        exposes.numeric('calibration_time', ea.STATE).withUnit('S').withDescription('Calibration time').withEndpoint('left'),
        
        e.cover_position().withEndpoint('right'),
        exposes.enum('moving', ea.STATE, ['UP', 'STOP', 'DOWN']).withEndpoint('right'),
        exposes.binary('calibration', ea.ALL, 'ON', 'OFF').withEndpoint('right'),
        exposes.binary('motor_reversal', ea.ALL, 'ON', 'OFF').withEndpoint('right'),
        exposes.numeric('calibration_time', ea.STATE).withUnit('S').withDescription('Calibration time').withEndpoint('right')
    ],
 };

module.exports = definition;
```

</p>
</details>

hopsor, 2021-12-28:
Thank you @dominicusmento. I'll try it. If I figure out anything I'll share it here

dominicusmento, 2021-12-28:
I've added template covers in my HA configuration.yaml so I can use both covers now in automations / dashboard..

```yaml
# change 0xa4c1383333333333 with your zigbee2mqtt IEEE Address of the device
# change cover.original_cover_id to actual home assistant cover id
# if you DON'T use inverted covers: 
  # 1. exchange close_cover and open_cover places
  # 2. remove '100 -' so the position is calculated accurately 

cover:
  - platform: template
    covers:
      my_curtain_left:
        friendly_name: "My curtain Left"
        open_cover:
          service: cover.close_cover
          data:
            entity_id: cover.original_cover_id
        close_cover:
          service: cover.open_cover
          data:
            entity_id: cover.original_cover_id
        stop_cover:
          service: cover.stop_cover
          data:
            entity_id: cover.original_cover_id
        set_cover_position:
          service: mqtt.publish
          data:
            topic: zigbee2mqtt/0xa4c1383333333333/set
            payload: '{"position_left":{{ 100 - position }} }'
        position_template: "{{ 100 - state_attr('cover.original_cover_id', 'position_left') }}"
      my_curtain_right:
        friendly_name: "My curtain Right"
        open_cover:
          service: cover.close_cover
          data:
            entity_id: cover.original_cover_id
        close_cover:
          service: cover.open_cover
          data:
            entity_id: cover.original_cover_id
        stop_cover:
          service: cover.stop_cover
          data:
            entity_id: cover.original_cover_id
        set_cover_position:
          service: mqtt.publish
          data:
            topic: zigbee2mqtt/0xa4c1383333333333/set
            payload: '{"position_right":{{ 100 - position }} }'
        position_template: "{{ 100 - state_attr('cover.original_cover_id', 'position_right') }}"
```

hopsor, 2022-01-11:
Hi @dominicusmento 

Sorry for my late answer. I tried with a first device and I burnt it by doing a wrong wiring 😅. I had to reorder a new one so it took some time.

I've been able to try exactly what you posted and it worked flawlessly. Thank you very much! Now I've replaced my two wifi Shelly 2.5 with one single zigbee relay and it's working like a charm.

Maybe now it's worth considering opening a PR in [Koenkk/zigbee-herdsman-converters](https://github.com/Koenkk/zigbee-herdsman-converters) so other people can take advantage of this out of the box.

dominicusmento, 2022-01-11:
Opening a PR is fine but I feel that converter can be written better and striped down maybe.. Also after the last z2m update, I don't need cover templates in configuration.yaml of HA. Now both of the covers are available directly from z2m. I'll probably open it as soon as I find time to check few more things. Hopefully this week.