Koenkk/zigbee2mqtt #8707: OTA for Danfoss Ally doesn't find newest version

frauhottelmann, 2021-09-10:
Hey, I have a couple of Danfoss Ally. The one I got first was updated by z2m to firmware date 20200721 and firmware version 01.02.0008 01.02.
And the new one got delivered with firmware date 20210316 and firmware version 01.08.0008 01.08.
The 01.08 version is on their website at https://files.danfoss.com/download/Heating/Ally/Ally%20eTRV%201.08%20OTA%20file.zip. The same version appears to be at https://github.com/Koenkk/zigbee-OTA/tree/master/images/Danfoss.

When I check the OTA for the old one z2m doesn't find the new one.

## Debug info
Zigbee2MQTT version: 1.21.1-1 (Home Assistant Supervised)
Adapter hardware: Slaesh's CC2652RB stick
Adapter firmware version: zStack3x0 20200715

WouterKr, 2021-11-07:
Hi, the Danfoss Ally I am using does not update by OTA either.
Hope this will and can be solved.

frauhottelmann, 2021-12-13:
I was able to update to the new firmware.

closed 2021-12-13 by frauhottelmann