Koenkk/zigbee2mqtt #8709: Aqara Opple WXCJKG12LM and WXCJKG13LM (possibly WXCJKG11LM but I don't have those) aren't registering double click, hold and release, tripple click

dnedeltchev, 2021-09-10:
## What happened
I had WXCJKG12LM and WXCJKG13LM configured with various routines for double, triple and hold/release actions. This is all in the kids room so I am not sure when it stopped working exactly. So it could have been a while. I just started to configure another WXCJKG12LM and noticed that the top 2 buttons only register a single click. I re-paired the device but that didn't help. Then I picked up the WXCJKG13LM knowing that this was configured with double, triple and hold/release routines but the same thing happened with it as well. So it looks like this stopped working some time back and re-pairing the devices does not fix the issue.

## What did you expect to happen
I had these working with single, double, triple and hold/release. Now it seems that single click is available on all buttons but double, triple and hold/release is available on the bottom buttons but not all where it was available on all buttons.

## How to reproduce it (minimal and precise)
1. Pair the device
2. change mode to command
3. test buttons

## Debug info
Zigbee2MQTT version:
Adapter hardware: CC26X2R1 (zzh from Electrolama)
Adapter firmware version: 20210120 (it's not the latest version, yet)

dnedeltchev, 2021-09-10:
Updated to the latest firmware, boy was that a hairy experience..., had to repair some lights and still getting weird behaviour, but... the issue remains. The 2LM for example has the following
All buttons have single click
1,2 have only single click. Neither double nor hold release is reported. If you double click you see single click in the logs. That used to work. Buttons 3,4 have single and double but not hold and release. This also used to work. I had node red flows based on those. The 3LM is similar.

snorro80, 2021-09-17:
Having the same experience with WXCJKG11LM. Read about a kernel issue somewhere in the issues-section here. Maybe it's related? Al happend after updating zigbee2mqtt and domoticz plugin.

dnedeltchev, 2021-09-17:
> Having the same experience with WXCJKG11LM. Read about a kernel issue somewhere in the issues-section here. Maybe it's related? Al happend after updating zigbee2mqtt and domoticz plugin.

Thank you for that. I thought I was going crazy :)
I've redone my node red for now with the available actions but having to retrain the kids and the Mrs is proving challenging :D

dnedeltchev, 2021-09-20:
Hm... I just upgraded to core-2021.9.7 and the hold release are back. But double and triple click on button 3 and 4 are still not working... I'm not sure if it was the core upgrade. I didn't see anything obvious in the release notes.

dnedeltchev, 2021-10-02:
upgraded to 1.21.2-1 and still the same issue

dnedeltchev, 2021-11-02:
I can confirm the issue is still present after updating to 1.22.0-2.

dnedeltchev, 2021-11-02:
Somehow I am unable to remove the stale label... I do not see an option but the issue is still not fixed. I am considering moving away from Opple. It's impossible to redo all automations without the double-click, hold and release, etc. :(

closed 2021-11-10 by github-actions[bot]

dnedeltchev, 2021-12-16:
> Having the same experience with WXCJKG11LM. Read about a kernel issue somewhere in the issues-section here. Maybe it's related? Al happend after updating zigbee2mqtt and domoticz plugin.

I found a workaround... not sure if it's by design but I got all the actions - single, double, triple click as well as hold and release on all buttons. What I did was the following.

1. Pair the button
2. Change the operation mode to command as usual as per https://www.zigbee2mqtt.io/devices/WXCJKG11LM.html
3. Once device is paired, you go to the z2m web UI in HA and under devices you find the device you paired. Then click on reconfigure
[image]

after a while it will throw an error about not being able to reconfigure but... in actual fact, all actions are back!

I tried this on 2 devices a 2 button and a 6 one and this worked like a charm!

I hope this helps. It's still annoying and it appears that this is not an issue for anyone else but us two... and no fix is coming but at least this gets us back in business! :)