Koenkk/zigbee2mqtt #8710: Please add support for Yale YDF40

juslex, 2021-09-10:
## Information about the device + link
..https://www.yalehome.com.br/pt-BR/produtos1/products/fechaduras-digitais-biometricas/fechadura-digital-ydf40/

Yale Digital Lock YDF40

## data/database.db entry of the device

{"id":2,"type":"EndDevice","ieeeAddr":"0x000d6f00168a41f0","nwkAddr":16619,"manufId":4491,"manufName":"ASSA ABLOY iRevo","powerSource":"Battery","modelId":"c700000202","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":10,"inClusterList":[0,1,3,4,5,9,10,32,257],"outClusterList":[10,25],"clusters":{"genBasic":{"attributes":{"modelId":"c700000202","manufacturerName":"ASSA ABLOY iRevo","powerSource":3,"zclVersion":1,"appVersion":11,"hwVersion":0}},"genPowerCfg":{"attributes":{"batteryPercentageRemaining":90}}},"binds":[{"cluster":257,"type":"endpoint","deviceIeeeAddress":"0x00124b0022989931","endpointID":1},{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x00124b0022989931","endpointID":1}],"configuredReportings":[{"cluster":257,"attrId":0,"minRepIntval":0,"maxRepIntval":3600,"repChange":0},{"cluster":1,"attrId":33,"minRepIntval":3600,"maxRepIntval":62000,"repChange":0}],"meta":{}}},"appVersion":11,"hwVersion":0,"zclVersion":1,"interviewCompleted":true,"meta":{"configured":2},"lastSeen":1631283208589}
..

I added the following entry to yale.js using the same code as YMF40/YDM4109+ but I don't have information about the lock battery. Also, there is a delay when sending commands to it via z2m / ha. Finally, it occasionally stops working using this code. I ask for help for the correct support. PS. I haven't tested the pind code management yet.

{
        zigbeeModel: ['c700000202'],
        model: 'YDF40',
        vendor: 'Yale',
        description: 'Real living lock / Intelligent biometric digital lock',
        extend: lockExtend,
        // Increased timeout needed: https://github.com/Koenkk/zigbee2mqtt/issues/3290 for YDM4109+
        meta: {timeout: 20000},
    },

juslex, 2021-09-10:
[image]

juslex, 2021-09-10:
[image]

juslex, 2021-09-10:
[image]
[image]

juslex, 2021-09-16:
@Koenkk could you kindly help us? Thank you so much

juslex, 2022-07-16:
@Koenkk 

after some zigbee2mqtt updates the YDF40 is not working properly.  until recently, I couldn't finish the interview.  in the last version, I managed to finish the interview, but the lock keeps failing the commands, it doesn't update the status correctly and the information of the person who opened the lock is also not reported.

Koenkk, 2022-07-17:
Could you provide the herdsman debug log when the device is interviewed?

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable the herdsman debug logging. Note that this is only logged to STDOUT and not to log files.

andrepia, 2022-07-18:
@Koenkk Something similar is happening to my YDF40!
I can finish the device interview and everything works but it's a lot laggy and the state update is not happening automatically, I have to manually refresh it from the "exposes" tab do get an update. Besides, I have no battery status or action, action_source_name and action_source_user information.
After some online research, I could manage to workaround some of the issues by setting lockState and batteryPercentageRemaining from the Reporting tab as in this screenshot: 

[image]

Now the lag is gone and I have state automatically updating and battery status, but still no action, action_source_name and action_source_user information.
Maybe these screenshots can illustrate it better:

[image]

[image]

I tried to get the herdsman debug log (don't know if I managed to get the right output - if not, could you please elaborate on how to proceed?) when trying to reconfigure the device and got this error:

> Zigbee2MQTT:info  2022-07-17 23:05:44: Configuring 'Fechadura Sala'
> Zigbee2MQTT:error 2022-07-17 23:05:55: Failed to configure 'Fechadura Sala', attempt 1 (Error: Bind 0x000d6f0016fac4ad/1 genPowerCfg from '0x00124b00258b7ed4/1' failed (AREQ - ZDO - bindRsp after 10000ms)
>     at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35)
>     at listOnTimeout (node:internal/timers:559:17)
>     at processTimers (node:internal/timers:502:7))
> Zigbee2MQTT:info  2022-07-17 23:05:55: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/configure', payload '{"data":{"id":"Fechadura Sala"},"error":"Failed to configure (Bind 0x000d6f0016fac4ad/1 genPowerCfg from '0x00124b00258b7ed4/1' failed (AREQ - ZDO - bindRsp after 10000ms))","status":"error","transaction":"i93vq-1"}'

I hope it helps you to help us!!
If you need any further information, let me know!
Thanks in advance!!

andrepia, 2022-07-18:
@Koenkk Sure!

> {"id":6,"type":"EndDevice","ieeeAddr":"0x000d6f0016fac4ad","nwkAddr":752,"manufId":4491,"manufName":"ASSA ABLOY iRevo","powerSource":"Battery","modelId":"c700000202","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":10,"inClusterList":[0,1,3,4,5,9,10,32,257],"outClusterList":[10,25],"clusters":{"genBasic":{"attributes":{"hwVersion":0}},"closuresDoorLock":{"attributes":{"lockState":1}},"genPowerCfg":{"attributes":{"batteryPercentageRemaining":90}}},"binds":[{"cluster":257,"type":"endpoint","deviceIeeeAddress":"0x00124b00258b7ed4","endpointID":1},{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x00124b00258b7ed4","endpointID":1}],"configuredReportings":[{"cluster":1,"attrId":33,"minRepIntval":"3600","maxRepIntval":"43200","repChange":0},{"cluster":257,"attrId":0,"minRepIntval":"0","maxRepIntval":"3600","repChange":"0"}],"meta":{}}},"appVersion":11,"hwVersion":0,"zclVersion":1,"interviewCompleted":true,"meta":{},"lastSeen":1658157327317,"defaultSendRequestWhen":"fastpoll"}

I hope this is what you've asked for!

Koenkk, 2022-07-18:
@andrepia can you:

- stop z2m
- in the database.db change `,"defaultSendRequestWhen":"fastpoll"` to `,"defaultSendRequestWhen":"immediate"`
- start z2m
- see if the issue is fixed now.

juslex, 2022-07-18:
@Koenkk 

My database:

{"id":14,"type":"EndDevice","ieeeAddr":"0x000d6f00168a4334","nwkAddr":5768,"manufId":4491,"manufName":"ASSA ABLOY iRevo","powerSource":"Battery","modelId":"c700000202","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":10,"inClusterList":[0,1,3,4,5,9,10,32,257],"outClusterList":[10,25],"clusters":{"genBasic":{"attributes":{"hwVersion":0}},"genPollCtrl":{"attributes":{"checkinInterval":360}},"closuresDoorLock":{"attributes":{"lockState":1}}},"binds":[{"cluster":32,"type":"endpoint","deviceIeeeAddress":"0x00124b0024c0f070","endpointID":1}],"configuredReportings":[],"meta":{}}},"appVersion":11,"hwVersion":0,"zclVersion":1,"interviewCompleted":true,"meta":{},"lastSeen":1658161075471,"defaultSendRequestWhen":"fastpoll"}

andrepia, 2022-07-18:
@Koenkk
I changed defaultSendRequestWhen as you said and got the same result...
I even tried to remove the device and make it join the network again, and then changed the defaultSendRequestWhen once more, but nothing changed...

Koenkk, 2022-07-19:
@andrepia can you provide the debug log when you unlock the lock (this is when I expect an `action` to be send).

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable debug logging.

andrepia, 2022-07-19:
@Koenkk ok... This is what I got:

```
Debug 2022-07-19 12:51:32 Received MQTT message on 'zigbee2mqtt/sala_fechadura/set' with data '{"state":"UNLOCK"}'
Debug 2022-07-19 12:51:32 Publishing 'set' 'state' to 'sala_fechadura'
Debug 2022-07-19 12:51:38 Received Zigbee message from 'sala_fechadura', type 'commandOperationEventNotification', cluster 'closuresDoorLock', data '{"data":0,"opereventcode":2,"opereventsrc":1,"pin":{"data":[],"type":"Buffer"},"userid":0,"zigbeelocaltime":1770445695}' from endpoint 1 with groupID 0
Debug 2022-07-19 12:51:38 Received Zigbee message from 'sala_fechadura', type 'attributeReport', cluster 'closuresDoorLock', data '{"lockState":2}' from endpoint 1 with groupID 0
Info 2022-07-19 12:51:38 MQTT publish: topic 'zigbee2mqtt/sala_fechadura', payload '{"action":"unlock","action_source":1,"action_source_name":"rf","action_source_user":null,"action_user":0,"battery":45,"linkquality":51,"lock_state":"locked","state":"LOCK"}'
Info 2022-07-19 12:51:38 MQTT publish: topic 'zigbee2mqtt/sala_fechadura', payload '{"action":null,"action_source_name":null,"action_source_user":null,"battery":45,"linkquality":51,"lock_state":"unlocked","state":"UNLOCK"}'
Info 2022-07-19 12:51:38 MQTT publish: topic 'zigbee2mqtt/sala_fechadura', payload '{"action":"","action_source_name":null,"action_source_user":null,"battery":45,"linkquality":51,"lock_state":"unlocked","state":"UNLOCK"}'
Info 2022-07-19 12:51:38 MQTT publish: topic 'zigbee2mqtt/sala_fechadura/action', payload 'unlock'
Debug 2022-07-19 12:51:42 Received Zigbee message from 'sala_fechadura', type 'commandCheckIn', cluster 'genPollCtrl', data '{}' from endpoint 1 with groupID 0
```

Just to be clear, the lock was locked and I unlocked it via Home Assistant / Zigbee2Mqtt.

Koenkk, 2022-07-19:
It shows an `action` right?

```
Info 2022-07-19 12:51:38 MQTT publish: topic 'zigbee2mqtt/sala_fechadura', payload '{"action":"unlock","action_source":1,"action_source_name":"rf","action_source_user":null,"action_user":0,"battery":45,"linkquality":51,"lock_state":"locked","state":"LOCK"}'
```

Note that when HA integration is enabled, the action will be cleared as explained in the FAQ: https://www.zigbee2mqtt.io/guide/faq/#why-is-the-action-property-always-empty

andrepia, 2022-07-19:
Yes! You're right...
Looking in HA logbook I can see that the entity is properly updating and cleared right after:

[image]

Great!! I haven't checked that before!! Thanks a lot!!
So, if this is the expected behavior, than it's looking good after manually adding the reporting attributes...

One last thing: action and action_source_name are properly updating it's HA entities, but action_source_user is not... Looking at the debug, it looks like the user information is registered thru "action_user" and not thru "action_source_user" (which is not being exposed to home assistant)...
Is there a way to add the user information to the home assistant entity?

Koenkk, 2022-07-19:
> After some online research, I could manage to workaround some of the issues by setting lockState and batteryPercentageRemaining from the Reporting tab as in this screenshot:

What exactly did you do? I checked the code and these reporting are already being setup.

> One last thing: action and action_source_name are properly updating it's HA entities, but action_source_user is not... 

Fixed!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

juslex, 2022-07-19:
I'm waiting for the commit koenkk made to go live to test on the dev branch. As soon as I test it, I'll send the feedback here.

juslex, 2022-07-20:
after that i deleted and re-added the lock.  after the interview, the situation remained the same, with no response to commands, status without updating and no battery percentage.  so I decided to do what Andrepia did as picture[image] below.  now the lock works perfectly regarding the commands.  both directly on the front end of the z2m and on the home assistant, with speed of response.  but I don't have a battery percentage and the user's answer that activated it is also not complete.

[image]

juslex, 2022-07-20:
After re-added the lock 

My database is now with the option "defaultSendRequestWhen":"immediate"}

{"id":14,"type":"EndDevice","ieeeAddr":"0x000d6f00168a4334","nwkAddr":8588,"manufId":4491,"manufName":"ASSA ABLOY iRevo","powerSource":"Battery","modelId":"c700000202","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":10,"inClusterList":[0,1,3,4,5,9,10,32,257],"outClusterList":[10,25],"clusters":{"genBasic":{"attributes":{"modelId":"c700000202","manufacturerName":"ASSA ABLOY iRevo","powerSource":3,"zclVersion":1,"appVersion":11,"hwVersion":0}},"closuresDoorLock":{"attributes":{"lockState":1}}},"binds":[{"cluster":257,"type":"endpoint","deviceIeeeAddress":"0x00124b0024c0f070","endpointID":1},{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x00124b0024c0f070","endpointID":1}],"configuredReportings":[{"cluster":257,"attrId":0,"minRepIntval":"0","maxRepIntval":"900","repChange":0},{"cluster":1,"attrId":33,"minRepIntval":"3600","maxRepIntval":"43200","repChange":0}],"meta":{}}},"appVersion":11,"hwVersion":0,"zclVersion":1,"interviewCompleted":true,"meta":{},"lastSeen":1658280660892,"defaultSendRequestWhen":"immediate"}

Koenkk, 2022-07-20:
Instead of manually setting up the reporting does this work?
1. Force remove device from z2m
2. restart z2m
3. pair lock
4. now configure should fail and lock is slow
5. configure device by pressing the yellow refresh button on the device page -> does this succeed?
6. lock should be fast now.

andrepia, 2022-07-21:
Sorry I couldn’t check it today…
I’ll try to do it tomorrow and report back!

juslex, 2022-07-21:
> Instead of manually setting up the reporting does this work?
> 
> 1. Force remove device from z2m
> 2. restart z2m
> 3. pair lock
> 4. now configure should fail and lock is slow
> 5. configure device by pressing the yellow refresh button on the device page -> does this succeed?
> 6. lock should be fast now.

@Koenkk 
I followed the instructions carefully. After setting up the device, I got the error message attached. And the lock remains with the same problems reported by me and @andrepia. The unlock or lock state is not updated in either HA or Z2m, making it necessary to press the update state button on the frontend.

[image]

[image]

juslex, 2022-07-21:
After again doing the @andrepia procedure, the lock was quick again. However, in my case, the percentage of the battery still does not appear, only appearing some time later.
[image]

andrepia, 2022-07-21:
Same here....
This is the debug log output:

**Pairing:**
```
Debug 2022-07-21 13:13:09 Received MQTT message on 'zigbee2mqtt/bridge/request/permit_join' with data '{"device":null,"time":254,"transaction":"3gqru-1","value":true}'
Info 2022-07-21 13:13:09 Zigbee: allowing new devices to join.
Info 2022-07-21 13:13:09 MQTT publish: topic 'zigbee2mqtt/bridge/response/permit_join', payload '{"data":{"time":254,"value":true},"status":"ok","transaction":"3gqru-1"}'
Info 2022-07-21 13:13:31 Device '0x000d6f0016fac4ad' joined
Info 2022-07-21 13:13:31 MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x000d6f0016fac4ad","ieee_address":"0x000d6f0016fac4ad"},"type":"device_joined"}'
Info 2022-07-21 13:13:31 MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0x000d6f0016fac4ad"},"type":"device_connected"}'
Info 2022-07-21 13:13:31 Starting interview of '0x000d6f0016fac4ad'
Info 2022-07-21 13:13:31 MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x000d6f0016fac4ad","ieee_address":"0x000d6f0016fac4ad","status":"started"},"type":"device_interview"}'
Info 2022-07-21 13:13:31 MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0x000d6f0016fac4ad"},"type":"pairing"}'
Debug 2022-07-21 13:13:31 Device '0x000d6f0016fac4ad' announced itself
Info 2022-07-21 13:13:31 MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x000d6f0016fac4ad","ieee_address":"0x000d6f0016fac4ad"},"type":"device_announce"}'
Info 2022-07-21 13:13:31 MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"0x000d6f0016fac4ad"},"type":"device_announced"}'
Debug 2022-07-21 13:13:51 Received Zigbee message from '0x000d6f0016fac4ad', type 'readResponse', cluster 'genBasic', data '{"modelId":"c700000202"}' from endpoint 1 with groupID 0
Debug 2022-07-21 13:13:51 Skipping message, definition is undefined and still interviewing
Debug 2022-07-21 13:14:01 Received Zigbee message from '0x000d6f0016fac4ad', type 'readResponse', cluster 'genBasic', data '{"manufacturerName":"ASSA ABLOY iRevo"}' from endpoint 1 with groupID 0
Debug 2022-07-21 13:14:01 Skipping message, definition is undefined and still interviewing
Debug 2022-07-21 13:14:11 Received Zigbee message from '0x000d6f0016fac4ad', type 'readResponse', cluster 'genBasic', data '{"powerSource":3}' from endpoint 1 with groupID 0
Debug 2022-07-21 13:14:11 Skipping message, definition is undefined and still interviewing
Debug 2022-07-21 13:14:21 Received Zigbee message from '0x000d6f0016fac4ad', type 'readResponse', cluster 'genBasic', data '{"zclVersion":1}' from endpoint 1 with groupID 0
Debug 2022-07-21 13:14:21 Skipping message, definition is undefined and still interviewing
Debug 2022-07-21 13:14:31 Received Zigbee message from '0x000d6f0016fac4ad', type 'readResponse', cluster 'genBasic', data '{"appVersion":11}' from endpoint 1 with groupID 0
Debug 2022-07-21 13:14:31 Skipping message, definition is undefined and still interviewing
Debug 2022-07-21 13:14:41 Received Zigbee message from '0x000d6f0016fac4ad', type 'readResponse', cluster 'genBasic', data '{}' from endpoint 1 with groupID 0
Debug 2022-07-21 13:14:41 Skipping message, definition is undefined and still interviewing
Debug 2022-07-21 13:14:51 Received Zigbee message from '0x000d6f0016fac4ad', type 'readResponse', cluster 'genBasic', data '{"hwVersion":0}' from endpoint 1 with groupID 0
Debug 2022-07-21 13:14:51 Skipping message, definition is undefined and still interviewing
Debug 2022-07-21 13:15:01 Received Zigbee message from '0x000d6f0016fac4ad', type 'readResponse', cluster 'genBasic', data '{}' from endpoint 1 with groupID 0
Debug 2022-07-21 13:15:01 Skipping message, definition is undefined and still interviewing
Debug 2022-07-21 13:15:11 Received Zigbee message from '0x000d6f0016fac4ad', type 'readResponse', cluster 'genBasic', data '{}' from endpoint 1 with groupID 0
Debug 2022-07-21 13:15:11 Skipping message, definition is undefined and still interviewing
Info 2022-07-21 13:15:21 Successfully interviewed '0x000d6f0016fac4ad', device has successfully been paired
Info 2022-07-21 13:15:21 Device '0x000d6f0016fac4ad' is supported, identified as: Yale Real living lock / Intelligent biometric digital lock (YDF40)
Info 2022-07-21 13:15:21 MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"Real living lock / Intelligent biometric digital lock","exposes":[{"features":[{"access":7,"description":"State of the lock","name":"state","property":"state","type":"binary","value_off":"UNLOCK","value_on":"LOCK"},{"access":1,"description":"Actual state of the lock","name":"lock_state","property":"lock_state","type":"enum","values":["not_fully_locked","locked","unlocked"]}],"type":"lock"},{"access":1,"description":"Remaining battery in %","name":"battery","property":"battery","type":"numeric","unit":"%","value_max":100,"value_min":0},{"features":[{"access":2,"description":"User ID to set or clear the pincode for","name":"user","property":"user","type":"numeric"},{"access":2,"description":"Type of user, unrestricted: owner (default), (year|week)_day_schedule: user has ability to open lock based on specific time period, master: user has ability to both program and operate the door lock, non_access: user is recognized by the lock but does not have the ability to open the lock", [... cut ...] e_change":0,"transaction":"zl86j-4"}'
Info 2022-07-21 13:23:55 Configured reporting for '0x000d6f0016fac4ad/1', 'genPowerCfg.batteryPercentageRemaining'
Info 2022-07-21 13:23:55 MQTT publish: topic 'zigbee2mqtt/bridge/response/device/configure_reporting', payload '{"data":{"attribute":"batteryPercentageRemaining","cluster":"genPowerCfg","id":"0x000d6f0016fac4ad/1","maximum_report_interval":"43200","minimum_report_interval":"3600","reportable_change":0},"status":"ok","transaction":"zl86j-4"}'
```

Koenkk, 2022-07-21:
Could you check if the issue is fixed with the following external converter:

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['c700000202', '06ffff2029'],
    model: 'YDF40',
    vendor: 'Yale',
    description: 'Real living lock / Intelligent biometric digital lock',
    fromZigbee: [fz.lock, fz.battery, fz.lock_operation_event, fz.lock_programming_event, fz.lock_pin_code_response,
        fz.lock_user_status_response],
    toZigbee: [tz.lock, tz.pincode_lock, tz.lock_userstatus],
    meta: {pinCodeCount: 250, ...meta},
    exposes: [e.lock(), e.battery(), e.pincode(), e.lock_action(), e.lock_action_source_name(), e.lock_action_user()],
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['closuresDoorLock']);
        await reporting.lockState(endpoint);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genPowerCfg']);
        await reporting.batteryPercentageRemaining(endpoint);
    },
};

module.exports = definition;

```

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, check if issue is fixed

andrepia, 2022-07-21:
I created the ext_converter.js with code you provided and added it to the configuration.yaml, but now I cannot start the Zigbee2MQTT add-on... Here's the log containing the error (which loops every few seconds):

```
ReferenceError: meta is not defined
    at /app/dist/util/externally-loaded.js:19:34
    at Script.runInContext (node:vm:139:12)
    at Script.runInNewContext (node:vm:144:17)
    at Object.runInNewContext (node:vm:298:38)
    at loadModuleFromText (/app/lib/util/utils.ts:148:8)
    at loadModuleFromFile (/app/lib/util/utils.ts:155:12)
    at Object.getExternalConvertersDefinitions (/app/lib/util/utils.ts:165:25)
    at getExternalConvertersDefinitions.next (<anonymous>)
    at new ExternalConverters (/app/lib/extension/externalConverters.ts:12:20)
    at new Controller (/app/lib/controller.ts:84:58)
[18:03:36] INFO: Handing over control to Zigbee2MQTT Core ...
> zigbee2mqtt@1.26.0 start
> node index.js
```

Maybe I need some extra reference or definition for "meta"??

Koenkk, 2022-07-23:
@andrepia I've updated https://github.com/Koenkk/zigbee2mqtt/issues/8710#issuecomment-1191917686, please try again.

andrepia, 2022-07-24:
@Koenkk Thanks! I have some progress to share now...
I updated the code but couldn't start zigbee2mqtt due to "e.lock_action_user is not a function" error (as the log shows below). I think this error refers to the action_user issue already treated on https://github.com/Koenkk/zigbee2mqtt/issues/8710#issuecomment-1189388575.

```
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
cont-init: info: running /etc/cont-init.d/socat.sh
[20:29:02] INFO: Socat not enabled, marking service as down
cont-init: info: /etc/cont-init.d/socat.sh exited 0
cont-init: info: running /etc/cont-init.d/zigbee2mqtt.sh
cont-init: info: /etc/cont-init.d/zigbee2mqtt.sh exited 0
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
services-up: info: copying legacy longrun socat (no readiness notification)
services-up: info: copying legacy longrun zigbee2mqtt (no readiness notification)
[20:29:03] INFO: Handing over control to Zigbee2MQTT Core ...
> zigbee2mqtt@1.26.0 start
> node index.js
/app/dist/util/externally-loaded.js:20
    exposes: [e.lock(), e.battery(), e.pincode(), e.lock_action(), e.lock_action_source_name(), e.lock_action_user()],
                                                                                                  ^
TypeError: e.lock_action_user is not a function
    at /app/dist/util/externally-loaded.js:20:99
    at Script.runInContext (node:vm:139:12)
    at Script.runInNewContext (node:vm:144:17)
    at Object.runInNewContext (node:vm:298:38)
    at loadModuleFromText (/app/lib/util/utils.ts:148:8)
    at loadModuleFromFile (/app/lib/util/utils.ts:155:12)
    at Object.getExternalConvertersDefinitions (/app/lib/util/utils.ts:165:25)
    at getExternalConvertersDefinitions.next (<anonymous>)
    at new ExternalConverters (/app/lib/extension/externalConverters.ts:12:20)
    at new Controller (/app/lib/controller.ts:84:58)
[20:29:06] INFO: Handing over control to Zigbee2MQTT Core ...
> zigbee2mqtt@1.26.0 start
> node index.js
```

So I tried to start zigbee2mqtt edge and all went well!
After that I removed the lock, restarted zigbee2mqtt edge and readded the device. The interview was completed but the configure failed. See the log:

```
Info 2022-07-23 20:40:10Device 'sala_fechadura' joined
Info 2022-07-23 20:40:11MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"sala_fechadura","ieee_address":"0x000d6f0016fac4ad"},"type":"device_joined"}'
Info 2022-07-23 20:40:11MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"sala_fechadura"},"type":"device_connected"}'
Info 2022-07-23 20:40:11Starting interview of 'sala_fechadura'
Info 2022-07-23 20:40:11MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"sala_fechadura","ieee_address":"0x000d6f0016fac4ad","status":"started"},"type":"device_interview"}'
Info 2022-07-23 20:40:11MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"sala_fechadura"},"type":"pairing"}'
Debug 2022-07-23 20:40:11Device 'sala_fechadura' announced itself
Info 2022-07-23 20:40:11MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"sala_fechadura","ieee_address":"0x000d6f0016fac4ad"},"type":"device_announce"}'
Info 2022-07-23 20:40:11MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"sala_fechadura"},"type":"device_announced"}'
Debug 2022-07-23 20:40:31Received Zigbee message from 'sala_fechadura', type 'readResponse', cluster 'genBasic', data '{"modelId":"c700000202"}' from endpoint 1 with groupID 0
Debug 2022-07-23 20:40:41Received Zigbee message from 'sala_fechadura', type 'readResponse', cluster 'genBasic', data '{"manufacturerName":"ASSA ABLOY iRevo"}' from endpoint 1 with groupID 0
Debug 2022-07-23 20:40:51Received Zigbee message from 'sala_fechadura', type 'readResponse', cluster 'genBasic', data '{"powerSource":3}' from endpoint 1 with groupID 0
Debug 2022-07-23 20:41:01Received Zigbee message from 'sala_fechadura', type 'readResponse', cluster 'genBasic', data '{"zclVersion":1}' from endpoint 1 with groupID 0
Debug 2022-07-23 20:41:11Received Zigbee message from 'sala_fechadura', type 'readResponse', cluster 'genBasic', data '{"appVersion":11}' from endpoint 1 with groupID 0
Debug 2022-07-23 20:41:21Received Zigbee message from 'sala_fechadura', type 'readResponse', cluster 'genBasic', data '{}' from endpoint 1 with groupID 0
Debug 2022-07-23 20:41:31Received Zigbee message from 'sala_fechadura', type 'readResponse', cluster 'genBasic', data '{"hwVersion":0}' from endpoint 1 with groupID 0
Debug 2022-07-23 20:41:41Received Zigbee message from 'sala_fechadura', type 'readResponse', cluster 'genBasic', data '{}' from endpoint 1 with groupID 0
Debug 2022-07-23 20:41:51Received Zigbee message from 'sala_fechadura', type 'readResponse', cluster 'genBasic', data '{}' from endpoint 1 with groupID 0
Info 2022-07-23 20:42:01Successfully interviewed 'sala_fechadura', device has successfully been paired
Info 2022-07-23 20:42:01Device 'sala_fechadura' is supported, identified as: Yale Real living lock / Intelligent biometric digital lock (YDF40)
[... cut ...] al/timers:502:7))

```

Despite that, the lock is working fast and it was properly updating it's status... Reporting tab shows that "lockState" is configured and I got the battery info on About tab...
I also noticed on the log that every once in a while it automatically tries to configure the device, but fails everytime...
So, the bottom line is: despite the "failed to configure" message on the log, the lock is working properly without having to manually set the parameters on the reporting tab!!

juslex, 2022-07-24:
In my case, as I mentioned in https://github.com/Koenkk/zigbee2mqtt/issues/8710#issuecomment-1191660985 soon after deleting and adding the lock again, the lock only received command again and had a status update after this procedure https://github.com/Koenkk/zig Yesterday I changed the batteries for new ones and it still keeps showing only 45%.

PS. Still don't test the external converter considering the errors that @andrepia is reporting.
[image]

Koenkk, 2022-07-24:
@andrepia can you check if the configure still fails with:

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['c700000202', '06ffff2029'],
    model: 'YDF40',
    vendor: 'Yale',
    description: 'Real living lock / Intelligent biometric digital lock',
    fromZigbee: [fz.lock, fz.battery, fz.lock_operation_event, fz.lock_programming_event, fz.lock_pin_code_response,
        fz.lock_user_status_response],
    toZigbee: [tz.lock, tz.pincode_lock, tz.lock_userstatus],
    meta: {pinCodeCount: 250},
    exposes: [e.lock(), e.battery(), e.pincode(), e.lock_action(), e.lock_action_source_name()],
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.lockState(endpoint);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genPowerCfg']);
        await reporting.batteryPercentageRemaining(endpoint);
    },
};

module.exports = definition;

```

juslex, 2022-07-25:
I include the external converter, deleted the lock, restart zigbee2mqtt and re-add the lock. After pairing, I tried to make a configuration, but it had an error. The lock began to present the same error again, it gave the command to the home assistant to open, it opened, but the status did not update that it was open neither in HA nor in z2m. So I changed from 3600 to 900 in the max rep interval in reporting using the @andrepia configuration. After that, the lock returned to working well and managed to configure without error. However, I still have no battery status.

Here is the log:

Debug 2022-07-25 09:35:15Received MQTT message on 'zigbee2mqtt/bridge/request/device/configure' with data '{"id":"Fechadura Yale","transaction":"bjixc-1"}'
Info 2022-07-25 09:35:15MQTT publish: topic 'zigbee2mqtt/bridge/response/device/configure', payload '{"data":{"id":"Fechadura Yale"},"status":"ok","transaction":"bjixc-1"}'

[image]

[image]

juslex, 2022-07-26:
Night update: after reinstall the docker image, the lock starting to configure without me having requested it and I received this error message:

022-07-25 23:00:26Failed to configure 'Fechadura Yale', attempt 3 (Error: ConfigureReporting 0x000d6f00168a4334/1 closuresDoorLock([{"attribute":"lockState","minimumReportInterval":0,"maximumReportInterval":3600,"reportableChange":0}], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'MAC no ack' (233)) at ZStackAdapter.sendZclFrameToEndpointInternal (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:414:23) at runMicrotasks (<anonymous>) at processTicksAndRejections (node:internal/process/task_queues:96:5) at Queue.executeNext (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:32:32))

juslex, 2022-07-26:
One more update:

After the previous error, I removed the external converter, updated the dev version to 1.26.0-dev commit: 343ef7f, deleted the lock, restart z2m and added the lock again. This time, the lock was recognized quickly, it appeared immediately with battery reporting (still wrong at 45%) and after the interview I was able to configure it successfully without any error. A little slow commands, I just moved the max rep interval of endpoint 1 in the cloruesDoorLock cluster from 3600 to 900 and the speed returned. There is a perfect update of the action in home Assistant as well.

To summarize in the current stage of version 1.26.0-dev commit: 343ef7f: a) battery percentage appears at 45% even with new batteries; b) speed is better when set at 900 instead of 3600 max rep interval of endpoint 1 in the cloruesDoorLock cluster;

By solving only these two details, the lock will be 100% without any failure.

Koenkk, 2022-07-26:
@juslex can you check with this converter? Should fix both A and B.

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['c700000202', '06ffff2029'],
    model: 'YDF40',
    vendor: 'Yale',
    description: 'Real living lock / Intelligent biometric digital lock',
    fromZigbee: [fz.lock, fz.battery, fz.lock_operation_event, fz.lock_programming_event, fz.lock_pin_code_response,
        fz.lock_user_status_response],
    toZigbee: [tz.lock, tz.pincode_lock, tz.lock_userstatus],
    meta: {pinCodeCount: 250, dontDividePercentage: true},
    exposes: [e.lock(), e.battery(), e.pincode(), e.lock_action(), e.lock_action_source_name(), e.lock_action_user()],
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['closuresDoorLock', 'genPowerCfg']);
        await reporting.lockState(endpoint, {max: 900}));
        await reporting.batteryPercentageRemaining(endpoint);
    },
};

module.exports = definition;

``

juslex, 2022-08-02:
@Koenkk Sorry by delay

I created an external converter with the above code and z2m did not start, here is the log:

Using '/app/data' as data directory

/app/dist/util/externally-loaded.js:24

        await reporting.lockState(endpoint, {max: 900}));

                                                       ^

SyntaxError: Unexpected token ')'

    at new Script (node:vm:100:7)

    at createScript (node:vm:257:10)

    at Object.runInNewContext (node:vm:298:10)

    at loadModuleFromText (/app/lib/util/utils.ts:148:8)

    at loadModuleFromFile (/app/lib/util/utils.ts:155:12)

    at Object.getExternalConvertersDefinitions (/app/lib/util/utils.ts:165:25)

    at getExternalConvertersDefinitions.next (<anonymous>)

    at new ExternalConverters (/app/lib/extension/externalConverters.ts:12:20)

    at new Controller (/app/lib/controller.ts:84:58)

    at start (/app/index.js:108:18)

Using '/app/data' as data directory

/app/dist/util/externally-loaded.js:24

        await reporting.lockState(endpoint, {max: 900}));

                                                       ^

SyntaxError: Unexpected token ')'

    at new Script (node:vm:100:7)

    at createScript (node:vm:257:10)

    at Object.runInNewContext (node:vm:298:10)

    at loadModuleFromText (/app/lib/util/utils.ts:148:8)

    at loadModuleFromFile (/app/lib/util/utils.ts:155:12)

    at Object.getExternalConvertersDefinitions (/app/lib/util/utils.ts:165:25)

    at getExternalConvertersDefinitions.next (<anonymous>)

    at new ExternalConverters (/app/lib/extension/externalConverters.ts:12:20)

    at new Controller (/app/lib/controller.ts:84:58)

    at start (/app/index.js:108:18)

Using '/app/data' as data directory

/app/dist/util/externally-loaded.js:24

        await reporting.lockState(endpoint, {max: 900}));

                                                       ^

SyntaxError: Unexpected token ')'

    at new Script (node:vm:100:7)

    at createScript (node:vm:257:10)

    at Object.runInNewContext (node:vm:298:10)

    at loadModuleFromText (/app/lib/util/utils.ts:148:8)

    at loadModuleFromFile (/app/lib/util/utils.ts:155:12)

    at Object.getExternalConvertersDefinitions (/app/lib/util/utils.ts:165:25)

    at getExternalConvertersDefinitions.next (<anonymous>)

    at new ExternalConverters (/app/lib/extension/externalConverters.ts:12:20)

    at new Controller (/app/lib/controller.ts:84:58)

    at start (/app/index.js:108:18)

Using '/app/data' as data directory

/app/dist/util/externally-loaded.js:24

        await reporting.lockState(endpoint, {max: 900}));

                                                       ^

SyntaxError: Unexpected token ')'

    at new Script (node:vm:100:7)

    at createScript (node:vm:257:10)

    at Object.runInNewContext (node:vm:298:10)

    at loadModuleFromText (/app/lib/util/utils.ts:148:8)

    at loadModuleFromFile (/app/lib/util/utils.ts:155:12)

    at Object.getExternalConvertersDefinitions (/app/lib/util/utils.ts:165:25)

    at getExternalConvertersDefinitions.next (<anonymous>)

    at new ExternalConverters (/app/lib/extension/externalConverters.ts:12:20)

    at new Controller (/app/lib/controller.ts:84:58)

    at start (/app/index.js:108:18)

Using '/app/data' as data directory

/app/dist/util/externally-loaded.js:24

        await reporting.lockState(endpoint, {max: 900}));

juslex, 2022-08-02:
I'm using the latest dev version 1.27.0-dev commit: [4e590ec](https://github.com/Koenkk/zigbee2mqtt/commit/4e590ec)

Koenkk, 2022-08-02:
@juslex updated https://github.com/Koenkk/zigbee2mqtt/issues/8710#issuecomment-1195802534, it contained a small error.

juslex, 2022-08-03:
This time, I was successful in starting z2m with the external converter. However, even problems from before have returned. Status only updates if we press manually and failed to configure.[image][image]

Koenkk, 2022-08-04:
I update https://github.com/Koenkk/zigbee2mqtt/issues/8710#issuecomment-1195802534 , can you try again?

juslex, 2022-08-07:
@Koenkk The external converter allowed the completion of the configure successfully but I have no status update despite the commands being responding. There is also no update in the HA of the action. The battery is also not shown.

[image], [image][image][image][image]

Koenkk, 2022-08-07:
Can you check with https://github.com/Koenkk/zigbee2mqtt/issues/8710#issuecomment-1195802534, lock state should work, battery not yet.

juslex, 2022-08-08:
@Koenkk Good news, configure working, quick responses in the lock, with status and action update in Home Assistant! The battery, as already mentioned, remains unindicated.
[image]

Koenkk, 2022-08-09:
@juslex I updated https://github.com/Koenkk/zigbee2mqtt/issues/8710#issuecomment-1195802534 , does battery work now?

dsantos21, 2022-08-11:
Sorry to hijack this thread, but is this code the same for the yale 420d? The issue #8946 was closed (stale) and I think it was not committed, and has the same problem described here.

PS: Just bought one 420d :) zigbee2mqtt says its unsupported: (

juslex, 2022-08-12:
> @juslex I updated [#8710 (comment)](https://github.com/Koenkk/zigbee2mqtt/issues/8710#issuecomment-1195802534) , does battery work now?

@Koenkk I tried again with the new code, however, the battery still did not appear and I started to have errors again to configure the lock.

[image]

[image]

Koenkk, 2022-08-12:
Updated https://github.com/Koenkk/zigbee2mqtt/issues/8710#issuecomment-1195802534 again, please try again.

juslex, 2022-08-13:
@Koenkk from the previous to the current code, in addition to the battery not having worked, the lock commands were not working after some time. Earlier today before using the new code, the commands stopped responding and I got the following error:

[image]

After that, I put the last code, this time, the percentage of the battery appeared, but wrong, showing 125% instead of 100%. The configure was successful. configure. Soon after the pairing, some commands did not respond via home assistant and presented some delay, I will continue testing.

[image]
[image]

juslex, 2022-08-14:
@Koenkk update. The code is perfect. The only problem is the percentage of the battery that appears 125% instead of 100%.

Koenkk, 2022-08-14:
Updated https://github.com/Koenkk/zigbee2mqtt/issues/8710#issuecomment-1195802534 , may take +- 24 hour for the correct battery % to be reported. It should report 62.5% battery.

juslex, 2022-08-17:
Battery showing 45%[image]

Koenkk, 2022-08-17:
I guess battery percentage dropped further, does it show close to 100% when putting in a new battery?

juslex, 2022-08-19:
@andrepia Could you tested the last code posted by @Koenkk? How’s the battery there?

juslex, 2022-08-24:
@Koenkk I think that you can upload the last code to the dev branch, I'm out of new batteries for now, but the lock is working perfectly just reporting the probably wrong battery percentage. As soon as I change the batteries, I come here to confirm that the error remains.

Koenkk, 2022-08-24:
Great, integrated the fix.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

shing6326, 2022-08-26:
Hello, 
after upgraded to Zigbee2MQTT version 1.27.0-dev commit: [f1a1915](https://github.com/Koenkk/zigbee2mqtt/commit/f1a1915), I still got some display issues on the battery. The new battery shows around 33% from the UI. 

[image]

shing6326, 2022-08-26:
I have checked the logs and seems it reports reasonable reading from the zigbee message, but the UI shows about half of the reading

`01:40:51 user1@DESKTOP-PDMGKA8 tmp → k logs -f zigbee2mqtt-9cdcbd774-6fsdq | grep -F '大門門鎖' | grep batteryPercentageRemaining
Zigbee2MQTT:debug 2022-08-26 13:32:28: Received Zigbee message from '大門門鎖', type 'attributeReport', cluster 'genPowerCfg', data '{"batteryPercentageRemaining":60}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2022-08-26 14:32:29: Received Zigbee message from '大門門鎖', type 'attributeReport', cluster 'genPowerCfg', data '{"batteryPercentageRemaining":60}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2022-08-26 15:32:29: Received Zigbee message from '大門門鎖', type 'attributeReport', cluster 'genPowerCfg', data '{"batteryPercentageRemaining":56}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2022-08-26 16:32:30: Received Zigbee message from '大門門鎖', type 'attributeReport', cluster 'genPowerCfg', data '{"batteryPercentageRemaining":57}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2022-08-26 17:32:30: Received Zigbee message from '大門門鎖', type 'attributeReport', cluster 'genPowerCfg', data '{"batteryPercentageRemaining":60}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2022-08-26 18:32:31: Received Zigbee message from '大門門鎖', type 'attributeReport', cluster 'genPowerCfg', data '{"batteryPercentageRemaining":59}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2022-08-26 19:32:32: Received Zigbee message from '大門門鎖', type 'attributeReport', cluster 'genPowerCfg', data '{"batteryPercentageRemaining":59}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2022-08-26 20:32:32: Received Zigbee message from '大門門鎖', type 'attributeReport', cluster 'genPowerCfg', data '{"batteryPercentageRemaining":60}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2022-08-26 21:32:33: Received Zigbee message from '大門門鎖', type 'attributeReport', cluster 'genPowerCfg', data '{"batteryPercentageRemaining":59}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2022-08-26 22:32:34: Received Zigbee message from '大門門鎖', type 'attributeReport', cluster 'genPowerCfg', data '{"batteryPercentageRemaining":61}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2022-08-26 23:32:34: Received Zigbee message from '大門門鎖', type 'attributeReport', cluster 'genPowerCfg', data '{"batteryPercentageRemaining":59}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2022-08-27 00:32:35: Received Zigbee message from '大門門鎖', type 'attributeReport', cluster 'genPowerCfg', data '{"batteryPercentageRemaining":57}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2022-08-27 01:32:35: Received Zigbee message from '大門門鎖', type 'attributeReport', cluster 'genPowerCfg', data '{"batteryPercentageRemaining":59}' from endpoint 1 with groupID 0`

Koenkk, 2022-08-26:
@shing6326 that is expect, as per the ZCL spec battery % = batteryPercentageRemaining / 2. But the problem is that some firmwares do not conform to this spec. So for some battery % = batteryPercentageRemaining. Only way to find out what the YDF40 uses is to put in a new battery and see if it reports a batteryPercentageRemaining > 100

juslex, 2022-08-26:
@Koenkk I upgraded to the latest dev version, put new batteries, readitioned the lock and the battery percentage shows only 45%.

[image]

Koenkk, 2022-08-27:
Ok so the battery % should not be divided by 2 (it will become 90% in your case).

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

shing6326, 2022-08-27:
Hello @Koenkk ,
My YDM4109+ also have the same issue, could you also fix it as well? Thanks
` {
        zigbeeModel: ['iZBModule01', '0700000001'],
        model: 'YMF40/YDM4109+',
        vendor: 'Yale',
        description: 'Real living lock / Intelligent biometric digital lock',
        // Increased timeout needed: https://github.com/Koenkk/zigbee2mqtt/issues/3290 for YDM4109+
        extend: lockExtend({timeout: 20000}),
    },
`

[5 more replies]