Koenkk/zigbee2mqtt #8711: zigbee network freezes(?) around 1-2 times a week using CC2652RB adapter.

kpcz, 2021-09-10:
Hi all. I use zigbee2Mqtt with Slaesh's CC2652RB adapter. There are cases, at least once a week when zigbee network comes unavailable (no sensors or devices are responding and reporting in logs), Zigbee2mqtt addon restart helps every time.

I  have around 52 devices in network, from which 5 are powered devices - routers. 

Slaesh's CC2652RB adapter have the firmware version as it was shipped - 20200805
zigbee2mqtt is on 1.21.1-1 version.

In standard logs, it does not show anything when zigbee network hangs. For now turned on   log_level: debug, do I need to turn on more for debbugging the issue?

Could it be that the old FW of the adapter is causing some issues, and it needs to be reflashed to a newer FW? Or as the restart of the HA addon helps, it means the adapter is working fine, but there is something wrong with addon config? As I am avoiding reflashing the FW of the adapter, as I am afraid that I will need to re-join all the end devices, that would be real pain..

For now the config looks like this, maybe something wrong in there?

```
data_path: /share/zigbee2mqtt
external_converters: []
devices: devices.yaml
groups: groups.yaml
homeassistant: true
permit_join: false
mqtt:
  base_topic: zigbee2mqtt
  server: mqtt://core-mosquitto
  user: _deleted_
  password: _deleted_
serial:
  port: >-
    /dev/serial/by-id/usb-Silicon_Labs_slae.sh_cc2652rb_stick_-_slaesh_s_iot_stuff_00_12_4B_00_21_CB_F2_24-if00-port0
advanced:
  ikea_ota_use_test_url: true
  log_level: debug
  pan_id: 6755
  channel: 11
  network_key:
    - 1
    - 3
    - 5
    - 7
    - 9
    - 11
    - 13
    - 15
    - 0
    - 2
    - 4
    - 6
    - 8
    - 10
    - 12
    - 13
  availability_blocklist: []
  availability_passlist: []
device_options: {}
blocklist: []
passlist: []
queue: {}
frontend:
  port: 8099
experimental: {}
socat:
  enabled: false
  master: pty,raw,echo=0,link=/dev/ttyZ2M,mode=777
  slave: tcp-listen:8485,keepalive,nodelay,reuseaddr,keepidle=1,keepintvl=1,keepcnt=5
  options: '-d -d'
  log: false
zigbee_herdsman_debug: false

```

Looking forward for some ideas how I could debbug the issue, or at what direction to search, 
Side note, around 1m from zigbee dongle (what is connected via usb cord) there is also IKEA tradfri GW, with around 6 devices, if that could cause any issues.

thank you.

kpcz, 2021-09-11:
Hanged second Night un a row with error un logs:
`Zigbee2MQTT:debug 2021-09-11 08:26:13: Saving state to file /share/zigbee2mqtt/state.json
(node:329) UnhandledPromiseRejectionWarning: Error: SRSP - SYS - version after 6000ms
    at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35)
    at listOnTimeout (internal/timers.js:557:17)
    at processTimers (internal/timers.js:500:7)
(Use `node --trace-warnings ...` to show where the warning was created)
(node:329) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 1)
(node:329) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.`

lux73, 2021-09-12:
i would try without ser2net/socat - connect your CC2652RB direct to the z2m Machine. i could imagine there is the Problem.

btw. i have **switched all my SmartHome Software**: mosquitto, z2m, openhab, hostapd **to one physical dedicated APU2 Board.**

My System works much smoother now - before that all was installed in their own small VMs...

Koenkk, 2021-09-12:
I would also suggest to update the firmware, `20200805` could crash after some uptime, this is fixed in latest fw versions: https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_3.x.0/bin

kpcz, 2021-09-12:
Thanks for your answer. The CC2652RB is connected straight to the RPi4 USB. Now when looking at config I am not quite sure why the socat is in there at first place. But as it is set to   enabled: false  I guess it should not cause issues. Probably will delete it from config anyways.
@Koenkk Do I understand correctly, that updating the stick via procedure written in supported adapters, I will not need to repair my 50+ zigbee devices afterwards? Shut down HA, plug adapter to PC, flash, plug back to Rpi and start HA?

Koenkk, 2021-09-12:
> Shut down HA, plug adapter to PC, flash, plug back to Rpi and start HA?

Exactly, no repairing needed

kpcz, 2021-09-12:
Thanks :) Did the upgrade of the CC2652RB  successfully, and all devices connected also successfully. (had a little stress about it tho :) 
Will monitor with debugging logs enabled for about week, hopefully this will resolve the issue.

Thank You a lot for your great work.

lux73, 2021-09-12:
> 
> 
> ..The CC2652RB is connected straight to the RPi4 USB....
>

you should consider to attach your Stick via USB Extension Cable - directly connected to USB Port could trigger some heavy Delay and/or massive reduce the Range!

kpcz, 2021-10-13:
No issues ever since the upgrade of the coordinator FW. Thank you very much for your assistance.