Koenkk/zigbee2mqtt #8712: Endpoint not supported for cover type

tomidix, 2021-09-10:
## What happened
Trying to use a converter for 2ch zigbee module but can't start zigbee2mqtt with it defined in configuration.

## What did you expect to happen
Add the external converter into the config and use the device which supports to control two separate covers and therefore it has multiple zigbee endpoints

## How to reproduce it (minimal and precise)
Add the converter from https://github.com/Koenkk/zigbee2mqtt/issues/8705#issuecomment-916876546 to the configuration and restart the zigbee2mqtt

## Debug info
Zigbee2MQTT version: 1.21.1-1
Adapter hardware: CC2531
Adapter firmware version: -

## Error
If I've traced correctly, problem should be in this file: https://github.com/dimon78/zigbee2mqtt_standalone/blob/4fc16e3d577d9e98e02a3c8776b2fe9e30048a79/lib/extension/homeassistant.js
```
(node:1073) UnhandledPromiseRejectionWarning: AssertionError [ERR_ASSERTION]: Endpoint not supported for cover type
    at HomeAssistant.exposeToConfig (/app/lib/extension/homeassistant.js:283:13)
    at HomeAssistant.populateMapping (/app/lib/extension/homeassistant.js:591:54)
    at new HomeAssistant (/app/lib/extension/homeassistant.js:64:14)
    at new Controller (/app/lib/controller.js:83:34)
    at start (/app/index.js:94:18)
```

Koenkk, 2021-09-11:
Should be fixed now, can you check?

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

tomidix, 2021-09-11:
Thanks, I'll check it and report back as soon as I get back to place..

tomidix, 2021-09-11:
Seems fine now regarding this issue, tested with zb2mqtt edge.
Starts fine, pairing successful; device itself has some problems but this issue is solved. Thank you very much 😊

closed 2021-09-11 by tomidix