Koenkk/zigbee2mqtt #8729: After moving to 1.21.1 i have several (mostly IKEA Remote E1810) not reacting to all buttons anymore

njordan77, 2021-09-12:
Is this behaviour visible to others as well? I will try to switch back the version but right now part of my installation (mostly IKEA Remote) does react normal (fast) eg for the TOGGLE button, but others seem to be queued and the MQTT message comes about 1 minute or even more later......funny when pressing toggle afterwards the first not sent message is generated and sent together with the correct TOGGLE message.....

Tried to remove add these devices again but does not help at all....
btw. i'm on Raspi4 with a ZZH stick and the latest firmware from JULY. but did not change the firmware right now...so firmware should not be the problem....

Any help is highly appreciated. Norbert

Koenkk, 2021-09-12:
Might be due to https://github.com/Koenkk/zigbee2mqtt/issues/8663

njordan77, 2021-09-15:
Thanks, i searched in parallel and found several discussions based on this issue, did an rpi-update and after x.63 i now have all fine again. always annoying when these complex systems get stuck .... but happy that others already found the root cause....

closed 2021-09-15 by njordan77