Koenkk/zigbee2mqtt #8735: Error BUFFER not available in external converter (similar to another fixed issue)

systemiii, 2021-09-12:
# Bug Report
## What happened
I'm using Zigbee2MQTT Edge Addon for Home Assistant OS
I have written an external converter and i have the following error when i try to toggle the switch status

Publish 'set' 'state' to '0x00124b0022e14c01' failed: 'Error: Write 0x00124b0022e14c01/6 genPowerCfg({"1":{"value":{"0":1,"1":0,"2":0,"3":0,"4":0,"5":0,"6":0,"7":0},"type":1}}, {"sendWhenActive":false,"timeout":10000,"disableResponse":true,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":null,"reservedBits":3,"manufacturerCode":6866,"transactionSequenceNumber":233,"writeUndiv":false}) failed (Write for 'BUFFER' not available)'

This is the affected row
const payloadOn = {0x0001: {value: Uint8Array.from([1, 0, 0, 0, 0, 0, 0, 0]), type: 1}};

I don't have the error when i use a known cluster attribute, like this
const payloadOn = {mainsFrequency: {value: 1, type: 1}};

but it's not the same, obviously.

I think the following commit is related
https://github.com/Koenkk/zigbee-herdsman/commit/bc265563aa9fecc13f2356a8b5fa8fd65dcb7e9e

## What did you expect to happen
I expect i can use this payload in external converters

## How to reproduce it (minimal and precise)
Following part of code

const tzLocal = {

    livolo_socket_switch_on_off: {
        key: ['state'],
        convertSet: async (entity, key, value, meta) => {
            if (typeof value !== 'string') {
                return;
            }
            const state = value.toLowerCase();**
            const postfix = meta.endpoint_name || 'left';
            await entity.command('genOnOff', 'toggle', {}, {transactionSequenceNumber: 0});
            const payloadOn = {0x0001: {value: Uint8Array.from([1, 0, 0, 0, 0, 0, 0, 0]), type: 1}};
            const payloadOff = {0x0001: {value: Uint8Array.from([0, 0, 0, 0, 0, 0, 0, 0]), type: 1}};
            const payloadOnRight = {0x0001: {value: Uint8Array.from([2, 0, 0, 0, 0, 0, 0, 0]), type: 2}};
            const payloadOffRight = {0x0001: {value: Uint8Array.from([0, 0, 0, 0, 0, 0, 0, 0]), type: 2}};
            const payloadOnTop = {0x0001: {value: Uint8Array.from([4, 0, 0, 0, 0, 0, 0, 0]), type: 4}};
            const payloadOffTop = {0x0001: {value: Uint8Array.from([0, 0, 0, 0, 0, 0, 0, 0]), type: 4}};**
            if (postfix === 'left') {
                await entity.write('genPowerCfg', (state === 'on') ? payloadOn : payloadOff,
                    {
                        manufacturerCode: 0x1ad2, disableDefaultResponse: true, disableResponse: true,
                        reservedBits: 3, direction: 1, transactionSequenceNumber: 0xe9,
                    });
                return {state: {state_left: value.toUpperCase()}, readAfterWriteTime: 250};
            } else if (postfix === 'right') {
                await entity.write('genPowerCfg', (state === 'on') ? payloadOnRight : payloadOffRight,
                    {
                        manufacturerCode: 0x1ad2, disableDefaultResponse: true, disableResponse: true,
                        reservedBits: 3, direction: 1, transactionSequenceNumber: 0xe9,
                    });
                return {state: {state_right: value.toUpperCase()}, readAfterWriteTime: 250};
            } else if (postfix === 'top') {
                await entity.write('genPowerCfg', (state === 'on') ? payloadOnTop : payloadOffTop,
                    {
                        manufacturerCode: 0x1ad2, disableDefaultResponse: true, disableResponse: true,
                        reservedBits: 3, direction: 1, transactionSequenceNumber: 0xe9,
                    });
                return {state: {state_top: value.toUpperCase()}, readAfterWriteTime: 250};
            }
            return {state: {state: value.toUpperCase()}, readAfterWriteTime: 250};
        },
        convertGet: async (entity, key, meta) => {
            await entity.command('genOnOff', 'toggle', {}, {transactionSequenceNumber: 0});
        },
    },
};

## Debug Info
Zigbee2MQTT version: v1.21.1-1
Adapter hardware: CC2531
Adapter firmware version: 2020_11_02

Koenkk, 2021-09-12:
Instead of `Uint8Array.from([1, 0, 0, 0, 0, 0, 0, 0]`,can you try `Buffer.from([1, 0, 0, 0, 0, 0, 0, 0]`

systemiii, 2021-09-13:
I have this errore when i use "Buffer"

Publish 'set' 'state' to '0x00124b0022e14c01' failed: 'ReferenceError: Buffer is not defined'

systemiii, 2021-09-13:
Solved using

const Buffer = require('buffer').Buffer;

closed 2021-09-13 by systemiii