Koenkk/zigbee2mqtt #8738: Lock device only talks if routers present on network

davidmonro, 2021-09-13:
## What happened
Lockwood (Yale) lock will only connect via routers, falls off network rather than talk to coordinator

## What did you expect to happen
I would expect it to talk to the coordinator directly if there are no routers on the network

## How to reproduce it (minimal and precise)
* Join Lockwood/Yale YDD-D4F0-TSDB (rebadged Assure SL) to network.
* Remove power from all the router objects (IKEA lights in my case)
* Note that lock cannot be controlled and shows isolated on the zigbee map. 
* Note that attempts to control the lock result in messages like this:
```
Zigbee2MQTT:error 2021-09-13 12:31:17: Publish 'set' 'state' to '0x000ZZZZZZZZZZZZZ' failed: 'Error: Command 0x000ZZZZZZZZZZZZZ/1 closuresDoorLock.lockDoor({"pincodevalue":""}, {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'MAC no ack' (233))'
Zigbee2MQTT:info 2021-09-13 12:31:17: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'state' to '0x000ZZZZZZZZZZZZZ' failed: 'Error: Command 0x000ZZZZZZZZZZZZZ/1 closuresDoorLock.lockDoor({\"pincodevalue\":\"\"}, {\"sendWhenActive\":false,\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":true,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (Data request failed with error: 'MAC no ack' (233))'","meta":{"friendly_name":"0x000ZZZZZZZZZZZZZ"},"type":"zigbee_publish_error"}'
```
* Turn on a router, and the lock is now controllable and shows connected via the router on the map

## Debug info
Zigbee2MQTT version: 1.21.1
Adapter hardware: CC2538+CC2592
Adapter firmware version: 20201010 (jethome-ru latest)

The zigbee-herdsman-converters yale.js has been patched to support the lock (see https://github.com/Koenkk/zigbee-herdsman-converters/pull/3009)
I have a colleague seeing what looks like the same thing with the same lock when using an older coordinator (CC2530 or CC2531).

I also had a lot of trouble getting the lock paired in the first place until I got the coordinator right next to the lock (painful, since my coordinator is normally screwed to a beam in the roof...) but I now wonder if the problem was that no routers were online when I first tried. I am reluctant to unpair and repair it to find out though.

danieledwardgeorgehitchcock, 2021-09-13:
Hi,

It looks like the Yale device probably has quite a weak signal power (it is not uncommon for some cheaply made devices).

As you have managed to get this working with routers in your network and with the co-ordinator in close proximity, this goes to re-enforce this case.

How far away is your Lockwood device from your co-ordinator? Any walls, etc in-between?

I don't think this is a Zigbee2MQTT issue, I think you'd probably find this exact behaviour in whatever Zigbee network you had the device joined to.

davidmonro, 2021-09-13:
Well that is the odd thing; the coordinator is much closer than the router. The only thing between the coordinator and the lock is the ceiling and maybe 8m of air. But the lock will talk fine to an IKEA light 20m and 4 walls away, and won't talk to the coordinator at all.

The coordinator is a raspberry pi hooked up to a CC2538+CC2592 with an antenna attached, and talks very happily to all my other devices (mostly Xiaomi sensors and IKEA bulbs). The house is modern with stud internal walls so signal propagates quite well generally.

 As I said I don't know if I really had to move the coordinator to pair it, or if someone switched on an IKEA light between when I last tried with the coordinator in the ceiling and an hour or so later when I had moved it down to be near the lock.

(Because the Xiaomi devices don't generally like roaming I've got into the habit of switching off all the routers before trying to pair things, so that they pair direct with the coordinator. All my IKEA bulbs still have power switches on the walls, so I can't rely on them being powered up permanently).

Cheers

David

danieledwardgeorgehitchcock, 2021-09-13:
All of this sounds like semi-normal behaviour of Zigbee.

Due to its self-healing nature, it doesn't necessarily bind to the co-ordinator as a default, rather, what is giving it the greatest signal at that time, with a route back to the co-ordinator.

You may well find that over time, it does link itself back to the co-ordinator once things have bedded in.

You should be ok with using your Ikea devices as routers for Xiaomi (Aqara) and other picky brands as the Ikea devices use a pretty vanilla Zigbee spec, and therefore play nice with routing other devices. I would however, recommend wherever possible to keep any routers as permanently powered as best you can. This will prevent the network always having to self-heal, and you would probably get a slightly better reliability from your network.

davidmonro, 2021-09-13:
I'm surprised it didn't manage to find the coordinator for 8 hours overnight when all the ikea bulbs were off, 4 nights in a row. As soon as a bulb appears on the network, the lock comes back. It seems to be entirely reproducible.

(It would be nice if I could keep the ikea bulbs permanently powered, but that just isn't going to happen since messing with the switches would require an electrician; I really only use the zigbee functionality on them to adjust colour temperature anyway. If I can't solve the problem any other way I might mount one inside the roof space just so it won't get disconnected from power).

Kryzek, 2021-09-13:
How about adding an Ikea power outlet near your lock? If lights are getting turned on/off, maybe using one of those power outlets would be a solution? Just throwing alternative approach ideas in the air :)

MattWestb, 2021-09-14:
I think @Kryzek suggestion is one of the best.
If the outlet have more lights (routers) its connected 2 and some links is going down it can rerouting the lock to the coordinator if its possible (if no route is possibly or the distant for direct connecting, its failing but the lock without router is doing the same but earlier).

IKEA outlets have good RF performance and is using Zigbee 3 for making good handling of children and routing in the network but the outlet can have little week spots with the antenna (is one large capacitor near it on the PCB) but normally they working great and i have lot of them as "backbone" in my production network and is normally having one On/Off switch and one Aqara weather sensor and they is never having problems with its parent only the famous parent accountants bug can making little trouble but i hope IKEA is fixing it in the firmware.

closed 2021-10-22 by github-actions[bot]