Koenkk/zigbee2mqtt #8739: CC2652 Launchpad: using without XDS110 - "network commissioning timed out" - failed to start

danergo, 2021-09-13:
## What happened

I'd like to use LAUNCHXL-CC26X2R1 Evaluation board from TI with a RPi0.
Also, I'd like to use it _directly connected_ meaning that I don't need XDS110 (blue part), only the onboard CC2652R (green part).

For this purpose, I've removed all the jumpers (that interconnect XDS110 and CC2652 - yellow part):

[image]

After this, I have wired the CC2652's pins to RPi0:

```
GND - GND
3V3 - 3V3
RXD<< - GPIO14 (TXD0)
TXD>> - GPIO15 (RXD0)
RESET - serial 1kOhm - 3V3
```

TXD0 and RXD0 are visible on RPi0 as `/dev/ttyAMA0`. Root console is disabled, so nothing else is using this UART (for sure).

I have configured Z2M to use:
```
serial:
  port: /dev/ttyAMA0
```

However, Z2M's startup fails with
```
Error: network commissioning timed out - most likely network with the same panId or extendedPanId already exists nearby
```

I have read the FAQ, my CC2652 is Rev E, tried resetting it via jumpercable removal, tried many times. Flashed on Windows (by that time, jumpers were not removed yet).

## What did you expect to happen

I'd expect Z2M to successfully start.

## Debug info
Zigbee2MQTT version: 1.21.1 (commit #4a51e0c0)
Adapter hardware: CC26X2R1
Adapter firmware version: https://github.com/Koenkk/Z-Stack-firmware/raw/master/coordinator/Z-Stack_3.x.0/bin/CC2652R_coordinator_20210708.zip

## Full debug log
https://pastebin.com/uXLtuMzH

danergo, 2021-09-13:
Addition: it is also not starting if I use the common way (with jumpers and `/dev/ttyACM0`) on PI0. Is there any known blocking reason for this?

salopette, 2021-09-13:
Try to put all jumpers back on except TX RX.

danergo, 2021-09-13:
Thank you! Tried, but then there is not communication!

> Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)

danergo, 2021-09-13:
I had to let Z2M regenerating of the `pan_id` by adding `pan_id: GENERATE` into configuration's advanced section. This might be worth mentioning among the common startup fail reasons.

After this change, I could also start the Z2M without the XDS110 part of the launchpad.

closed 2021-09-13 by danergo