Koenkk/zigbee2mqtt #8746: HEIMAN HS2DB - Doorbell button

PEGARifrouf, 2021-09-13:
## Information about the device + link
This device is the a doorbell button from the HEIMAN gateway. This is a clone of the lidl door bell button :
`model: 'HG06668',
vendor: 'Lidl',
description: 'Silvercrest smart wireless door bell button',`

Link to the manufacturer: [http://www.heimantech.com/product/?type=detail&id=23](url)

I have added manually the device via the external converter method inspired by the lidl definition of it (tuya type). I was able to get:
- battery status with 100%
- battery low = 'False'
- Link quality which is varying at each action on the button
- action = N/A (not affected by an action on the button)
- temper = clear (not affected by an action on the button)

Nevertheless, just after I press the button I have the following message on the log:
`Received Zigbee message from 'Sonnette', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"extendedstatus":0,"zonestatus":32768}' from endpoint 1 with groupID 0`

So, for now I don't know how to retrieve the action of prressing the button from an home assistant entity, what can I do to link this message to a usable variable in home assistant ?

## data/database.db entry of the device
`{"id":4,"type":"EndDevice","ieeeAddr":"0x000d6f00161f67b7","nwkAddr":36600,"manufId":4619,"manufName":"HEIMAN","powerSource":"Battery","modelId":"DoorBell-EM","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[0,1,3,1280,2821],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"DoorBell-EM","manufacturerName":"HEIMAN","powerSource":3,"zclVersion":2,"appVersion":21,"stackVersion":2,"hwVersion":16,"dateCode":"2018.10.12"}},"ssIasZone":{"attributes":{"iasCieAddr":"0x00124b0024c0efc4","zoneState":1}},"genPowerCfg":{"attributes":{"batteryPercentageRemaining":200}}},"binds":[{"cluster":0,"type":"endpoint","deviceIeeeAddress":"0x00124b0024c0efc4","endpointID":1},{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x00124b0024c0efc4","endpointID":1}],"configuredReportings":[{"cluster":1,"attrId":33,"minRepIntval":3600,"maxRepIntval":62000,"repChange":0}],"meta":{}}},"appVersion":21,"stackVersion":2,"hwVersion":16,"dateCode":"2018.10.12","zclVersion":2,"interviewCompleted":true,"meta":{"configured":1618658434},"lastSeen":1631569772457}`

Koenkk, 2021-09-13:
Does `fromZigbee` contain `fz.tuya_doorbell_button`, if yes `Received Zigbee message from 'Sonnette', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"extendedstatus":0,"zonestatus":32768}' from endpoint 1 with groupID 0` should be followed up by an `action`.

PEGARifrouf, 2021-09-14:
Here is my custion definition of the device:
`const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require("zigbee-herdsman-converters/lib/tuya");

const definition = {
    // Since a lot of Tuya devices use the same modelID, but use different data points
    // it's usually necessary to provide a fingerprint instead of a zigbeeModel
    fingerprint: [
        {
            // The model ID from: Device with modelID 'TS0601' is not supported
            // You may need to add \u0000 at the end of the name in some cases
            modelID: 'DoorBell-EM',
            // The manufacturer name from: Device with modelID 'TS0601' is not supported.
            manufacturerName: 'HEIMAN'
        },
    ],
    model: 'DoorBell-EM',
    vendor: 'HEIMAN',
    description: 'Smat Door Bell',
    fromZigbee: [
        fz.battery,
        fz.tuya_doorbell_button,
        fz.ignore_basic_report
    ],
    toZigbee: [
        tz.tuya_data_point_test, // Another debug converter
    ],
    onEvent: tuya.setTime, // Add this if you are getting no converter for 'commandSetTimeRequest'
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genPowerCfg']);
        await reporting.batteryPercentageRemaining(endpoint);    
//    configure: async (device, coordinatorEndpoint, logger) => {
//       const endpoint = device.getEndpoint(1);
//        await reporting.bind(endpoint, coordinatorEndpoint, ['genBasic']);
    },
    exposes: [e.battery(), e.action(['pressed']), e.battery_low(), e.tamper()],
};

module.exports = definition;`

Is the 'fz.tuya_doorbell_button' shouldl be feeding 'e.action(['pressed'])' or a declaration is missing ?

danieledwardgeorgehitchcock, 2021-09-14:
Hi,

There is already a pull request that I submitted: Koenkk/zigbee-herdsman-converters#2873

Your definition above should now allow us to close that out and get it integrated into the codebase of Zigbee2MQTT

PEGARifrouf, 2021-09-14:
Hi,

Thanks I have not found this request before I post. Please tell me if you need some additional info on the device to get in into codebase.

Regards

danieledwardgeorgehitchcock, 2021-09-14:
In answer to your second question regarding the event,

The doorbell button when pressed should emit a momentary 'pressed' action which is the trigger that you would use in a Home Assistant automation.

I have confirmed that this is how the Lidl device works, and cannot see much of a reason why it shouldn't work in your code - although you do have a lot of Tuya related code in your external converter..

Maybe wait until the above is merged (hopefully this will happen now) and switch to the Dev branch to see if you have any better luck?

PEGARifrouf, 2021-09-14:
I tried to perform automation based on the 'presed' action with no success. The logbook always says 'none'.

I used tuya template to import the lidl version on it, I didn't change many things or cleanup comments so my definition is not very readable and, there is a possibility that unneeded code is the reason why it does not work. It is also my first one, I have migrated from ZHA because the lidl variant was supported on zigbee2MQTT.

Until the zigbee2MQTT is very pleasant to use, I have fun to play with it.

I will keep the issue about the code merging and consider switching to dev branch (I have also a nodon product which I have declared that is supported on dev branch)

danieledwardgeorgehitchcock, 2021-09-15:
@PEGARifrouf FYI, this has now been merged into the master (dev) branch

PEGARifrouf, 2021-09-15:
Thanks, the community is very reactive! So if I remove my external converter an re-install the addon the device will be supported now (in edge addon)

danieledwardgeorgehitchcock, 2021-09-15:
> Thanks, the community is very reactive! So if I remove my external converter an re-install the addon the device will be supported now (in edge addon)

Correct!

I'd remove the device before removing the external converter just to make sure there is nothing residual remaining

PEGARifrouf, 2021-09-15:
I reinstall the addon following the addon instructions, the device in not supported.
It is maybe too soon for the repo to be updated, the device in not supported.

I imported the definition of the device made in "devices/heiman.js" from 1h ago to my external converter. The device is now "supported" but no improvement on action or temper entitites.

I noticed a ! next to the temper attribute in Z2Mqtt dashboard, note sure about the meaning.
[image]

Regards

danieledwardgeorgehitchcock, 2021-09-15:
Yeah, may be a bit too early for the HA addon edge build.. keep an eye on it!

I get the same screen on mine:

[image]

But then, if I stay on the dashboard and push the button, I get this appear:

[image]

Yours should do the same..?

PEGARifrouf, 2021-09-15:
Action never appear, but LQI value is refreshed each time I press meaning the device send somthing.

danieledwardgeorgehitchcock, 2021-09-15:
If you enable the debug logs in your Zigbee2MQTT settings, do you see any events come through when the button is pressed?

PEGARifrouf, 2021-09-15:
Each time I press:
[image]

danieledwardgeorgehitchcock, 2021-09-15:
Ahh interesting! you have a different zonestatus than me! Can you post your latest External Converter file, and I will add something to it for you to test?

PEGARifrouf, 2021-09-15:
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require("zigbee-herdsman-converters/lib/tuya");

const definition = {
    // Since a lot of Tuya devices use the same modelID, but use different data points
    // it's usually necessary to provide a fingerprint instead of a zigbeeModel
            fingerprint: [{modelID: 'DoorBell-EM', manufacturerName: 'HEIMAN'}],
        model: 'HS2DB',
        vendor: 'HEIMAN',
        description: 'Smart doorbell button',
        fromZigbee: [fz.battery, fz.tuya_doorbell_button, fz.ignore_basic_report],
        toZigbee: [],
        configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint = device.getEndpoint(1);
            await reporting.bind(endpoint, coordinatorEndpoint, ['genPowerCfg']);
            await reporting.batteryPercentageRemaining(endpoint);
        },
        exposes: [e.battery(), e.action(['pressed']), e.battery_low(), e.tamper()],
};

module.exports = definition;

danieledwardgeorgehitchcock, 2021-09-15:
Can you give this a go?

```javascript
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const e = exposes.presets;

const fzLocal = {
   heiman_doorbell_button: {
        cluster: 'ssIasZone',
        type: 'commandStatusChangeNotification',
        convert: (model, msg, publish, options, meta) => {
            if (hasAlreadyProcessedMessage(msg)) return;
            const lookup = {32768: 'pressed'};
            const zoneStatus = msg.data.zonestatus;
            return {
                action: lookup[zoneStatus & 32768],
                tamper: (zoneStatus & 1<<2) > 0,
                battery_low: (zoneStatus & 1<<3) > 0,
            };
        },
    },
};

const definition = {
        fingerprint: [{modelID: 'DoorBell-EM', manufacturerName: 'HEIMAN'}],
        model: 'HS2DB',
        vendor: 'HEIMAN',
        description: 'Smart doorbell button',
        fromZigbee: [fz.battery, fzLocal.heiman_doorbell_button, fz.ignore_basic_report],
        toZigbee: [],
        configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint = device.getEndpoint(1);
            await reporting.bind(endpoint, coordinatorEndpoint, ['genPowerCfg']);
            await reporting.batteryPercentageRemaining(endpoint);
        },
        exposes: [e.battery(), e.action(['pressed']), e.battery_low(), e.tamper()],
    },

module.exports = definition;
```

PEGARifrouf, 2021-09-15:
I update the file, reboot the addon.
Behavior is the same, action is not updated

PEGARifrouf, 2021-09-15:
Is your action state is 'None' when not pressed ?

danieledwardgeorgehitchcock, 2021-09-15:
Sorry, a bit of trial and error.. What about this?

```javascript
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const e = exposes.presets;

const fzLocal = {
   heiman_doorbell_button: {
        cluster: 'ssIasZone',
        type: 'commandStatusChangeNotification',
        convert: (model, msg, publish, options, meta) => {
            if (hasAlreadyProcessedMessage(msg)) return;
            const lookup = {32768: 'pressed'};
            const zoneStatus = msg.data.zonestatus;
            return {
                action: lookup[zoneStatus],
                tamper: (zoneStatus & 1<<2) > 0,
                battery_low: (zoneStatus & 1<<3) > 0,
            };
        },
    },
};

const definition = {
    fingerprint: [{modelID: 'DoorBell-EM', manufacturerName: 'HEIMAN'}],
    model: 'HS2DB',
    vendor: 'HEIMAN',
    description: 'Smart doorbell button',
    fromZigbee: [fz.battery, fzLocal.heiman_doorbell_button, fz.ignore_basic_report],
    toZigbee: [],
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genPowerCfg']);
        await reporting.batteryPercentageRemaining(endpoint);
    },
    exposes: [e.battery(), e.action(['pressed']), e.battery_low(), e.tamper()],
};

module.exports = definition;
```

danieledwardgeorgehitchcock, 2021-09-15:
> Is your action state is 'None' when not pressed ?

My action is 'N/A' when not pressed

PEGARifrouf, 2021-09-15:
Same issue

I try to understand, action is  looking into msg.data.zonestatus for '32768: pressed'. In log there is only 32768, or maybe 'pressed' il associated to it by ":"?

danieledwardgeorgehitchcock, 2021-09-15:
Yeah if you look above, I am trying to look for the action of 3276, then I have a constant called lookup above which should map that value to 'pressed' if it works! :-D

PEGARifrouf, 2021-09-15:
I commented the line "if (hasAlreadyProcessedMessage(msg)) return;"
Got N/A when not pressed and pressed when pressesd !

Is function "hasAlreadyProcessedMessage" declared on external converters ?

danieledwardgeorgehitchcock, 2021-09-15:
Ah! I bet it needs a reference to utils!

PEGARifrouf, 2021-09-15:
It works perfectly.

Thanks

closed 2021-10-23 by github-actions[bot]